home-assistant/operating-system #1351: This seems to be closed, but I (and many others, so it seems) have this same problem:

iotinkognito, 2021-05-06:
This seems to be closed, but I (and many others, so it seems) have this same problem:

https://community.home-assistant.io/t/ext4-fs-error-validating-inode-bitmap/163287/4

https://community.home-assistant.io/t/ext4-mb-generate-buddy-747/104860/6

Nothing is being self-healed, HA seems to get more and more corrupt until it doesn't boot at all.

_Originally posted by @roybosch in https://github.com/home-assistant/operating-system/issues/441#issuecomment-601652957_

roybosch, 2021-05-06:
In my case, not the vm but the harddisk was corrupt. I had to reinstall HA on a fresh disk and that solved the problem.

iotinkognito, 2021-05-06:
Definitely not corrupted here. Everything runs fine on my local Win10 and the VM.

roybosch, 2021-05-06:
I tested this by reinstalling a HA VM. They kept getting corrupted. My windows was running fine too, but I didn't have other VM's running.

agners, 2021-05-06:
@iotinkognito ext4 corruption can occure for many many reasons. It seems that in @roybosch's case it was due to the underlying harddrive. It can also be due to software bugs in VirtualBox etc.

Please provide exact details: What kind of errors do you see (e.g. dmesg/Host logs?), what version of VirtualBox are you using? Did the issue reappear even when starting over with a new Home Assistant OS installation?

iotinkognito, 2021-07-28:
A few restarts and most errors cleared, been rock solid since recent update.

closed 2021-07-28 by iotinkognito