home-assistant/operating-system #1354: Supervisor and cli look for "hassos" on download

pgsylvestre, 2021-05-07:
Cli command and supervisor-2021.04.3 in ODROID-N2 and others look for the original spelling 'hassos' and report error 404.  

"INFO (MainThread) [supervisor.hassos] Fetch OTA update from https://github.com/home-assistant/operating-system/releases/download/6.0.rc1/hassos_odroid-n2-6.0.rc1.raucb"

wovka88, 2021-05-07:
Some on Raspberry:

21-05-07 18:17:05 INFO (MainThread) [supervisor.hassos] Fetch OTA update from https://github.com/home-assistant/operating-system/releases/download/6.0.rc1/hassos_rpi4-64-6.0.rc1.raucb
21-05-07 18:17:06 ERROR (MainThread) [supervisor.hassos] Error raise form OTA Webserver: 404

agners, 2021-05-26:
FWIW, this was due to a problem in the download URL and has been fixed back then:
https://github.com/home-assistant/version/commit/189c3a5996fd6848340843ed77cbf4af3ba4472d

closed 2021-05-26 by agners