home-assistant/operating-system #1361: Feature request: Support remote serial devices via usbip

poldim, 2021-05-15:
**Hardware Environment**

- [X] Raspberry Pi 4
- [X] Synology with HA via docker container

## System Health

version | core-2021.5.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
virtualenv | false
python_version | 3.8.9
os_name | Linux
os_version | 5.4.83-v7l
arch | armv7l
timezone | America/Los_Angeles

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 4991
Installed Version | 1.12.3
Stage | running
Available Repositories | 819
Installed Repositories | 1

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 5.13
-- | --
update_channel | stable
supervisor_version | supervisor-2021.04.3
docker_version | 19.03.15
disk_total | 28.6 GB
disk_used | 6.8 GB
healthy | true
supported | true
board | rpi4
supervisor_api | ok
version_api | ok
installed_addons | Home Assistant Google Drive Backup (0.104.3), Samba share (9.3.1), AdGuard Home (4.0.0), SSH & Web Terminal (8.2.1), IDE (2.2.0), Check Home Assistant configuration (3.7.1), AirCast (3.1.0), Z-Wave JS to MQTT (0.15.0)

</details>

<details><summary>Lovelace</summary>

dashboards | 1
-- | --
resources | 0
views | 6
mode | storage

</details>

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**

This was added to the month of what the heck (https://community.home-assistant.io/t/wth-can-remote-serial-devices-not-be-easily-mounted-as-dev-tty-virtual-devices/228148) but isn't getting any love.
I'm not sure if this belongs at the supervisor level or the OS, but hoping it could get some attention.

This post, https://community.home-assistant.io/t/rpi-as-z-wave-zigbee-over-ip-server-for-hass/23006,  outlines how you  can install this on a rpi but this doesnt work in HASSos.

thehaxxa, 2021-06-20:
I support this feature request, without it integrations have to implement ways to support remote serial devices.

Or users have to resort to using Home Assistant on Generic Linux. This is a common use case that should be supported by HASSOS.

bartgrefte, 2022-03-19:
I'm interested in support for usbip well (in the OVA image), since there are virtual machine hypervisors where passthroughing USB devices is only possible at the controller level, not individual devices. Support for usbip would help get around that.

TheBinaryLoop, 2022-04-10:
I'm giving this feature also a thumbs up

OptimusGREEN, 2022-08-04:
Yup, this would be a huge step toward high availability. thumbs up

mjkl-gh, 2022-08-05:
While researching this  I found this interesting documentation. Apparantly an USB attached to Home assistant OS can be used for more than just filling an authorized_keys file:

https://github.com/home-assistant/operating-system/blob/fd7d0253f3304b24377155252a0105fdba7920da/Documentation/configuration.md

Specifically in this case you can use it to add files to /etc/modules-load.d or /etc/modbprobe.d

> `/etc/modules-load.d` is persistent and you can add your configuration files there. See [Systemd modules load](https://www.freedesktop.org/software/systemd/man/modules-load.d.html). You can add the modules configuration files in `/etc/modprobe.d`

This persists and will be reloaded when updating. This way you can load vhci-hcd which will persist through updates. Then a [RUDY](https://github.com/git-developer/rudy) container converted to add-on would be required to mount devices over USB/IP or so it seems

Would be nice to have vhci-hcd enabled out of the box maybe?

agners, 2022-08-05:
Yeah above comment triggered that PR :smile: 

It should be possible to create an add-on which has kernel modules permissions. This could then setup USB/IP host or client depending on configuration.

warlordattack, 2022-08-13:
hello is it possible to mount usb devices over ip into HA OS ? (VM inside synology nas)
bt dongle, camera ... from other linux box

mjkl-gh, 2022-08-14:
> hello is it possible to mount usb devices over ip into HA OS ? (VM inside synology nas) bt dongle, camera ... from other linux box

Not yet, preparing changes were merged, but are not part of the current version yet.

Moreover, Additional work is required to make this possible.

Currently it shoulde be possible to load the missing kernel modules through:

https://github.com/home-assistant/operating-system/blob/fd7d0253f3304b24377155252a0105fdba7920da/Documentation/configuration.md

And another docker container is required to do the actual mounten of the hardware. [RUDY](https://github.com/git-developer/rudy) is a possibility for that. It will require some tinkering, but it should work.

Apocrathia, 2022-12-31:
I've made a preliminary stab at creating a RUDYmentary add-on. I don't have a USB/IP server to work with at the moment, so I've gotten it to the point that I can configure the script and successfully connect to my MQTT broker. If anyone wants to test it, feel free to try. I'll hopefully get a chance to get this more fleshed out soon.

https://github.com/Apocrathia/home-assistant-addons

yepeekai, 2023-04-05:
With skyconnect and "year of the voice" I think this feature make more and more sense because skyconnect is a usb device and year of the voice will require a lot of computing power so having HA OS as a VM will enable to give it a lot of power without having to give it a full machine.  Also, it makes it possible to have the skyconnect at the best spot (for signal strengh) in the house while the server sit in a not used place in the basement.

I have HA OS in a VM in HyperV.  I do blame hyper V for not having the usb assignment to VM functionality!!!  but still if HA OS can get around that limitation, it would be a welcome addition.  

I am currently working on an alternative which require HA to be installed in supervised mode which is not recommended at a lot of places in the documentation and it make stuff a lot more complex to setup ...

yepeekai, 2023-10-05:
I see that 11 RC was released but this issue is not part of it :(
When I sort active issues by how many likes it has (thumbs up) it is the first one on the list.
When will it receive the love and attention it deserves?

jschneekloth, 2023-10-14:
This would be a really useful feature.  I'm virtualizing HA, but can't really enable high availability with it due to lack of USBIP support (or similar) at OS level.  I can likely just install supervised on plain linux to get things working, but the convenience of home assistant os is really nice to have.

Timmo3, 2023-12-09:
Hi, 
I would like to see this feature as well as a built-in option. I tried the AverageJoe89 usbip mounter, but it constantly crashed (I'm running on HA Core 11.2). 
My goal would be moving HA into a proxmox environment...

edwin-2023, 2024-02-05:
And another vote for some kind of usb over ip. We really need this. At the moment it's going to become a nono for me to use HA :(

lildadou, 2024-03-17:
https://github.com/AverageJoe89/hassio-usbip-mounter seems to do the trick.

As an alternative solution, I've noticed that the zigbee2mqtt add-on lets you connect to a remote dongle using socat. It's not usb-over-IP, but when the device simply communicates via serial, it's an easier solution to implement.