home-assistant/operating-system #1362: Uhhuh. NMI received for unknown reason 00 on CPU X. Do you have a strange power saving mode enabled? Dazed and confused, but trying to continue

SixFive7, 2021-05-15:
**Hardware Environment**

- Image: hassos_ova-5.13.vhdx
- OS: Windows server 2019
- Hypervisor: Hyper-V
- Hardware: AMD Ryzen 9 5950X on a x570 chipset

**Home Assistant OS release:**
- Fresh installation of release 5.13

**Description of problem:**
- Every few minutes the log is spammed with the message:

Uhhuh. NMI received for unknown reason 00 on CPU X.
Do you have a strange power saving mode enabled?
Dazed and confused, but trying to continue

Where X is the ID of the CPU. There is a line per core added to the Hyper-V VM.

agners, 2021-05-19:
Can you try using 6.0.rc1? It comes with a new kernel, and since this is a new CPU/platform it might solved by upgrading.

SixFive7, 2021-05-20:
I'm afraid I have to report nothing has changed. Dynamic memory is possible with 6.0.rc1 so that's great. But this stream of watchdog/NMI errors and resets is still there unfortunately.

agners, 2021-05-26:
It seems that are also other reports of those messages in our Community as well as in others (https://community.clearlinux.org/t/hyper-v-dazed-and-confused/333). From what I can tell Linux receives a Non-Maskable Interrupt without reason, which it just doesn't know how to handle. Not sure who exactly throws those NMI, it might be Hyper-V or the host CPU itself (AMD Ryzen).

Maybe it can be fixed with different settings in Hyper-V? I am not familiar with Hyper-V...

But maybe its also host CPU related, updating your host BIOS and trying different settings might make sense too (is virtualization enabled in the BIOS?)

How often do the messages appear? Are they causing problems? Maybe simply ignoring them..?

SixFive7, 2021-05-26:
Hi, thanks for thinking with me. Unfortunately there aren't many settings in hyper-v to fiddle with. The bios was just brought up to date (due to some other issues) and of course virtualization is enable; otherwise Hyper-v just doesn't start.

I'm pretty sure it is related to the Home Assistant OS as I am running multiple other VMs on this machine without issue. At least windows server, debian, ubuntu and fedora coreos seem to work without issue on the same hyper-v/hardware. I've also tried to get home assistant on another host featuring an Xeon processor with the same result.

Both machines show the same result with extra CPU compatibility enabled:
[image]

So my personal differential diagnoses is that there is something with the linux/kernel version in the home assistant hyper-v image that is broken. Unfortunately I'm far removed from understanding anything on that level to help with this. I am available however to test whatever needs to be tested.

To answer your last question: I can just ignore the NMI errors; but the VM also sometimes resets with the related hyper-v log error. This prevents me from running home assistant on hyper-v outside of testing.

agners, 2021-05-27:
What would be interesting what kind of kernel version the OS's which don't show the problem come with (`uname -a`). It might be that they come with much older/or even newer kernel which don't show the problem.

From what I can tell, we have all the important kernel support for x86 machines (ACPI, AMD/Intel specific CPU drivers etc.) as well as paravirtualization support for Hyper-V (CONFIG_HYPERV=y etc.). We use a rather vanilla upstream kernel, so either we miss a configuration or something is really borked with Hyper-V and vanilla Linux (which would be a bit surprising given the rather good support for Hyper-V for Linux these days).

One thing which would be interesting is what kernel modules are loaded (`lsmod`) as well as what kernel configurations are enabled (zcat /proc/config.gz > kernel.conig, or sometimes the kernel config is stored alongside the kernel in /boot or similar).

PecceG2, 2021-10-17:
The problem persists (HASS OS v6.5)
Did anyone find a solution? Also, my HASS server encounters spontaneous reboots or hangs since I moved from Intel to Ryzen with Hyper-V.