home-assistant/operating-system #1363: Supervisor not listening

mschiegg, 2021-05-17:
**Hardware Environment**

- [x] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Generic x86-64 (like Intel NUC)
- [ ] OVA (Open Virtualization Appliance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [x] Fresh installation of release 6.0.rc1
- [ ] Updated from version x.y
- Additional information (if accessible):

## System Health

version | core-2021.5.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
virtualenv | false
python_version | 3.8.9
os_name | Linux
os_version | 5.10.17-v8
arch | aarch64
timezone | Europe/Vienna

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 4964
Installed Version | 1.12.3
Stage | running
Available Repositories | 830
Installed Repositories | 4

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 6.0.rc1
-- | --
update_channel | stable
supervisor_version | supervisor-2021.04.3
docker_version | 20.10.6
disk_total | 57.8 GB
disk_used | 4.7 GB
healthy | true
supported | true
board | rpi3-64
supervisor_api | ok
version_api | ok
installed_addons | Portainer (1.5.0), Grafana (6.3.4), Mosquitto broker (5.1.1), File editor (5.3.0), Terminal & SSH (9.1.1), AdGuard Home (4.0.0), Duck DNS (1.12.5), SQLite Web (3.1.0), WireGuard (0.5.1)

</details>

<details><summary>Lovelace</summary>

dashboards | 1
-- | --
resources | 0
mode | auto-gen

</details>

**Supervisor logs:**
```
2021-05-17 19:51:38 ERROR (MainThread) [homeassistant.components.hassio.handler] Timeout on /homeassistant/restart request
s6-svscanctl: fatal: unable to control /var/run/s6/services: supervisor not listening
[cont-finish.d] executing container finish scripts...
[cont-finish.d] done.
[s6-finish] waiting for services.
[s6-finish] sending all processes the TERM signal.
[s6-finish] sending all processes the KILL signal and exiting.
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] udev.sh: executing... 
[cont-init.d] udev.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
2021-05-17 20:21:08 WARNING (MainThread) [homeassistant.loader] You are using a custom integration alarmo which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2021-05-17 20:21:08 WARNING (MainThread) [homeassistant.loader] You are using a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2021-05-17 20:21:08 WARNING (MainThread) [homeassistant.loader] You are using a custom integration fritzbox_tools which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2021-05-17 20:21:08 WARNING (MainThread) [homeassistant.loader] You are using a custom integration deebot which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2021-05-17 20:21:32 WARNING (MainThread) [homeassistant.setup] Setup of input_text is taking over 10 seconds.
2021-05-17 20:21:32 WARNING (MainThread) [homeassistant.setup] Setup of timer is taking over 10 seconds.
2021-05-17 20:21:32 WARNING (MainThread) [homeassistant.setup] Setup of counter is taking over 10 seconds.
2021-05-17 20:21:32 WARNING (MainThread) [homeassistant.setup] Setup of input_select is taking over 10 seconds.
2021-05-17 20:21:32 WARNING (MainThread) [homeassistant.setup] Setup of python_script is taking over 10 seconds.
2021-05-17 20:21:32 WARNING (MainThread) [homeassistant.setup] Setup of zone is taking over 10 seconds.
2021-05-17 20:21:32 WARNING (MainThread) [homeassistant.setup] Setup of input_boolean is taking over 10 seconds.
2021-05-17 20:21:32 WARNING (MainThread) [homeassistant.setup] Setup of input_datetime is taking over 10 seconds.
2021-05-17 20:21:32 WARNING (MainThread) [homeassistant.setup] Setup of input_number is taking over 10 seconds.
2021-05-17 20:21:46 WARNING (MainThread) [homeassistant.config_entries] Config entry 'Flur LED' for shelly integration not ready yet; Retrying in background
2021-05-17 20:21:46 WARNING (MainThread) [homeassistant.config_entries] Config entry 'Kinderzimmer' for shelly integration not ready yet; Retrying in background
2021-05-17 20:21:46 WARNING (MainThread) [homeassistant.config_entries] Config entry 'Flur' for shelly integration not ready yet; Retrying in background
2021-05-17 20:21:46 WARNING (MainThread) [homeassistant.config_entries] Config entry 'Wohnzimmer' for shelly integration not ready yet; Retrying in background
2021-05-17 20:21:46 WARNING (MainThread) [homeassistant.config_entries] Config entry 'Wohnwand O' for shelly integration not ready yet; Retrying in background
2021-05-17 20:21:46 WARNING (MainThread) [homeassistant.config_entries] Config entry 'Schlafzimmer LED' for shelly integration not ready yet; Retrying in background
2021-05-17 20:21:46 WARNING (MainThread) [homeassistant.config_entries] Config entry 'Wohnwand U' for shelly integration not ready yet; Retrying in background
2021-05-17 20:21:46 WARNING (MainThread) [homeassistant.config_entries] Config entry 'Katzenbrunnen' for shelly integration not ready yet; Retrying in background
2021-05-17 20:21:46 WARNING (MainThread) [homeassistant.components.binary_sensor] Setup of binary_sensor platform mqtt is taking over 10 seconds.
2021-05-17 20:22:02 WARNING (MainThread) [homeassistant.components.homekit.accessories] binary_sensor.rauchmelder: Battery sensor state missing: None
2021-05-17 20:22:02 ERROR (MainThread) [pyhap.characteristic] SecuritySystemCurrentState: value=0 is an invalid value.
2021-05-17 20:22:02 ERROR (MainThread) [pyhap.characteristic] SecuritySystemTargetState: value=0 is an invalid value.
~ $ ha core log
Get "http://supervisor/core/logs": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
```

**Journal logs:**

**Kernel logs:**

**Description of problem:**

Supervisor is often unreachable. If i click a installed Addon or the Supervisor from the Frontend Menu, nothing happens and the request timed out. Sometimes it worked fine...
I'm on a fresh installed 6.0.rc1.
It's hard to get some logs because often this happens.
[image]
Once i was lucky, see core log above

agners, 2021-05-19:
Hm, haven't seen that so far. Is that on a Raspberry Pi 4? Is the system overloaded or something of that sort?

mschiegg, 2021-05-20:
> Hm, haven't seen that so far. Is that on a Raspberry Pi 4? Is the system overloaded or something of that sort?

It is on a 3b+

Maybe it's a DNS Problem. Deactivated Adguard and set DNS on the Router back to a public one. Since then the system runs faultless for a few hours. I will now watch this

mschiegg, 2021-05-21:
Okay, it was not a DNS Problem.

Are these "normal" stats?
[image]

After a reboot, the System runs fine a few hours.
And den things begin to crash
[image]

ha os reboot doesn't work either anymore.
[image]

agners, 2021-05-26:
I am guessing that those issues are all related since they can be explained by DNS type of issues. Are you sure you deactivated AdGuard completely?

mschiegg, 2021-05-31:
Recently i rollback to OS 5.13 and now everthing work's fine. I will close this Issue now,

closed 2021-05-31 by mschiegg