home-assistant/operating-system #1371: USB Device with ZWave stick disappears from HASS OS host in VM

wigster, 2021-05-21:
**Hardware Environment**

- [ ] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Generic x86-64 (like Intel NUC)
- [ ] OVA (Open Virtualization Appliance, on Intel NUC or any other hardware, please add the Hypervisor you are using)
- [x] VirtualBox 6.1.22 on Windows Server 2012R2

**Home Assistant OS release:**
- [ ] Fresh installation of release x.y
- [x] Updated from version 5.11
- Additional information (if accessible):

```text
arch: amd64
channel: stable
docker: 19.03.15
features:
- reboot
- shutdown
- services
- network
- hostname
- hassos
hassos: "5.13"
homeassistant: 2021.5.5
hostname: homeassistant
logging: info
machine: qemux86-64
operating_system: Home Assistant OS 5.13
state: running
supervisor: 2021.04.3
supported: true
supported_arch:
- amd64
- i386
timezone: Europe/Prague
```
**Supervisor logs:**

```
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[08:26:49] INFO: Starting local supervisor watchdog...
21-05-21 08:26:56 INFO (MainThread) [__main__] Initializing Supervisor setup
21-05-21 08:26:56 INFO (MainThread) [supervisor.bootstrap] Initializing Supervisor Sentry
21-05-21 08:26:56 INFO (MainThread) [supervisor.bootstrap] Seting up coresys for machine: qemux86-64
21-05-21 08:26:56 INFO (SyncWorker_0) [supervisor.docker.supervisor] Attaching to Supervisor homeassistant/amd64-hassio-supervisor with version 2021.04.3
21-05-21 08:26:56 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.INITIALIZE
21-05-21 08:26:57 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
21-05-21 08:26:57 INFO (MainThread) [__main__] Setting up Supervisor
21-05-21 08:26:57 INFO (MainThread) [supervisor.api] Starting API on 172.30.32.2
21-05-21 08:26:57 INFO (MainThread) [supervisor.hardware.monitor] Started Supervisor hardware monitor
21-05-21 08:26:58 INFO (MainThread) [supervisor.host.info] Updating local host information
21-05-21 08:26:58 INFO (MainThread) [supervisor.host.services] Updating service information
21-05-21 08:26:58 INFO (MainThread) [supervisor.host.network] Updating local network information
21-05-21 08:26:59 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
21-05-21 08:26:59 INFO (MainThread) [supervisor.host] Host information reload completed
21-05-21 08:26:59 INFO (MainThread) [supervisor.host.apparmor] Loading AppArmor Profiles: {'hassio-supervisor'}
21-05-21 08:26:59 INFO (MainThread) [supervisor.host.services] Reloading local service hassos-apparmor.service
21-05-21 08:26:59 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/amd64-hassio-dns with version 2021.04.0
21-05-21 08:26:59 INFO (MainThread) [supervisor.plugins.dns] Starting CoreDNS plugin
21-05-21 08:27:02 INFO (SyncWorker_0) [supervisor.docker.dns] Starting DNS homeassistant/amd64-hassio-dns with version 2021.04.0 - 172.30.32.3
21-05-21 08:27:02 INFO (MainThread) [supervisor.plugins.dns] Updated /etc/resolv.conf
21-05-21 08:27:02 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to homeassistant/amd64-hassio-audio with version 2021.04.0
21-05-21 08:27:03 INFO (MainThread) [supervisor.plugins.audio] Starting Audio plugin
21-05-21 08:27:05 INFO (SyncWorker_1) [supervisor.docker.audio] Starting Audio homeassistant/amd64-hassio-audio with version 2021.04.0 - 172.30.32.4
21-05-21 08:27:05 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/amd64-hassio-cli with version 2021.05.0
[... 740 lines cut ...]
[   83.171749] hassio: port 7(vethf8dfb4b) entered disabled state
[   83.171849] device vethf8dfb4b entered promiscuous mode
[   84.253189] kauditd_printk_skb: 2 callbacks suppressed
[   84.253191] audit: type=1325 audit(1621585661.577:125): table=nat family=2 entries=0
[   84.264716] audit: type=1325 audit(1621585661.588:126): table=filter family=2 entries=0
[   84.274196] audit: type=1325 audit(1621585661.597:127): table=nat family=2 entries=5
[   84.277518] audit: type=1325 audit(1621585661.600:128): table=nat family=2 entries=7
[   84.284998] audit: type=1325 audit(1621585661.608:129): table=nat family=2 entries=8
[   84.293109] audit: type=1325 audit(1621585661.617:130): table=nat family=2 entries=10
[   84.299196] audit: type=1325 audit(1621585661.623:131): table=nat family=2 entries=11
[   84.303757] audit: type=1325 audit(1621585661.628:132): table=nat family=2 entries=12
[   84.309026] audit: type=1325 audit(1621585661.633:133): table=nat family=2 entries=13
[   84.313273] audit: type=1325 audit(1621585661.636:134): table=nat family=2 entries=14
[   84.323847] eth0: renamed from veth1a0c044
[   84.340302] IPv6: ADDRCONF(NETDEV_CHANGE): vethf8dfb4b: link becomes ready
[   84.340360] hassio: port 7(vethf8dfb4b) entered blocking state
[   84.340363] hassio: port 7(vethf8dfb4b) entered forwarding state
[  144.183560] hrtimer: interrupt took 14290015 ns
[  170.521575] nf_conntrack: default automatic helper assignment has been turned off for security reasons and CT-based  firewall rule not found. Use the iptables CT target to attach helpers instead.
```
**Description of problem:**

I have a Z-Wave stick (AEOTEC) plugged into the server, which is forwarded as a USB port to the HASS VM. Every so often (once a day?), the USB device disappears from HASS, and my Z-Wave network stops working. Restarting the host from Supervisor/System brings the device back and everything works until it again disappears. I don't see anything in the logs apart from the ZWave server beginning to complain that it can't reach devices. 

I owuld stay this started happening around the 2021.4 upgrade, but I'only started using HASS in March this year, so it may have been accidental that I didn't notice this problem.

agners, 2021-05-21:
Probably also related: https://github.com/home-assistant/core/issues/21291

It seems the culprit is the `frame counter not updating; disabled` message first, which then subsequently disconnects the device.

It seems that this is a common issue with VirtualBox. Not sure if that is really Home Assistant OS's fault, it could also be a problem of VirtualBox itself (see https://www.virtualbox.org/ticket/16724).

Maybe the workaround posted here helps?
https://community.home-assistant.io/t/hassio-kernel-usb-bug-ohci-pci-frame-counter-not-updating-disabled/100871/5

wigster, 2021-05-22:
Aha that explains why this started happening: I switched USB3 to USB2 to try to problem solve some other issue. Indeed now that I've switched it back, the USB stick is not disappearing anymore. Thanks a lot!

closed 2021-05-22 by wigster

pzol, 2023-03-04:
Any news on this?