home-assistant/operating-system #1374: Feature request: addition of SNMP agent (SNMPD) to HAOS to allow and be included in monitoring by external system

ClearlyDazed, 2021-05-22:
**Hardware Environment**

- [ ] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Generic x86-64 (like Intel NUC)
- [ ] OVA (Open Virtualization Appliance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [ ] Fresh installation of release x.y
- [ ] Updated from version x.y
- Additional information (if accessible):

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**

ClearlyDazed, 2021-05-22:
Would be helpful to include HAOS in monitoring by external system even though Home Assistant can monitor itself.

agners, 2021-05-26:
Hm, I am guessing also needs configuration? Daemons which needs customizing we typically configure through D-Bus from Supervisor etc..

What specific metrics would you like to monitor?

ClearlyDazed, 2021-05-26:
> Hm, I am guessing also needs configuration? Daemons which needs customizing we typically configure through D-Bus from Supervisor etc..
> 
> What specific metrics would you like to monitor?

Thanks for the reply.
Well, I don't want to limit the ability to monitor whatever other people may thinks as important; so if the metric is available with SMNP then it should be available to monitor.

JacobErnst98, 2024-10-08:
Settings in snmpd.conf that will likely need configuring:

- sysLocation
- sysContact
- agentaddress
- rocommunity
- rouser
- extend

Adding an ability to upload files to be used with extend would make what people could monitor limitless

sairon, 2026-09-17:
Closing as not planned - HAOS is an appliance for home users, and external SNMP polling is a need of a small group running larger fleets. Doing it properly would touch several layers - it would require the net-snmp package in the OS, a way to configure `snmpd` through Supervisor and D-Bus as Stefan mentioned, and a UI for it in the frontend. That is a lot to build and maintain for rather niche feature.

Michael's [SNMPD add-on](https://github.com/mib1185/ha-addon-snmpd) covers this already. It runs `snmpd` with `host_network`, so it answers on UDP 161 on the host IP, exposes the host interfaces and system stats, and ships `extend` entries with the HAOS, Supervisor and Core versions. Community string, name, location and contact are configurable options.

If the add-on is missing a metric you need, its repo is the right place to ask.

closed 2026-09-17 by sairon