home-assistant/operating-system #1375: [Preparing Home Assistant Screen] Fresh install - Can't install homeassistant/raspberrypi4-homeassistant:2021.5.5 -> 500 Server Error

piotrkochan, 2021-05-22:
**Hardware Environment**

Raspberry Pi 4 4GB RAM
32 GB SD Card Samsung Endurance PRO
PoE Hat

**Home Assistant OS release:**
Fresh installation 

Tried 6 RC2, 5 build 13, 5 build 12

**Supervisor logs:**

I can't use ha command since it is stuck on the "preparing home assistant" page.

**Journal logs:**
---

**Kernel logs:**
---

**Description of problem:**

Fresh installation, tried multiple versions.
HassOS is stuck on the "Preparing Home Assistant" page.

In the developer tools I can see logs:

```
21-05-22 19:49:41 INFO (SyncWorker_0) [supervisor.docker.interface] Downloading docker image homeassistant/raspberrypi4-homeassistant with tag 2021.5.5.
21-05-22 19:49:57 ERROR (SyncWorker_0) [supervisor.docker.interface] Can't install homeassistant/raspberrypi4-homeassistant:2021.5.5 -> 500 Server Error for http+docker://localhost/v1.41/images/create?tag=2021.5.5&fromImage=homeassistant%2Fraspberrypi4-homeassistant: Internal Server Error ("Get "https://registry-1.docker.io/v2/": context deadline exceeded (Client.Timeout exceeded while awaiting headers)").
21-05-22 19:49:57 WARNING (MainThread) [supervisor.homeassistant.core] Error on Home Assistant installation. Retry in 30sec
21-05-22 19:50:27 INFO (SyncWorker_4) [supervisor.docker.interface] Updating image homeassistant/raspberrypi4-homeassistant:landingpage to homeassistant/raspberrypi4-homeassistant:2021.5.5
21-05-22 19:50:27 INFO (SyncWorker_4) [supervisor.docker.interface] Downloading docker image homeassistant/raspberrypi4-homeassistant with tag 2021.5.5.
21-05-22 19:50:42 ERROR (SyncWorker_4) [supervisor.docker.interface] Can't install homeassistant/raspberrypi4-homeassistant:2021.5.5 -> 500 Server Error for http+docker://localhost/v1.41/images/create?tag=2021.5.5&fromImage=homeassistant%2Fraspberrypi4-homeassistant: Internal Server Error ("Get "https://registry-1.docker.io/v2/": context deadline exceeded").
21-05-22 19:50:42 WARNING (MainThread) [supervisor.homeassistant.core] Error on Home Assistant installation. Retry in 30sec
```

jeff21-jpg, 2021-05-22:
I am just a HA user, not a developer, but I was getting the same issue (stuck on Preparing Home Assistant) with the same errors in the logs today (21-05-22) as well except I was using a fresh HA VDI Virtual Box download and running within VirtualBox on my windows OS. I tried changing HA DNS servers (8.8.8.8, 8.8.4.4, 1.1.1.1) without any success. I ended up connecting my host OS to a VPN which allowed the Virtual Box to use a new IP to access docker and the docker download worked fine. Once it downloaded I was able to disconnect from the VPN and HA continued working.

I think this may be an issue with the docker network and something related to your IP address. Perhaps you could take the Pi to a different Wifi network to get the initial download to complete.

agners, 2021-05-26:
Docker uses the `172.17.0.0/16` network and an additional Home Assistant specific network `172.30.32.0/23`. If you use (overlapping) networks in your LAN, than this can lead to conflict networking not properly working.

@jeff21-jpg, @piotrkochan is your LAN using this network ranges?

@piotrkochan can you access the shell? (e.g. on the system console via HDMI, using `login` command?) If so, can you ping 8.8.8.8?

jeff21-jpg, 2021-05-26:
@agners - No my LAN is on 192.168.1.0/24. I did a fair amount of diagnostics on this issue since it was easily reproducible and I could get it to work by connecting my VM Host OS to a VPN. I grabbed a network trace of activity within Virtual Box (using VBoxManage --nictrace1 on) and a have a network trace of the VM when works (when host OS is connected to VPN) or doesn't work. I reviewed the network captures with Wireshark. I can see in the not working version, it never even attempts a TCP connection out to docker. I see it make two DNS requests for registry-1.docker.io, one for the A records, and one for the AAAA record. It gets a response to the A record request (and it looks the same as in the working capture) but it never receives a response for the AAAA request compared to it does get a response to the AAAA request in the working version and then it goes on to initial the TCP connection out to registry-1.docker.io and works. This is the case whether the HassOS DNS is pointed at my router, 8.8.8.8, 8.8.4.4, 1.1.1.1.

At this point I think it might be something with my router/AP. My next plan was to setup a DNS server on my VM host OS and make it "authoritative" for docker.io and make sure it is properly responding to A and AAAA requests and point the HassOS DNS at that DNS server so it would get responses to both DNS requests without traversing my router.

agners, 2021-05-27:
Do get a bit more insight you can also use the system screen (console) to login to the OS shell (use `root`, then `login`). From there, nmcli can give you some insight as well as resolvectl:

```
nmcli connection show
(check the DNS configuration section)
resolvectl query registry-1.docker.io
```

raidnet-ms, 2021-07-09:
I think the problem is back again with hassio 6.1

ezankel, 2021-07-13:
It seems that this error may be caused by the default docker dns settings. Manually changing them appears to force the connection to resolve.

https://community.home-assistant.io/t/cli-not-starting-on-new-install/317855/41?u=ezankel

raidnet-ms, 2021-07-14:
> I am just a HA user, not a developer, but I was getting the same issue (stuck on Preparing Home Assistant) with the same errors in the logs today (21-05-22) as well except I was using a fresh HA VDI [...]

I think so but the problem is becoming widespread

ezankel, 2021-07-14:
> I think so but the problem is becoming widespread

Exactly, and the issue is not present in 4.15 and earlier versions I have tried, which indicates that some change to HAOS is causing this.

vegascom-jeff, 2021-07-14:
I had this same issue awhile back running HA's VirtualBox image and did quite a bit of diagnostics on it. I tried changing the DNS to 8.8.8.8, 1.1.1.1, etc. and that didn't do anything. Ultimately if I connect the host to a VPN it would start working. 

I enabled VirtualBox creating a network trace file and traced both the working (while connected to VPN) and non-working scenario. The only difference I could see is when a DNS query was made for (IIRC) registry-1.docker.io, the failing scenario would not receive any response when querying for the AAAA (IPv6) record whereas the working scenario would. It's not connecting over IPv6 to docker so I'm not sure why that would cause an issue.

I was going to do more diagnostics on it, but once it was up and working I wasn't having any further issues. It didn't seem to be fixed but it would only fail when I needed to pull any docker items like installing new supervisor items, which I don't have to do very often. I would just connect to the VPN temporarily to bypass the issue if needed.

brentbrooks70, 2021-07-15:
> Docker uses the `172.17.0.0/16` network and an additional Home Assistant specific network `172.30.32.0/23`. If you use (overlapping) networks in your LAN, than this can lead to conflict networking not properly working.
> 

I do use 172.17.0.0/24 as my internal IOT network....Would this explain why nothing happens on the initial boot and the containers do not get installed using the RPI 4 64bit install image? How would I change the default IP scheme used by the docker environment?

raidnet-ms, 2021-07-15:
I changed ISP and the installation was finished properly in 3 minutes.
That's very curious.

skosnosky, 2021-08-25:
I've today attempted to get Home Assistant running on a new Canakit rPi 4, 4GB using instructions here for the 64bit image: [https://www.home-assistant.io/installation/raspberrypi#write-the-image-to-your-boot-media], resulting in the error message
```
Waiting for the Home Assistant CLI to be ready...
[WARN] Home Assistant CLI not starting! Jump into emergency console…
```
as well.

After following threads on
1. https://community.home-assistant.io/t/problem-with-installation-on-raspberry-pi4/322840/4
2. https://community.home-assistant.io/t/problem-with-installation-on-raspberry-pi4/322840/6
3. https://community.home-assistant.io/t/cli-not-starting-on-new-install/317855
4. https://www.reddit.com/r/homeassistant/comments/od0zyh/warn_home_assistant_cli_is_not_running_jump_into/
5. https://github.com/home-assistant/operating-system/issues/920#issuecomment-872683455
and using key combo `[ctrl]+[alt]+F2` to exit to HomeAssistant OS for commandline, using `root` with no password per documentation. 

For reference, I'm connected directly to my internet router via ethernet cable, and have suspended my Pi-Hole for each round of testing without any difference in results.   

Seems that `date` command shows the appropriate UTC time (indicating that I was able to hit the timeserver during boot), however `vi /etc/timezone` shows `Etc/UTC` rather than appropriate local time. 

Also used 
`nmcli> set ipv4.dns 1.1.1.1`
`nmcli> save`
`nmcli> quit`
just to be sure, and I'm seeing appropriate values in `/etc/resolv.conf`.

Results of `ha info`;
```
arch: aarch64
channel: stable
docker: 20.10.6
...
hassos: "6.2"
homeassistant: 2021.8.8
...
machine: raspberrypi4-64
operating_system: Home Assistant OS 6.2
...
```

Results of `ha docker info`:
```
logging: journald
registries: {}
storage: overlay2
version: 20.10.6
```

Pi-hole logs are showing the DNS queries are being passed appropriately to DNS `1.1.1.1`, and I can also `ping` `1.1.1.1`, as well as `www.google.com` from the HomeAssistantOS commandline with return of `Alive!`.  The hard error mentioned above occurs regardless if I put the HomeAssistant rPi into a DMZ or allow it to hit the PiHole for DNS.  Perhaps worth noting that there are some `dnsmasq` settings on the router, but since it's getting an internet connection, I doubt that's the issue here.

This has been pretty frustrating and time-consuming to troubleshoot, and still don't have a usable system, so I figured I'd add another data point for the H-A team regarding my out-of-box setup experience.

I'm open to recommendations or can gather any data/logs needed to assist with troubleshooting further.   Tomorrow I might try flashing SDcard with v4.15 to see if that gets H-A up.

raidnet-ms, 2021-08-25:
What is your ISP?

skosnosky, 2021-08-25:
> 
> 
> What is your ISP?

There is a service called Cox Communications here that I'm currently using.

skosnosky, 2021-08-28:
v4.15 testing summary:
----------------------4.15-----------------------
Attempted flash of SDcard using H-A v4.15 rPi4 image
https://github.com/home-assistant/operating-system/releases/download/4.15/hassos_rpi4-4.15.img.gz
from 
https://github.com/home-assistant/operating-system/releases/tag/4.15

It seemed from the `dmesg` outputs that the install was able to progress much further this time.

Using a browser to navigate to http://[HA-rPi-ipaddress]:8123, I see the following:
[image]

Also getting a lot of errors from the supervisor in console,  doesn't seem to be able to complete setup.

Pi-hole logs report a queries to pkg-containers.githubusercontent.com and ghcr.io . 

Ending testing unsuccessful.
----------------------/4.15-----------------------

Tested again with v6.2, because why not.
----------------------6.2-----------------------
Pi-hole query logs showing:

whoami.home-assistant.io
version.home-assistant.io
time.cloudflare.com

Otherwise, same issues as last time.   
----------------------/6.2-----------------------

At this point the last thing I'm going to attempt is install from another location with a different ISP.  Perhaps I'll try again in a year to see if these issues are resolved.

cncb-gh, 2021-09-04:
I don't have the option of using another ISP.  Is there any way to manually install the needed files?

munnecke, 2021-09-07:
I have the same problem... the web says waiting 20 minutes, for over an hour.  The monitor attached to my Pi now is asking for a login to homeassist, but I don't know the passwords (pi/raspberry doesn't work).   Any ideas?  I'm dead in the water at the moment.

philwagn, 2021-09-11:
+1
I think I have the same problem. I flashed my SD Card with the latest image of HA for Rpi3.
At startup I get this error:
WARN Home Assistant CLI not starting! Jump into emergency console...

But http://homeassistant:8123/ show the onboarding page  with message:
"Preparing Home Assistant"

After few minutes, the CLI give another errors like:
EXT4-fs error [....] dockerd : iget: checksum invalid

Do you have also this EXT4-fs error?

cncb-gh, 2021-09-16:
I still cannot get through the install/setup.  I don't really understand why everything is not included in the install image, and some things have to be downloaded?

closed 2022-02-03 by github-actions[bot]