home-assistant/operating-system #1383: Cannot update the system starting the release 6.0.rc1

gribouk, 2021-05-29:
**Hardware Environment**

Intel NUC, I7, GEN10
- [ ] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Generic x86-64 (like Intel NUC)
- [ ] OVA (Open Virtualization Appliance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
HASS OS 6.0.rc1
- [ ] Fresh installation of release x.y
- [ ] Updated from version x.y
- Additional information (if accessible):

**Supervisor logs:**

```
21-05-29 08:05:46 INFO (SyncWorker_0) [supervisor.docker.supervisor] Attaching to Supervisor homeassistant/amd64-hassio-supervisor with version 2021.05.3
21-05-29 08:05:46 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.INITIALIZE
21-05-29 08:05:46 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
21-05-29 08:05:46 INFO (MainThread) [__main__] Setting up Supervisor
21-05-29 08:05:46 INFO (MainThread) [supervisor.api] Starting API on 172.30.32.2
21-05-29 08:05:46 INFO (MainThread) [supervisor.hardware.monitor] Started Supervisor hardware monitor
21-05-29 08:05:47 INFO (MainThread) [supervisor.host.info] Updating local host information
21-05-29 08:05:47 INFO (MainThread) [supervisor.host.services] Updating service information
21-05-29 08:05:47 INFO (MainThread) [supervisor.host.network] Updating local network information
21-05-29 08:05:47 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
21-05-29 08:05:47 INFO (MainThread) [supervisor.host] Host information reload completed
21-05-29 08:05:47 INFO (MainThread) [supervisor.host.apparmor] Loading AppArmor Profiles: {'hassio-supervisor'}
21-05-29 08:05:47 INFO (MainThread) [supervisor.host.services] Reloading local service hassos-apparmor.service
21-05-29 08:05:47 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/amd64-hassio-dns with version 2021.04.0
21-05-29 08:05:47 INFO (MainThread) [supervisor.plugins.dns] Starting CoreDNS plugin
21-05-29 08:05:47 INFO (SyncWorker_0) [supervisor.docker.dns] Starting DNS homeassistant/amd64-hassio-dns with version 2021.04.0 - 172.30.32.3
21-05-29 08:05:47 INFO (MainThread) [supervisor.plugins.dns] Updated /etc/resolv.conf
21-05-29 08:05:47 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to homeassistant/amd64-hassio-audio with version 2021.04.0
21-05-29 08:05:47 INFO (MainThread) [supervisor.plugins.audio] Starting Audio plugin
21-05-29 08:05:47 INFO (SyncWorker_1) [supervisor.docker.audio] Starting Audio homeassistant/amd64-hassio-audio with version 2021.04.0 - 172.30.32.4
21-05-29 08:05:47 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/amd64-hassio-cli with version 2021.05.1
21-05-29 08:05:47 INFO (MainThread) [supervisor.plugins.cli] Starting CLI plugin
21-05-29 08:05:48 INFO (SyncWorker_0) [supervisor.docker.cli] Starting CLI ghcr.io/home-assistant/amd64-hassio-cli with version 2021.05.1 - 172.30.32.5
21-05-29 08:05:48 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to homeassistant/amd64-hassio-observer with version 2020.10.1
21-05-29 08:05:48 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to homeassistant/amd64-hassio-multicast with version 2021.04.0
21-05-29 08:05:48 INFO (MainThread) [supervisor.plugins.multicast] Starting Multicast plugin
21-05-29 08:05:48 INFO (SyncWorker_1) [supervisor.docker.multicast] Starting Multicast homeassistant/amd64-hassio-multicast with version 2021.04.0 - Host
21-05-29 08:05:48 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/beta.json
21-05-29 08:05:49 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/generic-x86-64-homeassistant with version 2021.5.5
21-05-29 08:05:49 INFO (MainThread) [supervisor.homeassistant.secrets] Loaded 6 Home Assistant secrets
21-05-29 08:05:49 INFO (MainThread) [supervisor.hassos] Detect Home Assistant Operating System 6.0.rc1 / BootSlot A
21-05-29 08:05:50 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/core repository
21-05-29 08:05:50 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/00185a40 repository
21-05-29 08:05:50 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/75f2ff60 repository
21-05-29 08:05:50 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/a0d7b954 repository
21-05-29 08:05:50 INFO (MainThread) [supervisor.store] Loading add-ons from store: 72 all - 72 new - 0 remove
21-05-29 08:05:50 INFO (MainThread) [supervisor.addons] Found 12 installed add-ons
21-05-29 08:05:50 INFO (SyncWorker_8) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/ftp/amd64 with version 4.1.0
21-05-29 08:05:50 INFO (SyncWorker_5) [supervisor.docker.interface] Attaching to homeassistant/amd64-addon-ssh with version 9.1.3
21-05-29 08:05:50 INFO (SyncWorker_3) [supervisor.docker.interface] Attaching to homeassistant/amd64-addon-dhcp_server with version 1.2
21-05-29 08:05:50 INFO (SyncWorker_7) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/portainer/amd64 with version 1.4.0
21-05-29 08:05:50 INFO (SyncWorker_4) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/ssh/amd64 with version 8.2.3
21-05-29 08:05:50 INFO (SyncWorker_10) [supervisor.docker.interface] Attaching to homeassistant/amd64-addon-check_config with version 3.7.1
21-05-29 08:05:50 INFO (SyncWorker_6) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/vscode/amd64 with version 3.0.3
21-05-29 08:05:50 INFO (SyncWorker_9) [supervisor.docker.interface] Attaching to homeassistant/amd64-addon-samba with version 9.3.0
21-05-29 08:05:50 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to homeassistant/amd64-addon-configurator with version 5.2.0
21-05-29 08:05:50 INFO (SyncWorker_2) [supervisor.docker.interface] Attaching to 75f2ff60/amd64-addon-rhasspy with version 2.5.9
21-05-29 08:05:50 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to hassioaddons/mqtt with version 1.2.0
21-05-29 08:05:50 INFO (SyncWorker_11) [supervisor.docker.interface] Attaching to homeassistant/amd64-addon-mosquitto with version 6.0.1
21-05-29 08:05:50 INFO (MainThread) [supervisor.snapshots] Found 7 snapshot files
21-05-29 08:05:50 INFO (MainThread) [supervisor.discovery] Loaded 1 messages
21-05-29 08:05:50 INFO (MainThread) [supervisor.ingress] Loaded 12 ingress sessions
[... 31 lines cut ...]
21-05-29 08:06:33 INFO (MainThread) [supervisor.host.info] Updating local host information
21-05-29 08:06:33 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/beta.json
21-05-29 08:06:33 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.RUNNING
21-05-29 08:06:33 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.PWNED/ContextType.ADDON
21-05-29 08:06:33 INFO (MainThread) [supervisor.host.services] Updating service information
21-05-29 08:06:33 INFO (MainThread) [supervisor.host.network] Updating local network information
21-05-29 08:06:33 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.FREE_SPACE/ContextType.SYSTEM
21-05-29 08:06:33 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.SECURITY/ContextType.CORE
21-05-29 08:06:33 INFO (MainThread) [supervisor.resolution.check] System checks complete
21-05-29 08:06:33 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.RUNNING
21-05-29 08:06:34 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
21-05-29 08:06:34 INFO (MainThread) [supervisor.host] Host information reload completed
21-05-29 08:06:35 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
21-05-29 08:06:35 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state CoreState.RUNNING
21-05-29 08:06:35 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
21-05-29 08:07:14 INFO (MainThread) [supervisor.snapshots] Found 7 snapshot files
21-05-29 08:07:14 INFO (MainThread) [supervisor.snapshots] Found 7 snapshot files
```
**Journal logs:**

**Kernel logs:**

**Description of problem:**

Hello, ever after the update to  Home Assistant OS 6.0.rc1 I cannot update the system to a newer version. Supervisor gets me a message that a new update is available, I try to update, thing reboots and ... I se the message that new update is available but my system is in old version... I've atttached the supervisor logs, but I have not found anything strange in it... Any ideas?

agners, 2021-05-29:
Just tried that on my Intel NUC installation, and it seems to update fine from rc1 to rc3. Maybe Ctrl+F5 to force refresh the update page helps? Also can you  check using the terminal with `ha os info`.

gribouk, 2021-05-30:
That's the responce from the comand line:

board: generic-x86-64
boot: A
update_available: true
version: 6.0.rc1
version_latest: 6.0.rc3

I tried to update few more times - it thinks, downloads something, long anticipation .... and  rc1 again

agners, 2021-05-30:
Do you have ssh access to the OS? The process to get ssh access to the operating system is outlined here: https://developers.home-assistant.io/docs/operating-system/debugging

Once in the OS shell (use `login` if you are in the HA shell), this command prints the full log of Supervisor before restart:
```
journalctl -u hassos-supervisor -b -1
```

Another interesting log is from the update service `rauc`:
```
journalctl -u rauc -b -1
```

The last ~50 lines of each would be helpful to better understand what is wrong on your end.

agners, 2021-06-12:
@gribouk is this still an issue?

gribouk, 2021-06-12:
Oh, sorry, I've forgotten to close it. The problem was solved on it's own with 6.0 release - its finally updated...

closed 2021-06-12 by gribouk