home-assistant/operating-system #1384: 6.0 RC 3 don´t work on a CM4 with EMMC.

boklubre, 2021-05-29:
Hi,

I just made a fresh installation of haos_rpi4-64-6.0.rc3.img.xz on a Raspberry Compute Modul 4 with 2GB Ram and 16 GB EMMC. Unfortunatley it doesn´t boot. Is the EMMC not supportet in Version 6 of Home Assistant IO?
Cheers

Boris

agners, 2021-05-29:
The CM4 needs a different device tree. Make sure to have the following change in `config.txt` (should be present in the boot partition, the first partition off the image):

```
 [all]
dtoverlay=dwc2,dr_mode=host
device_tree=bcm2711-rpi-cm4.dtb
```

boklubre, 2021-05-30:
I have again taken a fresh 64bit 6.0 RC3 installation and inserted the changes before the first boot process.
Unfortunately without success. The system does not boot.
Strangely enough, the 32-bit version of the official 5 version works without any changes!

agners, 2021-05-30:
> The system does not boot.

How does that look exactly? Black screen, or some text? Is the serial debug console showing anything?

boklubre, 2021-05-31:
I normally use the Raspberry without a monitor for Home Assistant. I install it on an SD card, boot and then look in my Unify controller for the IP.
But when I have written to the EMMC memory and boot with the 64bit version, nothing happens. No IP is assigned.
I have now connected a monitor. It remains black!No image is output; again with a fresh V6.0 RC3 system and the cinfig.txt changes.

agners, 2021-05-31:
How did you flash to eMMC? Using USB Boot? Did you remove the USB Boot jumper after flashing and power cycle the board?

boklubre, 2021-05-31:
I flash the emmc with the official tool in the latest version. The 5.13 32-bit version works fine. But the 64 bit version not.
Yes I have removed the boot jumper pin and repowered the board again.

boklubre, 2021-06-01:
BTW I have a Smart Manager 4.0 with a Compute Modul 3. There ist the same problem. I have tried the official Release 5.13 in 32 bit and 64 bit. The 32 bit starts out of the box and the 64 bit don´t start.

agners, 2021-06-05:
I cannot reproduce this here.

I am using a CM4 with 2GB RAM, 16GB eMMC on the official Compute Module 4 IO Board. Used `usbboot` to get the flash exported as USB mass storage device, and then used block by block copy using `dd` to get the rpi4-64 image version 6.0.rc3 flashed. After that, I mounted the first partition and added the following two lines in `config.txt` (after `[all]` section):

```
device_tree=bcm2711-rpi-cm4.dtb
dtoverlay=dwc2,dr_mode=host
```

Remove "disable eMMC Boot" jumper, and repowered the device. HDMI shows the RPi firmware screen and then the Linux kernel booting. HAOS is up and running fine.

boklubre, 2021-06-08:
After trying a few now, I think it's a hardware problem.

I made my first attempts with a CM4, 2GB RAM & 16GB eMMC and a Waveshare Compute Module 4 PoE board. The flashing worked without any problems. The 32bit version also works as intended. The 64bit version does not start with this board. Since I didn't have an official Raspberry Compute Module 4 IO board yet, I inserted the board into a Waveshare Mini Base Board (B). In this board, the CM4 module also starts with the 64bit version without a problem.
Now I also have a Raspberry Compute Module 4 IO board available and it also works as desired.
I think this solves the problem. Thanks for the help!

closed 2021-06-08 by boklubre

agners, 2021-06-08:
Ok, thanks for the update!

lcnittl, 2021-08-24:
Thank you @agners for the device tree settings. Should I file a PR for the docs, or do I just not find the info?