home-assistant/operating-system #1385: &.0 RC3: Odroid XU4/HC2: thermal thermal_zone0: critical temperature reached (125 C), shutting down

msperl, 2021-05-30:
**Hardware Environment**

- [ ] Raspberry Pi [1/2/3/4]
- [X] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Generic x86-64 (like Intel NUC)
- [ ] OVA (Open Virtualization Appliance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [X] Fresh installation of release 6.0 RC3
- [ ] Updated from version x.y
- Additional information (if accessible):

```
# ha info
arch: armv7
channel: stable
docker: 20.10.6
features:
- reboot
- shutdown
- services
- network
- hostname
- hassos
hassos: 6.0.rc3
homeassistant: 2021.5.5
hostname: homeassistant
logging: info
machine: odroid-xu
operating_system: Home Assistant OS 6.0.rc3
state: running
supervisor: 2021.04.3
supported: true
supported_arch:
- armv7
- armhf
timezone: UTC
```
**Supervisor logs:**

does not show anything after repowered system

**Journal logs:**

(```grep -A 20 -B 20 thermal_zone /tmp/journal.txt```)
```
May 30 07:35:51 homeassistant aa9deb870a1c[418]: [cont-finish.d] 99-message.sh: exited 0.
May 30 07:35:51 homeassistant aa9deb870a1c[418]: [cont-finish.d] done.
May 30 07:35:51 homeassistant aa9deb870a1c[418]: [s6-finish] waiting for services.
May 30 07:35:52 homeassistant aa9deb870a1c[418]: [s6-finish] sending all processes the TERM signal.
May 30 07:35:55 homeassistant aa9deb870a1c[418]: [s6-finish] sending all processes the KILL signal and exiting.
May 30 07:35:55 homeassistant dockerd[418]: time="2021-05-30T07:35:55.741388310Z" level=info msg="ignoring event" container=aa9deb870a1c282299fe426ac029ae06fec4db693aa1be5e7115deed7f92cf25 module=libcontainerd namespace=moby topic=/tasks/delete type="*events.TaskDelete"
May 30 07:35:55 homeassistant dockerd[430]: time="2021-05-30T07:35:55.756113394Z" level=info msg="shim disconnected" id=aa9deb870a1c282299fe426ac029ae06fec4db693aa1be5e7115deed7f92cf25
May 30 07:35:55 homeassistant dockerd[430]: time="2021-05-30T07:35:55.757098025Z" level=error msg="copy shim log" error="read /proc/self/fd/50: file already closed"
May 30 07:35:55 homeassistant systemd[1]: var-lib-docker-overlay2-3f9717227f86a46766bc09bcab40a2c64525753128f53a098f6d66126afdc28b-merged.mount: Succeeded.
May 30 07:35:55 homeassistant systemd[1]: mnt-data-docker-overlay2-3f9717227f86a46766bc09bcab40a2c64525753128f53a098f6d66126afdc28b-merged.mount: Succeeded.
May 30 07:36:08 homeassistant hassos-supervisor[925]: 21-05-30 07:36:08 WARNING (MainThread) [supervisor.misc.tasks] Watchdog found a problem with a0d7b954_nodered!
May 30 07:36:08 homeassistant 67c39ae20c38[418]: 21-05-30 07:36:08 WARNING (MainThread) [supervisor.misc.tasks] Watchdog found a problem with a0d7b954_nodered!
May 30 07:36:08 homeassistant 67c39ae20c38[418]: 21-05-30 07:36:08 INFO (SyncWorker_3) [supervisor.docker.interface] Cleaning addon_a0d7b954_nodered application
May 30 07:36:08 homeassistant hassos-supervisor[925]: 21-05-30 07:36:08 INFO (SyncWorker_3) [supervisor.docker.interface] Cleaning addon_a0d7b954_nodered application
May 30 07:36:08 homeassistant dockerd[418]: time="2021-05-30T07:36:08.954988972Z" level=warning msg="Security options with `:` as a separator are deprecated and will be completely unsupported in 17.04, use `=` instead."
May 30 07:36:09 homeassistant systemd[1]: var-lib-docker-overlay2-fc99f30b707d4291875d2e68662426267e70e09d4077e13950317ba93b10af19\x2dinit-merged.mount: Succeeded.
May 30 07:36:09 homeassistant systemd[1]: mnt-data-docker-overlay2-fc99f30b707d4291875d2e68662426267e70e09d4077e13950317ba93b10af19\x2dinit-merged.mount: Succeeded.
May 30 07:36:09 homeassistant dockerd[418]: time="2021-05-30T07:36:09.279883031Z" level=warning msg="Security options with `:` as a separator are deprecated and will be completely unsupported in 17.04, use `=` instead."
May 30 07:36:09 homeassistant dockerd[430]: time="2021-05-30T07:36:09.508918503Z" level=info msg="starting signal loop" namespace=moby path=/run/docker/containerd/daemon/io.containerd.runtime.v2.task/moby/7a91aa452f2a28107d77ca7a85befb6e342409a251d122edde58172350ac2429 pid=29719
May 30 07:36:09 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-7a91aa452f2a28107d77ca7a85befb6e342409a251d122edde58172350ac2429-runc.L1g2j0.mount: Succeeded.
May 30 07:36:09 homeassistant kernel: thermal thermal_zone0: critical temperature reached (125 C), shutting down
May 30 07:36:09 homeassistant systemd-resolved[344]: System hostname changed to 'homeassistant'.
May 30 07:36:09 homeassistant systemd-resolved[344]: Detected conflict on homeassistant.local IN A 10.10.10.202
[... 167 lines cut ...]
44000
47000
43000
42000
Sun May 30 09:13:57 UTC 2021
46000
48000
51000
48000
44000
Sun May 30 09:14:02 UTC 2021
48000
49000
53000
50000
46000
[  286.697359] thermal thermal_zone0: critical temperature reached (132 C), shutting down
[  OK  ] Removed slice system-modprobe.slice.
[  OK  ] Stopped target Multi-User System.
[  OK  ] Stopped target Login Prompts.
[  OK  ] Stopped target HassOS hardware targets.
[  OK  ] Stopped target HassOS zram targets.
[  OK  ] Stopped target Host and Network Name Lookups.
[  OK  ] Stopped target Remote Encrypted Volumes.
```

agners, 2021-05-30:
It seems that their TMU (temperature monitoring unit?) is broken and reports bogus values from time to time:
https://github.com/hardkernel/linux/commit/378d8f85c4f4708fc4266689283fa0202ca700a3

I guess it should be straight forward to add such a work around to HAOS as well.

Is this a new issue? E.g. did that same board work with Home Assistant OS 5.13? I don't see a kernel change which looks like it could have introduced that problem, so I assume it is also a problem in 5.13..?

msperl, 2021-05-30:
i just tried to use the HC2 for the first time with HA. so, I do not know...
As it is a trial I deciced to go with a 6.0 release immediately.

agners, 2021-06-04:
This should be fixed in the next release (which most likely will be 6.0).

closed 2021-06-04 by agners