home-assistant/operating-system #1389: Intel NUC - grep: /proc/net/wireless: No such file or directory

andrewjswan, 2021-06-02:
**Hardware Environment**

- [ ] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [X] Generic x86-64 (like Intel NUC)
- [ ] OVA (Open Virtualization Appliance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [X] Fresh installation of release 2021.5.5
- [ ] Updated from version x.y
- Additional information (if accessible):

## System Health

version | core-2021.5.5
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
virtualenv | false
python_version | 3.8.9
os_name | Linux
os_version | 5.4.109
arch | x86_64
timezone | Europe/Kiev

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 4853
Installed Version | 1.12.4
Stage | running
Available Repositories | 835
Installed Repositories | 23

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 5.13
-- | --
update_channel | stable
supervisor_version | supervisor-2021.05.4
docker_version | 19.03.15
disk_total | 116.7 GB
disk_used | 2.5 GB
healthy | true
supported | true
board | intel-nuc
supervisor_api | ok
version_api | ok
installed_addons | Samba share (9.5.0), SSH & Web Terminal (8.2.3)

</details>

<details><summary>Lovelace</summary>

[... 966 lines cut ...]
[33085.417594] Bluetooth: hci0: advertising data len corrected
[33958.843208] Bluetooth: hci0: advertising data len corrected
```

**Description of problem:**

I'm migrating from Pi3 to NUK, but before a complete migration of the configuration, I checked all my sensors, and ran into a problem that there is no file for WiFi, `grep: /proc/net/wireless: No such file or directory` and consequently my sensors do not work. 
Description of the sensors:
```
  # WiFi
  - platform: command_line
    name: Wifi Level
    command: "grep 'wlp' /proc/net/wireless | awk '{print $4}'"
    unit_of_measurement: "dBm"
    value_template: '{{ value | multiply(1) }}'
    scan_interval: 180 # every 3 minutes
    
  - platform: command_line
    name: Wifi Quality
    command: "grep 'wlp' /proc/net/wireless | awk '{print $3}'"
    unit_of_measurement: "%"
    value_template: '{{ value | multiply(100) | float / 70 }}'
    scan_interval: 180 # every 3 minutes
```
They worked fine on Pi3, other people I know also work on NUC. The NUC is connected via WiFI.

andrewjswan, 2021-06-02:
[image]

andrewjswan, 2021-06-02:
[image]

agners, 2021-06-05:
Hm, this seems to be controlled by the kernel configuration `CONFIG_WEXT_PROC`. This is part of Wireless Extension, which is deprecated in the upstream kernel (see https://wireless.wiki.kernel.org/wext-statement). Not sure if its a good idea to enable this as it might get removed at one point in the future.

The recommended interface to gain information about wireless is nl80211. The `iw` tool has many commands to get information, e.g. `iw dev wlp5s0 link` has signal quality.

andrewjswan, 2021-06-05:
@agners 
[image]

agners, 2021-06-05:
The command in you automation is run in the Home Assistant Core container, which has a different environment. However, also there `iw` is not installed, so that does not work out of the box. The package is available from Alpine linux, so it would need to get added to the base container.

andrewjswan, 2021-06-05:
@agners The strangest thing is that it works for others NUC, and on RPI too, or it depends on the driver WiFi card?

agners, 2021-06-06:
It depends on the kernel. It seems that the Raspberry Pi has this deprecated option enabled.

andrewjswan, 2021-06-06:
@agners On NUC6i3SYH with work without problem, but on my NUC5i5RYK no :(

agners, 2021-06-07:
Hm, from what I can tell its not enabled in our image, so it should not work on any Intel NUC (when you use Home Assistant OS...).

However, since wireless extension is deprecated by upstream Linux, I expect this file to disappear in more and more configurations.

andrewjswan, 2021-06-07:
@agners So in the current version I can not get this information? Or are there any options? :)

agners, 2021-06-07:
No options currently from what I can tell.

andrewjswan, 2021-06-07:
Bad news :(

andrewjswan, 2021-06-07:
Is it possible to add this in the next versions?

agners, 2021-06-07:
This is an open source project, we do accept PRs ;)

andrewjswan, 2021-06-07:
To do PR, I need to understand what to change, unfortunately I am not an expert in this. :)

agners, 2021-06-08:
This will be fixed in OS release 6.0.

closed 2021-06-08 by agners