home-assistant/operating-system #1390: Host Log Retention max entires

antoweb, 2021-06-03:
### Describe the issue you are experiencing

Today my Raspberry restart itself, 
I wanted to investigate the case but when I go to see the Host logs I noticed that the dmesg log starts from the start. Can't configure or even retrieve logs before reboot to pinpoint the problem?

### What is the used version of the Supervisor?

supervisor-2021.04.3

### What type of installation are you running?

Home Assistant OS

### Which operating system are you running on?

Home Assistant Operating System

### What is the version of your installed operating system?

5.13

### What version of Home Assistant Core is installed?

core-2021.5.5

### Steps to reproduce the issue

1. 
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 5.4.83-v8 (builder@3b343e44eee4) (gcc version 9.3.0 (Buildroot -gdd4918c8)) home-assistant/supervisor#1 SMP PREEMPT Thu Apr 1 18:30:43 UTC 2021
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.4
[    0.000000] efi: Getting EFI parameters from FDT:
[    0.000000] efi: UEFI not found.
```

agners, 2021-06-05:
That said, it would be nice to have access to previous logs in Supervisor. I'll label this issue as enhancement. Please open a new issue once you were able to get the logs of the reboot issue.

sairon, 2025-02-17:
Closing this, as `ha host logs` has `-b` argument for quite a while now.

closed 2025-02-17 by sairon