home-assistant/operating-system #1392: Missing firmware file rtl_bt/rtl8761b_fw.bin

agners, 2021-06-05:
Not sure if I should log a bug/enhancement as well, seem to be having the same issue. I see the commit is completed but still in Dev. (btw thanks @agners)

```
[16263.319768] usb 1-1.4: USB disconnect, device number 4
[16642.754635] usb 1-1.2: new full-speed USB device number 5 using dwc_otg
[16642.887871] usb 1-1.2: New USB device found, idVendor=0bda, idProduct=8771, bcdDevice= 2.00
[16642.887899] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[16642.887919] usb 1-1.2: Product: Bluetooth Radio
[16642.887936] usb 1-1.2: Manufacturer: Realtek
[16642.887953] usb 1-1.2: SerialNumber: 00E04C239987
[16642.907465] Bluetooth: hci0: RTL: examining hci_ver=0a hci_rev=000b lmp_ver=0a lmp_subver=8761
[16642.909456] Bluetooth: hci0: RTL: rom_version status=0 version=1
[16642.909486] Bluetooth: hci0: RTL: loading rtl_bt/rtl8761b_fw.bin
[16642.923271] bluetooth hci0: Direct firmware load for rtl_bt/rtl8761b_fw.bin failed with error -2
[16642.923306] Bluetooth: hci0: RTL: firmware file rtl_bt/rtl8761b_fw.bin not found
```

_Originally posted by @kelos-01 in https://github.com/home-assistant/operating-system/issues/1273#issuecomment-809614902_

agners, 2021-06-05:
Unfortunately `rtl_bt/rtl8761b_fw.bin` is not available from the kernel.org linux-firmware repository, [not even in the main branch](https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/rtl_bt).

I don't feel comfortable to ship firmwares from various sources mainly due to license. So this firmware is merged upstream, it is unlikely that Home Assistant OS is going to support that particular model.

vevsvevs, 2021-06-07:
@agners maybe it's possible to export `/usr/lib/firmware/` dir somehow to make it writable or loadable via `CONFIG` USB drive?

bwalshin, 2021-06-14:
@agners looking at the kernel.org linux-firmware repo link you provided, I can see the missing rtl firmwares (rtl8761b_fw, etc.) were added 5 days ago. Is there a way to update dev to support this update?

agners, 2021-06-17:
Oh nice I see, just days after claiming its missing :) Yeah it should be fairly straight forward to include it. I'll have a look at it to include it into the next update.

bwalshin, 2021-06-21:
@agners Thank you so much for added that firmware! I updated and watched the logging as it rebooted, but it still gives the "not found" error. 

**Startup**
```
[    3.906918] Bluetooth: Core ver 2.22
[    3.907193] NET: Registered protocol family 31
[    3.907208] Bluetooth: HCI device and connection manager initialized
[    3.907229] Bluetooth: HCI socket layer initialized
[    3.907250] Bluetooth: L2CAP socket layer initialized
[    3.907292] Bluetooth: SCO socket layer initialized
[    3.927324] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[    3.949462] usbcore: registered new interface driver btusb
[    3.958924] Bluetooth: hci0: RTL: examining hci_ver=0a hci_rev=000b lmp_ver=0a lmp_subver=8761
[    3.962034] Bluetooth: hci0: RTL: rom_version status=0 version=1
[    3.962069] Bluetooth: hci0: RTL: loading rtl_bt/rtl8761b_fw.bin
[    3.967076] bluetooth hci0: Direct firmware load for rtl_bt/rtl8761b_fw.bin failed with error -2
[    3.967109] Bluetooth: hci0: RTL: firmware file rtl_bt/rtl8761b_fw.bin not found
```

**After startup - retrying**
```
[    6.543286] Bluetooth: HCI UART driver ver 2.3
[    6.543300] Bluetooth: HCI UART protocol H4 registered
[    6.543372] Bluetooth: HCI UART protocol Three-wire (H5) registered
[    6.543518] Bluetooth: HCI UART protocol Broadcom registered
[    6.570289] bcmgenet fd580000.ethernet: configuring instance for external RGMII (RX delay)
[    6.570552] bcmgenet fd580000.ethernet eth0: Link is Down
[    6.613097] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[    6.684548] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    6.684560] Bluetooth: BNEP filters: protocol multicast
[    6.684575] Bluetooth: BNEP socket layer initialized
[    6.689325] Bluetooth: hci0: RTL: examining hci_ver=0a hci_rev=000b lmp_ver=0a lmp_subver=8761
[    6.692330] Bluetooth: hci0: RTL: rom_version status=0 version=1
[    6.692345] Bluetooth: hci0: RTL: loading rtl_bt/rtl8761b_fw.bin
[    6.692398] bluetooth hci0: Direct firmware load for rtl_bt/rtl8761b_fw.bin failed with error -2
[    6.692406] Bluetooth: hci0: RTL: firmware file rtl_bt/rtl8761b_fw.bin not found
```

agners, 2021-06-21:
@bwalshin what image/device are you using? Is that with HAOS 6.1 from the beta channel?

bwalshin, 2021-06-21:
@agners Here is my device info:

Core: core-2021.6.6
Supervisor: supervisor-2021.06.4
Channel: beta

I updated my device this morning to the latest of everything and then checked the logs as it rebooted (above) and noticed the error.

agners, 2021-06-21:
What hardware are you running on?

bwalshin, 2021-06-21:
Sorry, Raspberry Pi

agners, 2021-06-21:
1/2/3/4?

We don't ship firmwares for the ones which have built-in WiFi. If yours has built-in, any reason you are using an external Wi-Fi solution?

bwalshin, 2021-06-22:
I’m looking to extend the Bluetooth range because the Bluetooth antenna is super weak. 
The dongle I’m trying to use is just a Bluetooth usb with a much more increased range.

There is built-in wifi, but I have it plugged in with ethernet and not using the wifi antenna.

agners, 2021-06-22:
Ok, yeah we so far did not see a reason to support external WiFi/Bluetooth when there is on-board WiFi/Bluetooth support. Added a new feature request for this: #1427

For other uses like @kelos-01 @GreyTeardrop  the firmware should be present in OS 6.1.

closed 2021-06-22 by agners

MaximSN, 2021-06-24:
Hello. 
@agners 
I installed the haos_ova-6.1.qcow2 image in proxmox. But bluetooth still doesn't work. 
What could be the problem?
in logs:
[    1.278555] usb 2-1: new full-speed USB device number 2 using xhci_hcd
[    1.407557] usb 2-1: New USB device found, idVendor=0bda, idProduct=8771, bcdDevice= 2.00
[    1.408502] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    1.409410] usb 2-1: Product: Bluetooth Radio
[    1.410197] usb 2-1: Manufacturer: Realtek
[    1.411204] usb 2-1: SerialNumber: 00E04C239987
[    3.420529] Bluetooth: hci0: command 0x1001 tx timeout
[   11.868510] Bluetooth: hci0: RTL: HCI_OP_READ_LOCAL_VERSION failed (-110)

agners, 2021-07-08:
-110 means "Connection timed out". No idea, maybe this stick doesn't work well in virtualized environments...

denisj07, 2021-07-28:
I struggled with the exact same issue, a Proxmox VM with USB passthrough led to the same error messages (RTL: HCI_OP_READ_LOCAL_VERSION failed (-110))

I then tried to passthrough it by hooking the Bluetooth dongle on a USB hub instead of using the USB ports on the mainboard, and it seems to get recognized now.
But the following error messages appear:

```
[    1.746382] usb 3-1: New USB device found, idVendor=0bda, idProduct=8771, bcdDevice= 2.00
[    1.747622] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    1.748756] usb 3-1: Product: Bluetooth Radio
[    1.749622] usb 3-1: Manufacturer: Realtek
[    1.750469] usb 3-1: SerialNumber: 00E04C239987
[    1.762102] Bluetooth: hci0: RTL: examining hci_ver=0a hci_rev=000b lmp_ver=0a lmp_subver=8761
[    1.766225] Bluetooth: hci0: RTL: rom_version status=0 version=1
[    1.767303] Bluetooth: hci0: RTL: loading rtl_bt/rtl8761b_fw.bin
[    1.768375] bluetooth hci0: Direct firmware load for rtl_bt/rtl8761b_fw.bin failed with error -2
[    1.769715] Bluetooth: hci0: RTL: firmware file rtl_bt/rtl8761b_fw.bin not found
```

I'm using Home Assistant OS 6.1 and the firmware file exists in /lib/firmware/rtl_bt. What am I missing here?

denisj07, 2021-07-28:
Oh sorry, I just saw the link to #1427
I guess it's fixed over there...

DivanX10, 2023-03-11:
I have published a manual on how to install a driver for usb-bluetooth with a realtek rtl8671b chip and what to do if usb-bluetooth does not work and the error failed with error -2 and not found appears?

https://gist.github.com/DivanX10/7c6ca3f325dfd853b38c04da9dce28a6#file-usb-bluetooth-easy-idea-debian-11-6