home-assistant/operating-system #1397: mDNS and systemd-resolved high CPU usage

eatoff8, 2021-06-09:
**Hardware Environment**

VirtualBox/VMWare player (both gave same symptoms)

**Home Assistant OS release:**
 - Fresh installation of release 5.13, restore from backup 5.13 same issue
 - 
## System Health

version | core-2021.6.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
virtualenv | false
python_version | 3.8.9
os_name | Linux
os_version | 5.4.109
arch | x86_64
timezone | Australia/Perth

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 4925
Installed Version | 1.12.4
Stage | running
Available Repositories | 824
Installed Repositories | 12

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 14 June 2021, 8:00
relayer_connected | true
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | true
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 5.13
-- | --
update_channel | stable
supervisor_version | supervisor-2021.05.4
docker_version | 19.03.15
disk_total | 30.8 GB
disk_used | 5.5 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | Samba share (9.5.0), File editor (5.3.1), Home Assistant Google Drive Backup (0.104.3), Mosquitto broker (6.0.1), AdGuard Home (4.1.6), Glances (0.12.1), DOODS (2), Visual Studio Code (3.4.1)

</details>

<details><summary>Lovelace</summary>

dashboards | 1
-- | --
resources | 7
views | 18
mode | storage

</details>

**Description of problem:**
High CPU utilisation from systemd-resolved. CPU usage slowly creeps up and after about 20 mins is pegged at 100%. This happened with a fresh install (no backup image used, jsut a generic user created) and when i tried restoring from a backup. I have traced it back to running mDNS on my unifi system. If I disable mDNS the CPU usage of systemd-resolved drops back to under 1%. BUt without mDNS i cannot push my chromecast devices onto another VLAN.

Here is an example of a fresh install - https://imgur.com/sWLEeuf
Here is an example of my restored from backup - https://imgur.com/undefined

agners, 2021-06-09:
When checking journalctl, do you see messages like described in #1163?

Just to understand the architecture a bit better: Is your Home Assistant installation dual-homed (e.g. in the Chromecast VLAN as well as in a regular LAN?)

agners, 2021-06-09:
Also: Do you have maybe test installations running with the same Hostname or Instance name?

eatoff8, 2021-06-10:
The home assistant install is not dual homed, just on the main LAN. I only ever had one instance running at a time. I tried renaming the host but didn't seem to make a difference.

eatoff8, 2021-06-15:
> When checking journalctl, do you see messages like described in #1163?

Yes, I see exactly what is in that issue. For whatever reason i cant copy the text from the window, but here is a screenshot - https://imgur.com/a/vTsPxFa

No other instances of home assistant are running on my network.

agners, 2021-06-17:
Can you try changing the hostname to something other than `homeassistant`? You can do that in the System tab of the Supervisor section.

eatoff8, 2021-06-18:
> Can you try changing the hostname to something other than `homeassistant`? You can do that in the System tab of the Supervisor section.

I renamed it to "HA" which fixed it. But then a restart of the VM brought it back.

agners, 2021-06-21:
I am guessing this is the mDNS feature you are using? https://help.ui.com/hc/en-us/articles/360001004034-UniFi-Best-Practices-for-Managing-AirPlay-Chromecast-on-UniFi-Network#enable

From what understand this really plays back the same mDNS packets to Home Assistant, and it seems that systemd-resolved detects that as a hostname conflict. Can you try follow this advice and disable mDNS reflector (while keeping mDNS repeater active)?

https://github.com/home-assistant/operating-system/issues/1163#issuecomment-855441335

In any case, since this really seems to be the same issue as #1163 I am closing this. Please post your findings in #1163.

closed 2021-06-21 by agners