home-assistant/operating-system #1398: Odroid N2/N2+ boot from USB

pepsonEL, 2021-06-10:
Hi
Is any chance any news to add support boot from USB for board Odroid N2/N2+ ? Now we can only use eMMC and microSD but it is very big price... but microSD is not safty. Some developer of Odroid from forum Hardkernel can help to implement it.... He write firmware petiboot for Odroid N2.

DancesWithMrBill, 2021-06-10:
Thanks for the info, but I think I will go with Thanks for the info, but I
think I will go with virtual machine on Windows

agners, 2021-06-11:
This is a duplicate of #1299. The situation is still the same.

closed 2021-06-11 by agners

pepsonEL, 2021-06-11:
And is no any chance to rebuild or something to start support boot from USB ? No chance....

agners, 2021-06-12:
It is possible, but its potentially quite time consuming and I am not interested to work on it. I think USB attached SSD are overrated. The USB stack adds a lot of overhead and complexity, which brings unreliability (just check all the issues labeled usb-ssd in this issue in this tracker).

While USB SSD are generally fast, the eMMC really comes close. Plus per my measurements, the CPU load is lower when using the eMMC (reported by benchmarking tool `fio`, N2 eMMC requies about 11% vs. a Samsung USB SSD which maxes out the CPU at 100% when running random read benchmark). What does it help you when you have all the IO bandwidth of the world when your CPU is maxed out and can't process it anyways?

Booting from it is definitely overrated, especially when you have a reliable and rather fast boot media such as eMMC. Its much better to  have two storage devices (data disk approach), this gives you a dedicated storage device for binaries, and one for data. This increases total available random IO and throughput.

pepsonEL, 2021-06-12:
But you mean to have OS on eMMC and on SSD have data with Home Assistant ? But when create snapshot with this configuration and then try it restore it also works ok when use only eMMC or on other device like RPI4 ?

I user eMMC about year or more and show me now that it is 90% lifetime....

I try move data to SSD but i have a problem with running SSH and connect to it by putty.  I done as users write in this thread:
https://community.home-assistant.io/t/ssh-into-hassos/84909/39 
and i get error:
Unable to load private key file “C:\Users\Piotrek\Desktop\key_private.ppk” (PuTTY key format too new)

Maybe you can help me this ssh to run on Odroid N2?