home-assistant/operating-system #1401: Problem after update

robons17, 2021-06-12:
**Hardware Environment**

- [ ] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [X] Generic x86-64 (like Intel NUC)
- [ ] OVA (Open Virtualization Appliance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [ ] Fresh installation of release x.y
- [X] Updated from version 5.13y
- Additional information (if accessible):

## System Health

version | core-2021.6.4
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
virtualenv | false
python_version | 3.8.9
os_name | Linux
os_version | 5.10.42
arch | x86_64
timezone | Europe/Rome

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 4784
Installed Version | 1.12.4
Stage | running
Available Repositories | 832
Installed Repositories | 7

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 6.0
-- | --
update_channel | stable
supervisor_version | supervisor-2021.06.0
docker_version | 20.10.6
disk_total | 219.4 GB
disk_used | 15.8 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Samba share (9.5.0), NGINX Home Assistant SSL proxy (3.0.1), MariaDB (2.3.0), Mosquitto broker (6.0.1), File editor (5.3.1), Duck DNS (1.12.5), Check Home Assistant configuration (3.7.1), deCONZ (6.9.0), ESPHome (1.18.0), phpMyAdmin (0.4.1), Node-RED (9.1.3), SSH & Web Terminal (8.2.3), InfluxDB (4.1.1), Network UPS Tools (0.7.0), Grafana (6.3.5)

</details>

<details><summary>Lovelace</summary>

[... 1272 lines cut ...]
[  332.578150] EXT4-fs (sda7): error count since last fsck: 627
[  332.578159] EXT4-fs (sda7): initial error at time 1608666184: __ext4_find_entry:1541: inode 37
[  332.578168] EXT4-fs (sda7): last error at time 1623502311: __ext4_find_entry:1545: inode 37
[  530.689347] EXT4-fs warning (device sda7): ext4_dirblock_csum_verify:377: inode #37: comm journalctl: No space for directory leaf checksum. Please run e2fsck -D.
[  530.689352] EXT4-fs error (device sda7): __ext4_find_entry:1545: inode #37: comm journalctl: checksumming directory block 0
[  530.690436] EXT4-fs warning (device sda7): ext4_dirblock_csum_verify:377: inode #37: comm journalctl: No space for directory leaf checksum. Please run e2fsck -D.
[  530.690439] EXT4-fs error (device sda7): __ext4_find_entry:1545: inode #37: comm journalctl: checksumming directory block 0
[  530.692108] EXT4-fs warning (device sda7): ext4_dirblock_csum_verify:377: inode #37: comm journalctl: No space for directory leaf checksum. Please run e2fsck -D.
[  530.692111] EXT4-fs error (device sda7): __ext4_find_entry:1545: inode #37: comm journalctl: checksumming directory block 0
[  530.706376] EXT4-fs warning (device sda7): ext4_dirblock_csum_verify:377: inode #37: comm journalctl: No space for directory leaf checksum. Please run e2fsck -D.
[  530.706379] EXT4-fs error (device sda7): __ext4_find_entry:1545: inode #37: comm journalctl: checksumming directory block 0
[  530.708309] EXT4-fs warning (device sda7): ext4_dirblock_csum_verify:377: inode #37: comm journalctl: No space for directory leaf checksum. Please run e2fsck -D.
[  530.708311] EXT4-fs error (device sda7): __ext4_find_entry:1545: inode #37: comm journalctl: checksumming directory block 0
[  530.710400] EXT4-fs warning (device sda7): ext4_dirblock_csum_verify:377: inode #37: comm journalctl: No space for directory leaf checksum. Please run e2fsck -D.
[  530.710402] EXT4-fs error (device sda7): __ext4_find_entry:1545: inode #37: comm journalctl: checksumming directory block 0
[  530.712327] EXT4-fs warning (device sda7): ext4_dirblock_csum_verify:377: inode #37: comm journalctl: No space for directory leaf checksum. Please run e2fsck -D.
[  530.712329] EXT4-fs error (device sda7): __ext4_find_entry:1545: inode #37: comm journalctl: checksumming directory block 0
[  530.717147] EXT4-fs warning (device sda7): ext4_dirblock_csum_verify:377: inode #37: comm journalctl: No space for directory leaf checksum. Please run e2fsck -D.
[  530.717149] EXT4-fs error (device sda7): __ext4_find_entry:1545: inode #37: comm journalctl: checksumming directory block 0
[  530.719169] EXT4-fs warning (device sda7): ext4_dirblock_csum_verify:377: inode #37: comm journalctl: No space for directory leaf checksum. Please run e2fsck -D.
[  530.719176] EXT4-fs error (device sda7): __ext4_find_entry:1545: inode #37: comm journalctl: checksumming directory block 0
[  530.721339] EXT4-fs warning (device sda7): ext4_dirblock_csum_verify:377: inode #37: comm journalctl: No space for directory leaf checksum. Please run e2fsck -D.
[  530.721342] EXT4-fs error (device sda7): __ext4_find_entry:1545: inode #37: comm journalctl: checksumming directory block 0

**Description of problem:**

agners, 2021-06-13:
Do you have enough free space on the system? When pressing Ctrl+D, does it boot up normally?

robons17, 2021-06-13:
Yes after ctrl d the system boot.
I do a fresh install and all is ok now

agners, 2021-06-17:
Ok, thanks for the update. Assuming this was a file system corruption of some sort.

closed 2021-06-17 by agners