home-assistant/operating-system #1403: HA won't start after latest HAOS6 update

PizzaPino, 2021-06-13:
**Hardware Environment**
- Raspberry Pi 4B (4GB RAM) connected to an SSD.

**Home Assistant OS release:**
- Updated from version 5.12

**Description of problem:**
After updating to OS6 my HA won't come online anymore. It's stuck in a boot loop. It recognizes the new OS6 though but something happened with the SSD. The Pi still works with my older SD card with an older HA version. I unplugged my deConz stick and the LAN cable but there's no difference. Unplugging it 3 times won't change anything either. There were no startup problems before the update.

[image]

Here's a short video of the full log: https://imgur.com/a/hnAFjcJ

agners, 2021-06-13:
The main culprit seems to be this message:

```
Can not allocate SWIOTLB buffer earlier and can't now provide you with the DMA bounce buffer
```

Do you have changes in `config.txt`? Can you post your `config.txt` and `cmdline.txt` (both files should be in the first FAT partition.

PizzaPino, 2021-06-13:
I never touched anything in that partition.

config.txt
`# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# HAOS - don't change it!
disable_splash=1
kernel=u-boot.bin

# uncomment for aarch64 bit support
arm_64bit=1

# uncomment to enable primary UART console
#enable_uart=1

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

# Uncomment to disable continous SD-card poll (for USB SSD)
#dtparam=sd_poll_once=on

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable the lirc-rpi module
#dtoverlay=lirc-rpi

# Uncomment this to enable GPIO support for RPI-RF-MOD/HM-MOD-RPI-PCB
#enable_uart=1
#dtparam=i2c_arm=on
#dtoverlay=miniuart-bt
#dtoverlay=rpi-rf-mod

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[all]
#dtoverlay=vc4-fkms-v3d
#max_framebuffers=2
`

cmdline.txt
`dwc_otg.lpm_enable=0 console=tty1
`

roybosch, 2021-06-15:
Same problem here with an Odroid N2+

erwangrouj, 2021-06-15:
Boot loop also with RPi 4B 4GB + USB nVMe SSD :
- after upgrading from v5.13 
- or after rebooting after a new V6 install

sentinel43, 2021-06-23:
Any solution for this? I had same situation with rpi3b+ and usb SSD. I needed to revert back to 5.13 and I am afraid to update it anymore.

PizzaPino, 2021-06-23:
I would also like to know if it's possible to save the data (Snapshots or only Node-Red data) from my SSD or if I'm better off redoing all of it from scratch.

agners, 2021-06-23:
@PizzaPino not sure I understand your question, but yes you can take a snapshot in the Supervsior section download it.

> Same problem here with an Odroid N2+

This is unlikely the same problem as this is related to SSD boot. What does the screen shows during boot/just before rebooting?

PizzaPino, 2021-06-23:
> @PizzaPino not sure I understand your question, but yes you can take a snapshot in the Supervsior section download it.

@agners I meant from my broken HA partition. If I can still access the data or the snapshots somehow or if it’s possible to recover the whole drive/partition. Or if you will be able to fix this so it will just connect fine to my SSD again.

Supreme-Bob, 2021-06-25:
same issue on a nuc went from 6.0 to 6.1 nuc just boot loops. it gets to around the loading kernel section then just reboots. After it tries a few times you end up at a HA OS cli with boot options you try them and it says not bootable. Was working fine before the update. About to reinstall HA and restore the snap

roybosch, 2021-07-05:
> @roybosch
> 
> > Same problem here with an Odroid N2+
> 
> This is unlikely the same problem as this is related to SSD boot. What does the screen shows during boot/just before rebooting?

I didn't connect a screen to it I'm afraid. Now it's restored to 5.13 and stable again. As my whole house depends on it, I'm not planning to risk another bootloop as it costs me a lot of time to restore everything.

felix-1234, 2021-07-06:
Same issue here. Tried to upgrade from supervisor but it did not work, got sent to UEFI-shell. Flashed my drive with 5.13 and everything works like a charm. Tries to flash my drive with 6.1, no luck. Tried two ways. 
1st. I put the haos_generic-x86-64-6.1.img on an usb-drive, booted from ubuntu on another usb-stick, used disk-manager to recover drive from image and used the haos_generic-x86-64-6.1.img. Got to this point, see image.
[image]

2nd. I used dd to write to disk, got same result as above.

Then I tried to use the 5.13 image hassos_intel-nuc-5.13.img using same method as above and voila, everything works like a charm.

Running a NUC with i7, 8GB RAM, 256GB NvME-drive

closed 2021-10-13 by github-actions[bot]