home-assistant/operating-system #1407: Filtering message "Bluetooth: hci0: advertising data len corrected"?

hvorragend, 2021-06-13:
**Hardware Environment**
- [X ] Generic x86-64 (like Intel NUC)
- [X] Proxmox VE (doesn't matter here)
- BLE device: Gigaset GTag
- TP-Link UB400 Nano USB Bluetooth 4.0 Adapter Dongle (all others and china products, too)

**Home Assistant OS release:**
- core-2021.6.4 (fresh)
- supervisor-2021.06.0 (fresh)
- Home Assistant OS 6.0 (fresh)

**Supervisor logs:**
_irrelevant_

**Journal logs:**
_irrelevant_

**Kernel logs:**
```
[187730.806363] Bluetooth: hci0: advertising data len corrected 30 -> 21
[187731.463404] Bluetooth: hci0: advertising data len corrected 30 -> 21
[187731.809399] Bluetooth: hci0: advertising data len corrected 30 -> 21
[187731.815431] Bluetooth: hci0: advertising data len corrected 30 -> 21
[187732.473475] Bluetooth: hci0: advertising data len corrected 30 -> 21
[187732.811374] Bluetooth: hci0: advertising data len corrected 30 -> 21
[187733.474376] Bluetooth: hci0: advertising data len corrected 30 -> 21
[187733.814281] Bluetooth: hci0: advertising data len corrected 30 -> 21
[187734.476536] Bluetooth: hci0: advertising data len corrected 30 -> 21
[187735.486544] Bluetooth: hci0: advertising data len corrected 30 -> 21
[187735.823456] Bluetooth: hci0: advertising data len corrected 30 -> 21
[187735.829349] Bluetooth: hci0: advertising data len corrected 30 -> 21
[187736.495269] Bluetooth: hci0: advertising data len corrected 30 -> 21
[187737.498350] Bluetooth: hci0: advertising data len corrected 30 -> 21
[187737.831297] Bluetooth: hci0: advertising data len corrected 30 -> 21
[187738.839346] Bluetooth: hci0: advertising data len corrected 30 -> 21
[187739.513329] Bluetooth: hci0: advertising data len corrected 30 -> 21
[187739.847279] Bluetooth: hci0: advertising data len corrected 30 -> 21
[187740.517445] Bluetooth: hci0: advertising data len corrected 30 -> 21
[187741.516453] Bluetooth: hci0: advertising data len corrected 30 -> 21
[187741.856340] Bluetooth: hci0: advertising data len corrected 30 -> 21
[187742.522409] Bluetooth: hci0: advertising data len corrected 30 -> 21
[187743.523313] Bluetooth: hci0: advertising data len corrected 30 -> 21
[187743.862351] Bluetooth: hci0: advertising data len corrected 30 -> 21
[187744.533446] Bluetooth: hci0: advertising data len corrected 30 -> 21
[187744.863281] Bluetooth: hci0: advertising data len corrected 30 -> 21
[187745.532534] Bluetooth: hci0: advertising data len corrected 30 -> 21
[187745.864444] Bluetooth: hci0: advertising data len corrected 30 -> 21
[187746.537360] Bluetooth: hci0: advertising data len corrected 30 -> 21
[187746.869283] Bluetooth: hci0: advertising data len corrected 30 -> 21
[187747.542217] Bluetooth: hci0: advertising data len corrected 30 -> 21
[187747.871252] Bluetooth: hci0: advertising data len corrected 30 -> 21
[187748.553397] Bluetooth: hci0: advertising data len corrected 30 -> 21
[187748.884353] Bluetooth: hci0: advertising data len corrected 30 -> 21
[187749.550323] Bluetooth: hci0: advertising data len corrected 30 -> 21
[187749.887331] Bluetooth: hci0: advertising data len corrected 30 -> 21
[187750.565362] Bluetooth: hci0: advertising data len corrected 30 -> 21
[187750.892234] Bluetooth: hci0: advertising data len corrected 30 -> 21
[187751.564334] Bluetooth: hci0: advertising data len corrected 30 -> 21
[187752.568204] Bluetooth: hci0: advertising data len corrected 30 -> 21
[187752.899406] Bluetooth: hci0: advertising data len corrected 30 -> 21
```

**Description of problem:**
I am using a Bluetooth adapter on my x64 hardware which is looking for Bluetooth low energy devices.
I am using the "room-assistant" Add-On.

I cannot use my terminal console, because this one is flooded by bluetooth message (plz see above).
I am having this problem since years on other server und periphery hardware. 
It seems to be a problem of the Gigaset Gtag beacons.
Till now I avoid these messages as followed:

1. Create a file "01-blocklist.conf" in "/etc/rsyslog.d"
2. add the following lines
    :msg,contains,"Bluetooth: hci0 advertising data length corrected" ~
    :msg,contains,"bt_err_ratelimited:" ~
3. restart service with: "sudo service rsyslog restart"

But I cannot do this in HA OS.

Is it possible to add this to this repository?

Thank you.

agners, 2021-06-13:
>  Generic x86-64 (like Intel NUC)

Are you using the `generic-x86-64` image for virtualization? The `ova` image is intended for that purpose.

> Is it possible to add this to this repository?

I rather prefer to not suppress a specific message for all users. After all, this is an error message (see https://elixir.bootlin.com/linux/v5.10.42/source/net/bluetooth/hci_event.c#L5438).

On the system console, you can use the second virtual terminal (Ctrl+Alt+F2) to get rid of the messages.

You can suppress error messages in your installation using

```
echo "2" > /proc/sys/kernel/printk
```

Or by adding `loglevel=2` to `cmdline.txt` in the boot partition.

hvorragend, 2021-06-13:
Oh, sure. I am using the ova image. Sorry. :-)

Thank you for your explanation. I know the workaround with ctrl+alt+f2, but I want to get rid of these messages in HA host logs, too. 

I don't want to suppress all error messages. Just to filter out these special message on my installation.

But I understand that this isn't possible for all users.

ACrazyConcept, 2021-06-24:
I also have these messages (except for the `30 -> 21` part) in my terminal.
But I am running Debian on an Intel NUC and 2*HA in Docker.
In one of the HA containers I use the MiFlora integration and [ble_monitor](https://github.com/custom-components/ble_monitor).
I only discovered it a few days ago since I almost never use the main terminal. But it would still be nice to find the root cause.

agners, 2021-07-08:
Closing, as possible work arounds have been proposed: https://github.com/home-assistant/operating-system/issues/1407#issuecomment-860211796

Let's hope that a future Bluetooth/kernel update gets rid of the messages entirely.

closed 2021-07-08 by agners