home-assistant/operating-system #1409: HAOS shutting down after a while after upgrading to 6.0

dewitpj, 2021-06-13:
**Hardware Environment**

- [ ] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Generic x86-64 (like Intel NUC)
- [x] OVA (Open Virtualization Appliance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [ ] Fresh installation of release x.y
- [x] Updated from version 5.y
- Additional information (if accessible):

## System Health

version | core-2021.6.4
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
virtualenv | false
python_version | 3.8.9
os_name | Linux
os_version | 5.10.42
arch | x86_64
timezone | Pacific/Auckland

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 6.0
-- | --
update_channel | beta
supervisor_version | supervisor-2021.06.0
docker_version | 20.10.6
disk_total | 30.8 GB
disk_used | 5.0 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | Samba share (9.5.0), Hot Water Control (0.9.2d), Tasmota Native Zigbee to Home Assistant (0.1.6), SSH & Web Terminal (8.2.3)

</details>

<details><summary>Lovelace</summary>

dashboards | 4
-- | --
resources | 2
views | 13
mode | storage

</details>
**Supervisor logs:**

21-06-13 15:23:22 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/beta.json
21-06-13 15:23:26 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/qemux86-64-homeassistant with version 2021.6.4
21-06-13 15:23:26 INFO (MainThread) [supervisor.homeassistant.secrets] Loaded 1 Home Assistant secrets
[... 1563 lines cut ...]
[  104.388199] hassio: port 7(vethc1d52b6) entered disabled state
[  104.527608] eth0: renamed from vethb0c3010
[  104.531605] IPv6: ADDRCONF(NETDEV_CHANGE): vethc1d52b6: link becomes ready
[  104.531639] hassio: port 7(vethc1d52b6) entered blocking state
[  104.531641] hassio: port 7(vethc1d52b6) entered forwarding state
[  109.612817] kauditd_printk_skb: 12 callbacks suppressed
[  109.612820] audit: type=1334 audit(1623597863.536:157): prog-id=15 op=LOAD
[  109.612906] audit: type=1334 audit(1623597863.536:158): prog-id=16 op=LOAD
[  109.869918] audit: type=1334 audit(1623597863.793:159): prog-id=17 op=LOAD
[  109.870019] audit: type=1334 audit(1623597863.793:160): prog-id=18 op=LOAD
[  139.879383] audit: type=1334 audit(1623597893.804:161): prog-id=16 op=UNLOAD
[  139.879389] audit: type=1334 audit(1623597893.804:162): prog-id=15 op=UNLOAD
[  140.170926] audit: type=1334 audit(1623597894.096:163): prog-id=18 op=UNLOAD
[  140.170932] audit: type=1334 audit(1623597894.096:164): prog-id=17 op=UNLOAD

**Description of problem:**

- Is the problem reproducible?
- For me, yes

- Has this been working before (is this a regression?)
- Yes, system was stable before the upgrade

- Has there been attempt to rule out hardware issues? (different SD card etc.)
- No, no changes on VM/Server hosting VM

agners, 2021-06-14:
Hm, from the logs I can't see why the system would shut down. How much memory do you assign the VM?

Can you login on the terminal and try to get the journal from the last boot?

```
journalctl -b -1
```

dewitpj, 2021-06-14:
Hey,

I have 2gig assigned (can bump to 4 if needed) - 566 meg used according to "free". I don't have journalcrl for some reason. Is it included in the base image ?

Thanks

agners, 2021-06-16:
@dewitpj journalctl is only accessible from the OS shell, either via terminal or by using SSH to connect to the underlying OS (see https://developers.home-assistant.io/docs/operating-system/debugging).

piio, 2021-06-16:
Hi,

I got similar problem - after updating OS to 6.0 after some time host stops responding for ping on wifi connection.
I configured wired connection and it looks it is available after loss of wifi.

In journal I see something strange:
Jun 16 17:21:32 home-rpi wpa_supplicant[342]: wlan0: CTRL-EVENT-DISCONNECTED bssid=76:4d:28:f9:af:97 reason=2
Jun 16 17:21:32 home-rpi wpa_supplicant[342]: dbus: wpa_dbus_property_changed: no property SessionLength in object /fi/w1/wpa_supplicant1/Interfaces/0
Jun 16 17:21:32 home-rpi NetworkManager[334]: <warn>  [1623864092.4304] sup-iface[0x3aa380,wlan0]: connection disconnected (reason 2)

Larger journal log:
 16 17:21:27 home-rpi kernel: audit: type=1325 audit(1623864087.801:184): table=nat family=2 entries=13 op=xt_replace pid=4808 subj==unconfined comm="iptables"
Jun 16 17:21:27 home-rpi audit[4809]: NETFILTER_CFG table=nat family=2 entries=14 op=xt_replace pid=4809 subj==unconfined comm="iptables"
Jun 16 17:21:28 home-rpi kernel: eth0: renamed from veth4be4a5f
Jun 16 17:21:28 home-rpi NetworkManager[334]: <info>  [1623864088.0876] device (vethf79915f): carrier: link connected
Jun 16 17:21:28 home-rpi kernel: IPv6: ADDRCONF(NETDEV_CHANGE): vethf79915f: link becomes ready
Jun 16 17:21:28 home-rpi kernel: hassio: port 8(vethf79915f) entered blocking state
Jun 16 17:21:28 home-rpi kernel: hassio: port 8(vethf79915f) entered forwarding state
Jun 16 17:21:28 home-rpi 3c8b9ad8ec07[467]: 21-06-16 17:21:28 INFO (SyncWorker_5) [supervisor.docker.addon] Starting Docker add-on homeassistant/armv7-addon-configurator with version 5.3.1
Jun 16 17:21:28 home-rpi hassos-supervisor[1229]: 21-06-16 17:21:28 INFO (SyncWorker_5) [supervisor.docker.addon] Starting Docker add-on homeassistant/armv7-addon-configurator with version 5.3.1
Jun 16 17:21:28 home-rpi 69d69b86ebfb[467]: [s6-init] making user provided files available at /var/run/s6/etc...exited 0.
Jun 16 17:21:28 home-rpi 69d69b86ebfb[467]: [s6-init] ensuring user provided files have correct perms...exited 0.
Jun 16 17:21:28 home-rpi 69d69b86ebfb[467]: [fix-attrs.d] applying ownership & permissions fixes...
Jun 16 17:21:28 home-rpi 69d69b86ebfb[467]: [fix-attrs.d] done.
Jun 16 17:21:28 home-rpi 69d69b86ebfb[467]: [cont-init.d] executing container initialization scripts...
Jun 16 17:21:28 home-rpi 69d69b86ebfb[467]: [cont-init.d] done.
Jun 16 17:21:28 home-rpi 69d69b86ebfb[467]: [services.d] starting services
Jun 16 17:21:28 home-rpi 69d69b86ebfb[467]: [services.d] done.
Jun 16 17:21:30 home-rpi hassos-supervisor[1229]: 21-06-16 17:21:30 INFO (MainThread) [supervisor.api.middleware.security] /core/info access from cebe7a76_hassio_google_drive_backup
Jun 16 17:21:30 home-rpi 3c8b9ad8ec07[467]: 21-06-16 17:21:30 INFO (MainThread) [supervisor.api.middleware.security] /core/info access from cebe7a76_hassio_google_drive_backup
Jun 16 17:21:30 home-rpi 3c8b9ad8ec07[467]: 21-06-16 17:21:30 INFO (MainThread) [supervisor.api.middleware.security] /supervisor/info access from cebe7a76_hassio_google_drive_backup
Jun 16 17:21:30 home-rpi hassos-supervisor[1229]: 21-06-16 17:21:30 INFO (MainThread) [supervisor.api.middleware.security] /supervisor/info access from cebe7a76_hassio_google_drive_backup
Jun 16 17:21:30 home-rpi 37e95124579a[467]: 06-16 19:21:30 INFO Starting server on port 8099
Jun 16 17:21:30 home-rpi 37e95124579a[467]: 06-16 19:21:30 INFO Server started
Jun 16 17:21:30 home-rpi 37e95124579a[467]: 06-16 19:21:30 INFO Syncing Snapshots
Jun 16 17:21:30 home-rpi 3c8b9ad8ec07[467]: 21-06-16 17:21:30 INFO (MainThread) [supervisor.api.middleware.security] /snapshots access from cebe7a76_hassio_google_drive_backup
Jun 16 17:21:30 home-rpi hassos-supervisor[1229]: 21-06-16 17:21:30 INFO (MainThread) [supervisor.api.middleware.security] /snapshots access from cebe7a76_hassio_google_drive_backup
Jun 16 17:21:30 home-rpi 3c8b9ad8ec07[467]: 21-06-16 17:21:30 INFO (MainThread) [supervisor.api.middleware.security] /snapshots/3441155b/info access from cebe7a76_hassio_google_drive_backup
Jun 16 17:21:30 home-rpi hassos-supervisor[1229]: 21-06-16 17:21:30 INFO (MainThread) [supervisor.api.middleware.security] /snapshots/3441155b/info access from cebe7a76_hassio_google_drive_backup
Jun 16 17:21:30 home-rpi 3c8b9ad8ec07[467]: 21-06-16 17:21:30 INFO (MainThread) [supervisor.api.middleware.security] /snapshots/c80bd594/info access from cebe7a76_hassio_google_drive_backup
Jun 16 17:21:30 home-rpi hassos-supervisor[1229]: 21-06-16 17:21:30 INFO (MainThread) [supervisor.api.middleware.security] /snapshots/c80bd594/info access from cebe7a76_hassio_google_drive_backup
Jun 16 17:21:30 home-rpi 3c8b9ad8ec07[467]: 21-06-16 17:21:30 INFO (MainThread) [supervisor.api.middleware.security] /snapshots/8e2f7a51/info access from cebe7a76_hassio_google_drive_backup
Jun 16 17:21:30 home-rpi hassos-supervisor[1229]: 21-06-16 17:21:30 INFO (MainThread) [supervisor.api.middleware.security] /snapshots/8e2f7a51/info access from cebe7a76_hassio_google_drive_backup
Jun 16 17:21:30 home-rpi 3c8b9ad8ec07[467]: 21-06-16 17:21:30 INFO (MainThread) [supervisor.api.middleware.security] /snapshots/583cc028/info access from cebe7a76_hassio_google_drive_backup
Jun 16 17:21:30 home-rpi hassos-supervisor[1229]: 21-06-16 17:21:30 INFO (MainThread) [supervisor.api.middleware.security] /snapshots/583cc028/info access from cebe7a76_hassio_google_drive_backup
[... cut ...]
Jun 16 17:22:37 home-rpi f0609d1b5cd8[467]: [INFO] 127.0.0.1:41423 - 55167 "A IN www.googleapis.com. udp 47 true 2048" NOERROR - 0 30.000916297s
Jun 16 17:22:37 home-rpi f0609d1b5cd8[467]: [ERROR] plugin/errors: 2 www.googleapis.com. A: dial tcp 1.0.0.1:853: i/o timeout
Jun 16 17:22:38 home-rpi f0609d1b5cd8[467]: [INFO] 127.0.0.1:41911 - 45713 "A IN www.googleapis.com. udp 47 true 2048" NOERROR - 0 30.000599214s
Jun 16 17:22:38 home-rpi f0609d1b5cd8[467]: [ERROR] plugin/errors: 2 www.googleapis.com. A: dial tcp 1.0.0.1:853: i/o timeout

After downgrading os to 5.13 everything is working well as before.

agners, 2021-06-17:
@piio

> I got similar problem - after updating OS to 6.0 after some time host stops responding for ping on wifi connection.

Shutdown and disconnect from WiFi doesn't seem the same issue to me. This requires a separate issue.

sjukes69, 2021-06-17:
I have the same issue running on a VM. HA unavailable. Strange thing however, i find that if I just hit enter in the cli, it starts working again for a short while. HA been rock solid for me since latest update

dewitpj, 2021-06-20:
I did a manual reboot of the box and since then it's been stable - been a few days so happy to close it for me

closed 2021-06-20 by dewitpj