home-assistant/operating-system #1410: Update failed "Can't fetch OTA update..."

grrygh, 2021-06-13:
**Hardware Environment**

-  Raspberry Pi 4

**Home Assistant OS release:**
- Updated from version 5.13
- 

**Supervisor logs:**
21-06-13 21:30:42 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
21-06-13 21:43:59 INFO (MainThread) [supervisor.api.middleware_security] /snapshots access from cebe7a76_hassio_google_drive_backup
21-06-13 21:59:19 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
21-06-13 21:59:20 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.RUNNING
21-06-13 21:59:20 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.SECURITY/ContextType.CORE
21-06-13 21:59:20 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.PWNED/ContextType.ADDON
21-06-13 21:59:20 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.FREE_SPACE/ContextType.SYSTEM
21-06-13 21:59:20 INFO (MainThread) [supervisor.resolution.check] System checks complete
21-06-13 21:59:20 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.RUNNING
21-06-13 21:59:21 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
21-06-13 21:59:21 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state CoreState.RUNNING
21-06-13 21:59:21 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
21-06-13 22:00:00 INFO (MainThread) [supervisor.api.middleware_security] /snapshots access from cebe7a76_hassio_google_drive_backup
21-06-13 22:00:01 INFO (MainThread) [supervisor.api.middleware_security] /core/info access from cebe7a76_hassio_google_drive_backup
21-06-13 22:00:01 INFO (MainThread) [supervisor.api.middleware_security] /supervisor/info access from cebe7a76_hassio_google_drive_backup
21-06-13 22:00:01 INFO (MainThread) [supervisor.api.middleware_security] /snapshots/new/full access from cebe7a76_hassio_google_drive_backup
21-06-13 22:00:01 INFO (MainThread) [supervisor.snapshots] Creating new full-snapshot with slug 6ad8c475
21-06-13 22:00:01 INFO (MainThread) [supervisor.snapshots] Snapshotting 6ad8c475 store Add-ons
21-06-13 22:00:01 INFO (MainThread) [supervisor.addons.addon] Building snapshot for add-on core_configurator
21-06-13 22:00:01 INFO (MainThread) [supervisor.addons.addon] Finish snapshot for addon core_configurator
21-06-13 22:00:01 INFO (MainThread) [supervisor.addons.addon] Building snapshot for add-on core_ssh
21-06-13 22:00:01 INFO (MainThread) [supervisor.addons.addon] Finish snapshot for addon core_ssh
21-06-13 22:00:01 INFO (MainThread) [supervisor.addons.addon] Building snapshot for add-on core_samba
21-06-13 22:00:01 INFO (MainThread) [supervisor.addons.addon] Finish snapshot for addon core_samba
21-06-13 22:00:01 INFO (MainThread) [supervisor.addons.addon] Building snapshot for add-on cebe7a76_hassio_google_drive_backup
21-06-13 22:00:01 INFO (MainThread) [supervisor.addons.addon] Finish snapshot for addon cebe7a76_hassio_google_drive_backup
21-06-13 22:00:01 INFO (MainThread) [supervisor.addons.addon] Building snapshot for add-on a0d7b954_vscode
21-06-13 22:00:14 INFO (MainThread) [supervisor.addons.addon] Finish snapshot for addon a0d7b954_vscode
21-06-13 22:00:14 INFO (MainThread) [supervisor.addons.addon] Building snapshot for add-on a0d7b954_nodered
21-06-13 22:00:14 INFO (MainThread) [supervisor.addons.addon] Finish snapshot for addon a0d7b954_nodered
21-06-13 22:00:14 INFO (MainThread) [supervisor.snapshots] Snapshotting 6ad8c475 store folders
21-06-13 22:00:14 INFO (SyncWorker_6) [supervisor.snapshots.snapshot] Snapshot folder homeassistant
21-06-13 22:00:48 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
21-06-13 22:01:08 INFO (SyncWorker_6) [supervisor.snapshots.snapshot] Snapshot folder homeassistant done
21-06-13 22:01:08 INFO (SyncWorker_2) [supervisor.snapshots.snapshot] Snapshot folder ssl
21-06-13 22:01:08 INFO (SyncWorker_2) [supervisor.snapshots.snapshot] Snapshot folder ssl done
21-06-13 22:01:08 INFO (SyncWorker_3) [supervisor.snapshots.snapshot] Snapshot folder addons/local
21-06-13 22:01:08 INFO (SyncWorker_3) [supervisor.snapshots.snapshot] Snapshot folder addons/local done
21-06-13 22:01:08 INFO (SyncWorker_1) [supervisor.snapshots.snapshot] Snapshot folder media
21-06-13 22:01:09 INFO (SyncWorker_1) [supervisor.snapshots.snapshot] Snapshot folder media done
21-06-13 22:01:09 INFO (SyncWorker_2) [supervisor.snapshots.snapshot] Snapshot folder share
21-06-13 22:01:09 INFO (SyncWorker_2) [supervisor.snapshots.snapshot] Snapshot folder share done
21-06-13 22:01:10 INFO (MainThread) [supervisor.snapshots] Creating full-snapshot with slug 6ad8c475 completed
21-06-13 22:01:10 INFO (MainThread) [supervisor.api.middleware_security] /snapshots access from cebe7a76_hassio_google_drive_backup
21-06-13 22:01:10 INFO (MainThread) [supervisor.api.middleware_security] /snapshots/6ad8c475/info access from cebe7a76_hassio_google_drive_backup
21-06-13 22:01:11 INFO (MainThread) [supervisor.api.middleware_security] /snapshots/eb6a084f/remove access from cebe7a76_hassio_google_drive_backup
21-06-13 22:01:11 INFO (MainThread) [supervisor.snapshots] Removed snapshot file eb6a084f
21-06-13 22:01:11 INFO (MainThread) [supervisor.api.middleware_security] /snapshots/6ad8c475/download access from cebe7a76_hassio_google_drive_backup
21-06-13 22:01:11 INFO (MainThread) [supervisor.api.snapshots] Downloading snapshot 6ad8c475
21-06-13 22:01:30 INFO (MainThread) [supervisor.api.middleware_security] /snapshots access from cebe7a76_hassio_google_drive_backup
21-06-13 22:01:31 INFO (MainThread) [supervisor.api.middleware_security] /snapshots access from cebe7a76_hassio_google_drive_backup
21-06-13 22:26:06 INFO (MainThread) [supervisor.host.info] Updating local host information
21-06-13 22:26:06 INFO (MainThread) [supervisor.host.services] Updating service information
21-06-13 22:26:06 INFO (MainThread) [supervisor.host.network] Updating local network information
21-06-13 22:26:09 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
21-06-13 22:26:09 INFO (MainThread) [supervisor.host] Host information reload completed
21-06-13 22:30:54 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
21-06-13 22:46:55 INFO (MainThread) [supervisor.hassos] Fetch OTA update from https://github.com/home-assistant/operating-system/releases/download/6.0/haos_rpi4-64-6.0.raucb
21-06-13 22:46:57 ERROR (MainThread) [supervisor.hassos] Can't fetch OTA update from https://github.com/home-assistant/operating-system/releases/download/6.0/haos_rpi4-64-6.0.raucb: Response payload is not completed
21-06-13 22:59:20 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/hassio-addons/repository repository
21-06-13 22:59:20 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/home-assistant/addons repository
[... 522 lines cut ...]
[  113.928206] audit: type=1325 audit(1623590207.290:132): table=nat family=2 entries=12
[  113.931912] audit: type=1325 audit(1623590207.294:133): table=nat family=2 entries=13
[  113.976138] eth0: renamed from veth0ad64b2
[  113.991430] IPv6: ADDRCONF(NETDEV_CHANGE): vethd27ce07: link becomes ready
[  113.991582] hassio: port 7(vethd27ce07) entered blocking state
[  113.991590] hassio: port 7(vethd27ce07) entered forwarding state
[  115.254582] hassio: port 8(vethf1d93c0) entered blocking state
[  115.254590] hassio: port 8(vethf1d93c0) entered disabled state
[  115.254841] device vethf1d93c0 entered promiscuous mode
[  115.902062] eth0: renamed from veth2c2e00e
[  115.917879] IPv6: ADDRCONF(NETDEV_CHANGE): vethf1d93c0: link becomes ready
[  115.917989] hassio: port 8(vethf1d93c0) entered blocking state
[  115.917994] hassio: port 8(vethf1d93c0) entered forwarding state
[ 2659.651176] kauditd_printk_skb: 15 callbacks suppressed
[ 2659.651182] audit: type=1400 audit(1623592752.983:149): apparmor="STATUS" operation="profile_replace" info="same as current profile, skipping" profile="unconfined" name="hassio-supervisor" pid=7044 comm="apparmor_parser"
[ 2659.651200] audit: type=1400 audit(1623592752.983:149): apparmor="STATUS" operation="profile_replace" info="same as current profile, skipping" profile="unconfined" name="hassio-supervisor///usr/bin/gdbus" pid=7044 comm="apparmor_parser"
[... cut ...]
[27988.210438] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off

**Description of problem:**
I am trying to upgrade the Operating System from 5.13 to the latest 6.0 but encountering an error prompt "Update failed" with the message "Can't fetch OTA update from https://github.com/home-assistant/operating-system/releases/download/6.0/haos_rpi4-64-6.0.raucb: Response payload is not completed"
Had tried to reload/restart supervisor, restart core, reboot host, the issue still persist.

agners, 2021-06-16:
Hm, looks like a network type of issue. Can you successfully download the file from a regular desktop? Is the system time of the Raspberry Pi 4 correct? You can use the `date` command in the Terminal to check.

closed 2021-07-03 by grrygh

bitboss-ca, 2023-12-06:
I believe this is a network issue, in that [for some people fetching objects directly from GitHub can be extremely slow](https://www.google.ca/search?q=github+raw+fetch+slow).  Is it possible to either extend the download timeout so it will eventually work, or manually download the file(s) and place them somewhere where the update process will find them?