home-assistant/operating-system #1411: RaspBerry Pi 4 with PoE+ HAT, fan not working

shubhamkastwar1, 2021-06-14:
Hi,

I am new to HA and I bought a Rpi4B and installed Rpi new PoE+ hat but the fan is not working on it, Please help.

**Hardware Environment**

- [* ] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Generic x86-64 (like Intel NUC)
- [ ] OVA (Open Virtualization Appliance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [* ] Fresh installation of release x.y
- [ ] Updated from version x.y
- Additional information (if accessible):

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**

shubhamkastwar1, 2021-06-14:
System Health

version: core-2021.6.4
installation_type: Home Assistant OS
dev: false
hassio: true
docker: true
virtualenv: false
python_version: 3.8.9
os_name: Linux
os_version: 5.10.17-v8
arch: aarch64
timezone: Asia/Kolkata

GitHub API: ok
Github API Calls Remaining: 4925
Installed Version: 1.12.4
Stage: running
Available Repositories: 842
Installed Repositories: 3

logged_in: false
can_reach_cert_server: ok
can_reach_cloud_auth: ok
can_reach_cloud: ok

host_os: Home Assistant OS 6.0
update_channel: stable
supervisor_version: supervisor-2021.06.0
docker_version: 20.10.6
disk_total: 28.6 GB
disk_used: 5.9 GB
healthy: true
supported: true
board: rpi4-64
supervisor_api: ok
version_api: ok
installed_addons: File editor (5.3.1), Mosquitto broker (6.0.1), Samba share (9.5.0), SSH & Web Terminal (8.2.3), TasmoAdmin (0.15.0)

dashboards: 1
resources: 0
views: 5
mode: storage

shubhamkastwar1, 2021-06-14:
I have gone through this but no luck

https://github.com/home-assistant/operating-system/issues/518#issuecomment-670978810

shubhamkastwar1, 2021-06-14:
Hi adding below worked for me

**dtoverlay=rpi-poe**
dtparam=poe_fan_temp0=10000,poe_fan_temp0_hyst=1000
dtparam=poe_fan_temp1=55000,poe_fan_temp1_hyst=5000
dtparam=poe_fan_temp2=60000,poe_fan_temp2_hyst=5000
dtparam=poe_fan_temp3=65000,poe_fan_temp3_hyst=5000

closed 2021-06-14 by shubhamkastwar1

georgeboot, 2021-07-04:
Slightly related, but I can't measure the power the PoE+ hat provides through the OS.

It's supposed to be measured using:
```bash
cat /sys/devices/platform/rpi-poe-power-supply@0/power_supply/rpi-poe/current_now
```

Perhaps there are some modules that need to be enabled in the kernel to support the new hat out of the box?

agners, 2021-07-08:
@georgeboot can you open a new issue for that?

We use the Raspberry Pi kernel along with their kernel configuration, so it should work if it does on Raspbian.

georgeboot, 2021-07-09:
@agners See #1450 

Shouldn't this issue be reopened as it still required manual config changes to the `config.txt` to get it working?