home-assistant/operating-system #1420: Error 404 since upgrade from 5.x to 6.0.

Schloerhome, 2021-06-17:
**Hardware Environment**

- [x] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Generic x86-64 (like Intel NUC)
- [ ] OVA (Open Virtualization Appliance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [ ] Fresh installation of release x.y
- [x] Updated from version x.y
- Additional information (if accessible):

**Supervisor logs:**

```Home Assistant logo
Preparing Home Assistant
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] udev.sh: executing...
[10:46:28] INFO: Using udev information from host
[cont-init.d] udev.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[10:46:28] INFO: Starting local supervisor watchdog...
21-06-17 10:46:30 INFO (MainThread) [__main__] Initializing Supervisor setup
21-06-17 10:46:30 INFO (MainThread) [supervisor.bootstrap] Initializing Supervisor Sentry
21-06-17 10:46:30 INFO (MainThread) [supervisor.bootstrap] Seting up coresys for machine: raspberrypi4-64
21-06-17 10:46:30 INFO (SyncWorker_0) [supervisor.docker.supervisor] Attaching to Supervisor homeassistant/aarch64-hassio-supervisor with version 2021.06.3
21-06-17 10:46:30 INFO (SyncWorker_0) [supervisor.docker.supervisor] Connecting Supervisor to hassio-network
21-06-17 10:46:31 INFO (SyncWorker_0) [supervisor.docker.interface] Cleanup images: ['homeassistant/aarch64-hassio-supervisor:2021.05.4']
21-06-17 10:46:32 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.INITIALIZE
21-06-17 10:46:32 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
21-06-17 10:46:32 INFO (MainThread) [__main__] Setting up Supervisor
21-06-17 10:46:32 INFO (MainThread) [supervisor.api] Starting API on 172.30.32.2
21-06-17 10:46:33 INFO (MainThread) [supervisor.hardware.monitor] Started Supervisor hardware monitor
21-06-17 10:46:34 INFO (MainThread) [supervisor.host.info] Updating local host information
21-06-17 10:46:34 INFO (MainThread) [supervisor.host.services] Updating service information
21-06-17 10:46:34 INFO (MainThread) [supervisor.host.network] Updating local network information
21-06-17 10:46:36 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
21-06-17 10:46:36 INFO (MainThread) [supervisor.host] Host information reload completed
21-06-17 10:46:36 INFO (MainThread) [supervisor.host.apparmor] Loading AppArmor Profiles: {'hassio-supervisor'}
21-06-17 10:46:36 INFO (MainThread) [supervisor.host.services] Reloading local service hassos-apparmor.service
21-06-17 10:46:36 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/aarch64-hassio-dns with version 2021.04.0
21-06-17 10:46:36 INFO (MainThread) [supervisor.plugins.dns] Updated /etc/resolv.conf
21-06-17 10:46:36 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/aarch64-hassio-audio with version 2021.04.0
21-06-17 10:46:36 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/aarch64-hassio-cli with version 2021.05.1
21-06-17 10:46:36 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/aarch64-hassio-observer with version 2021.06.0
21-06-17 10:46:36 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/aarch64-hassio-multicast with version 2021.04.0
21-06-17 10:46:36 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
21-06-17 10:46:38 INFO (MainThread) [supervisor.plugins] dns does not have the latest version 2021.06.0, updating
21-06-17 10:46:38 INFO (SyncWorker_0) [supervisor.docker.interface] Updating image homeassistant/aarch64-hassio-dns:2021.04.0 to ghcr.io/home-assistant/aarch64-hassio-dns:2021.06.0
21-06-17 10:46:38 INFO (SyncWorker_0) [supervisor.docker.interface] Downloading docker image ghcr.io/home-assistant/aarch64-hassio-dns with tag 2021.06.0.
21-06-17 10:47:14 INFO (SyncWorker_0) [supervisor.docker.interface] Stopping hassio_dns application
21-06-17 10:47:18 INFO (SyncWorker_0) [supervisor.docker.interface] Cleaning hassio_dns application
21-06-17 10:47:18 INFO (SyncWorker_0) [supervisor.docker.interface] Cleanup images: ['homeassistant/aarch64-hassio-dns:2021.04.0']
21-06-17 10:47:18 INFO (MainThread) [supervisor.plugins.dns] Starting CoreDNS plugin
21-06-17 10:47:20 INFO (SyncWorker_0) [supervisor.docker.dns] Starting DNS ghcr.io/home-assistant/aarch64-hassio-dns with version 2021.06.0 - 172.30.32.3
21-06-17 10:47:20 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
21-06-17 10:47:20 INFO (MainThread) [supervisor.homeassistant.secrets] Loaded 0 Home Assistant secrets
21-06-17 10:47:20 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/raspberrypi4-64-homeassistant with version landingpage
21-06-17 10:47:20 INFO (MainThread) [supervisor.homeassistant.core] Starting HomeAssistant landingpage
21-06-17 10:47:20 INFO (MainThread) [supervisor.hassos] Detect Home Assistant Operating System 6.0 / BootSlot A
21-06-17 10:47:21 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/core repository
21-06-17 10:47:21 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/a0d7b954 repository
21-06-17 10:47:21 INFO (MainThread) [supervisor.store] Loading add-ons from store: 63 all - 63 new - 0 remove
[... 299 lines cut ...]
21-06-17 12:00:46 ERROR (SyncWorker_1) [supervisor.docker.interface] Can't install ghcr.io/home-assistant/raspberrypi4-64-homeassistant:2021.6.5: 404 Client Error for http+docker://localhost/v1.41/images/ghcr.io/home-assistant/raspberrypi4-64-homeassistant:2021.6.5/json: Not Found ("no such image: ghcr.io/home-assistant/raspberrypi4-64-homeassistant:2021.6.5: No such image: ghcr.io/home-assistant/raspberrypi4-64-homeassistant:2021.6.5")
21-06-17 12:00:46 WARNING (MainThread) [supervisor.homeassistant.core] Error on Home Assistant installation. Retry in 30sec
21-06-17 12:01:16 INFO (SyncWorker_2) [supervisor.docker.interface] Updating image homeassistant/raspberrypi4-64-homeassistant:landingpage to ghcr.io/home-assistant/raspberrypi4-64-homeassistant:2021.6.5
21-06-17 12:01:16 INFO (SyncWorker_2) [supervisor.docker.interface] Downloading docker image ghcr.io/home-assistant/raspberrypi4-64-homeassistant with tag 2021.6.5.```
\\\
**Journal logs:**
<!--
- Enable SSH on OS level and login, then use `journalctl`
-->

**Kernel logs:**
<!--
- use this command: dmesg
- Enable SSH on OS level and login, then use `dmesg`.
-->

**Description of problem:**
<!--
- Is the problem reproducible?
- Has this been working before (is this a regression?)
- Has there been attempt to rule out hardware issues? (different SD card etc.)
-->
Updated to 6.0 from 5.13 and supervisor started acting weird, ended update having to restore back to 5.13 but the problem persisted. Did a fresh Install and same thing. any ideas ? 

I'm using a 120GB SSD for my installation.

agners, 2021-06-17:
Looks like internet connectivity issues. Can you [ssh into the underlying OS](https://developers.home-assistant.io/docs/operating-system/debugging) and see if internet connectivity works (`ping 8.8.8.8` as well as `ping google.com`).

Schloerhome, 2021-06-17:
> Looks like internet connectivity issues. Can you [ssh into the underlying OS](https://developers.home-assistant.io/docs/operating-system/debugging) and see if internet connectivity works (`ping 8.8.8.8` as well as `ping google.com`).

Ping 8.8.8.8 is alive.

agners, 2021-06-17:
And does `ping google.com` work?

Schloerhome, 2021-06-17:
> And does `ping google.com` work?

Yes Google.com is Alive too.

agners, 2021-06-17:
And you still get 404 error in `ha su logs`?

Can you try manually pulling the image?

```
docker pull ghcr.io/home-assistant/raspberrypi4-64-homeassistant:2021.6.5
```

Schloerhome, 2021-06-17:
```Use `ha` to access the Home Assistant CLI.

# docker pull ghcr.io/home-assistant/raspberrypi4-64-homeassistant:2021.6.5

2021.6.5: Pulling from home-assistant/raspberrypi4-64-homeassistant

d2f70382dc9a: Pull complete

583f7c33801c: Pull complete

1dad098da82f: Pull complete

72a4d1581b72: Pull complete

6c1940940b10: Downloading  20.44MB/20.44MB

ecaaae6db58b: Download complete

6a2de11d8fa5: Download complete

296ae4c21377: Download complete

ae0fca5eeb0a: Downloading  8.547MB/8.547MB

dc6d1b588c79: Download complete

fa6e324a03b7: Download complete

f046b13746ff: Download complete

12a372b12da3: Download complete

4ea5238d6181: Download complete

57ce4320995f: Download complete

ebc785f7a5f2: Download complete

966b464a25d2: Download complete

60802da30d9e: Downloading
[==================================================>]  188.4MB/188.4MB

f6d83a41ea95: Download complete

b2dc8d37cc61: Download complete

8ba7db6c1331: Download complete

03af87c74700: Download complete

43623c2e4690: Download complete

local error: tls: bad record MAC

agners, 2021-06-17:
So it aborted in the middle of the oepration with this?
```
local error: tls: bad record MAC
```

Is the date on your system correct (you can check using `date` command in the command line).

Schloerhome, 2021-06-17:
> So it aborted in the middle of the oepration with this?
> 
> ```
> local error: tls: bad record MAC
> ```
> 
> Is the date on your system correct (you can check using `date` command in the command line).

Yup correct. Date is indeed correct but in UTC.

agners, 2021-06-17:
Time is relevant when checking validity of TLS certificates, but when the code is doing that it takes time zones into account. So the date being in UTC is fine.

So it aborted in the middle of the oepration with this?

```
local error: tls: bad record MAC
```

To me it seems like a firewall or similar device is messing with our (SSL) http and/or TCP/IP connection.

Schloerhome, 2021-06-17:
> Time is relevant when checking validity of TLS certificates, but when the code is doing that it takes time zones into account. So the date being in UTC is fine.
>
> So it aborted in the middle of the [...]

Oh man! think it might be my wireguard.  Let me deactivate it and try again. 

Thanks!!

Schloerhome, 2021-06-17:
Yup it was! thanks a lot! life saver.

closed 2021-06-17 by Schloerhome