home-assistant/operating-system #1421: Update of Supervisor failed: Can't install homeassistant/aarch64-hassio-supervisor:2021.06.3: 500 Server Error for http+docker://localhost/v1.41/images/create?tag=2021.06.3&fromImage=homeassistant%2Faarch64-hassio-supervisor: Internal Server Error ("Get "https://registry-1.docker.io/v2/": x509: certificate has expired or is not yet valid: current time 2021-02-02T17:41:39Z is before 2021-04-25T00:00:00Z")

uxiogeno, 2021-06-18:
**Hardware Environment**

- [ ] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Generic x86-64 (like Intel NUC)
- [ ] OVA (Open Virtualization Appliance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [ ] Fresh installation of release x.y
- [ ] Updated from version x.y
- Additional information (if accessible):

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**

uxiogeno, 2021-06-18:
soy un novato en esto, y no se cual es el problema, ni como resolverlo

agners, 2021-06-19:
Please fill out the issue form with all data you have (like OS version).

It seems that your Raspberry Pi RTC is not set correctly. This likely resolves after a reboot.

Also the latest release fixes this problem by making sure the RTC is set correctly before continue booting, see #1360. Installing Home Assistant Version 6 likely resolves the problem.

closed 2021-06-19 by agners