home-assistant/operating-system #1426: Failed to start Grow File System on /mnt/data

MrAlfabet, 2021-06-20:
Installed HassOS OVA 3.6 in Proxmox hypervisor way back when, upgrades always worked fine (all the way up to the last 5.X release). After updating to 6.0 however, I'm greeted with this message upon boot in the terminal/VNC:

```
[FAILED] Failed to start Grow File System on /mnt/data
See 'systemctl status systemd-growfs@mnt-data.service' for details.
```

The service status is the same as the guy in this thread: https://community.home-assistant.io/t/hass-os-6-0-vdi-after-upgrade-fails-every-boot-failed-to-start-grow-file-system-on-mnt-data/315819

I've tried to disable the service with systemctl disable, but that doesn't seem to work/persist through reboot.

Ctrl-D to continue works like a charm, no difference in performance is noticed from HA. Only after my full backup once a week (with reboot) do I need to login to the console and press ctrl-D. 

My VM has a single disk (on zfs on my hypervisor) of 32GB and an EFI disk of 128K, OVMF (UEFI) bios, i440fx machine type (default), LSI 53C895A SCSI controller (default) and a virtio network device.

**Home Assistant OS release:**

System Health

version: core-2021.6.5
installation_type: Home Assistant OS
dev: false
hassio: true
docker: true
virtualenv: false
python_version: 3.8.9
os_name: Linux
os_version: 5.10.42
arch: x86_64
timezone: Europe/Amsterdam

GitHub API: ok
Github API Calls Remaining: 4714
Installed Version: 1.12.4
Stage: running
Available Repositories: 914
Installed Repositories: 13

logged_in: false
can_reach_cert_server: ok
can_reach_cloud_auth: ok
can_reach_cloud: ok

host_os: Home Assistant OS 6.0
update_channel: stable
supervisor_version: supervisor-2021.06.3
docker_version: 20.10.6
disk_total: 30.8 GB
disk_used: 17.3 GB
healthy: true
supported: true
board: ova
supervisor_api: ok
version_api: ok
installed_addons: Check Home Assistant configuration (3.7.1), Terminal & SSH (9.1.3), Grafana (7.0.1), InfluxDB (4.1.1), File editor (5.3.1), Node-RED (9.1.4), Glances (0.12.2), Samba share (9.5.1), TasmoAdmin (0.15.0), Visual Studio Code (3.4.1), ESPHome (1.19.1), SQLite Web (3.1.0), Portainer (1.5.1), Zigbee2mqtt (1.19.1-1)

dashboards: 3
resources: 8
views: 7
mode: storage

**Supervisor logs:**
21-06-20 20:12:14 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to homeassistant/amd64-hassio-cli with version 2021.05.1
21-06-20 20:12:14 INFO (MainThread) [supervisor.plugins.cli] Starting CLI plugin
21-06-20 20:12:14 INFO (SyncWorker_1) [supervisor.docker.interface] Cleaning hassio_cli application
21-06-20 20:12:15 INFO (SyncWorker_1) [supervisor.docker.cli] Starting CLI homeassistant/amd64-hassio-cli with version 2021.05.1 - 172.30.32.5
21-06-20 20:12:15 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/amd64-hassio-observer with version 2021.06.0
21-06-20 20:12:15 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/amd64-hassio-multicast with version 2021.04.0
21-06-20 20:12:15 INFO (MainThread) [supervisor.plugins.multicast] Starting Multicast plugin
21-06-20 20:12:15 INFO (SyncWorker_0) [supervisor.docker.interface] Cleaning hassio_multicast application
[... 75 lines cut ...]
21-06-20 20:13:13 WARNING (MainThread) [supervisor.addons.options] Option 'interface' does not exist in the schema for Samba share (core_samba)
21-06-20 20:13:13 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.SECURITY/ContextType.CORE
21-06-20 20:13:13 INFO (MainThread) [supervisor.resolution.check] System checks complete
21-06-20 20:13:13 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.RUNNING
21-06-20 20:13:14 INFO (MainThread) [supervisor.host.services] Updating service information
21-06-20 20:13:14 INFO (MainThread) [supervisor.host.network] Updating local network information
21-06-20 20:13:15 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
21-06-20 20:13:15 INFO (MainThread) [supervisor.host] Host information reload completed
21-06-20 20:13:16 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
21-06-20 20:13:16 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state CoreState.RUNNING
21-06-20 20:13:16 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
21-06-20 20:13:54 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
21-06-20 20:13:54 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
21-06-20 20:13:54 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request running
21-06-20 20:13:58 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
21-06-20 20:13:58 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
21-06-20 20:13:58 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request running

**Journal logs:**

**Kernel logs:**

I don't know why, but I can't for the life of me enable ssh on the host VM. Did the whole shebang with a usbstick (passthrough) with an authorized_keys file in ANSI and linux line endings, no dice ([supervisor.hardware.monitor] Ignore device /sys/devices/pci0000:00/0000:00:1e.0/0000:01:1b.0/usb2/2-2/2-2:1.0/host8/target8:0:0/8:0:0:0/block/sdb / failes to initialize by udev). Any other way to ssh into the root os? Or maybe access it from the terminal?

**Description of problem:**

agners, 2021-06-21:
You get root shell access through virtual terminal (login using root, then use `login` to get to the root shell).

What happens if you try to run systemd-growfs manually by pressing enter to get to the root shell and executing `/usr/lib/systemd/systemd-growfs /mnt/data/`?

agners, 2021-06-21:
Can you also check `dmesg` after trying to resize.

MrAlfabet, 2021-06-21:
> You get root shell access through virtual terminal (login using root, then use `login` to get to the root shell).
> 
> What happens if you try to run systemd-growfs manually by pressing enter to get to the root shell and executing `/usr/lib/systemd/systemd-growfs /mnt/data/`?

What's the user/password for the core-ssh login? It's apparently not the same as logging into the ssh addon.

MrAlfabet, 2021-06-21:
I see there's a new HA core update, would you rather have me try updating, or fixing this first?

MrAlfabet, 2021-06-21:
I've finally managed to login over ssh with port 22222, but when I do 'login <enter>' the ssh connection gets closed.

agners, 2021-06-21:
Hm, it seems that this is the culprit:

```
[ 5.883500] EXT4-fs warning (device sda8): ext4_resize_begin:72: There are errors in the filesystem, so online resizing is not allowed
...
[ 18.038847] EXT4-fs warning (device sda8): ext4_resize_begin:72: There are errors in the filesystem, so online resizing is not allowed
```

Normally the file system should get checked on startup, can you check the logs using:

```
systemctl status systemd-fsck@dev-disk-by\\x2dlabel-hassos\\x2ddata.service
```

I think manually running the check again should involve something like
```
systemctl stop mnt-data.service
systemctl stop systemd-fsck@dev-disk-by\\x2dlabel-hassos\\x2ddata.service
systemctl start systemd-fsck@dev-disk-by\\x2dlabel-hassos\\x2ddata.service
```

MrAlfabet, 2021-06-21:
# systemctl status systemd-fsck@dev-disk-by\\x2dlabel-hassos\\x2ddata.service
● systemd-fsck@dev-disk-by\x2dlabel-hassos\x2ddata.service - File System Check on /dev/disk/by-label/hassos-data
     Loaded: loaded (/usr/lib/systemd/system/systemd-fsck@.service; static)
     Active: active (exited) since Mon 2021-06-21 11:18:52 UTC; 1h 28min ago
       Docs: man:systemd-fsck@.service(8)
    Process: 415 ExecStart=/usr/lib/systemd/systemd-fsck /dev/disk/by-label/hassos-data (code=exited, status=0/SUCCESS)
   Main PID: 415 (code=exited, status=0/SUCCESS)

Jun 21 11:18:52 homeassistant systemd[1]: Starting File System Check on /dev/disk/by-label/hassos-data...
Jun 21 11:18:52 homeassistant systemd[1]: Finished File System Check on /dev/disk/by-label/hassos-data.

agners, 2021-06-21:
> During startup I get the message [FAILED] Failed to start File System Check on /dev/disk/by-label/hassos-data, and then to run the command you specified above.

That explains why growing the partition then fails.

What is fsck saying?

```
umount /mnt/data
fsck.ext4 /dev/disk/by-label/hassos-data
```

MrAlfabet, 2021-06-21:
> > During startup I get the message [FAILED] Failed to start File System Check on /dev/disk/by-label/hassos-data, and then to run the command you specified above.
> 
> That explains why growing the partition then fails.
> 
> What is fsck saying?
> 
> ```
> umount -A /dev/disk/by-label/hassos-data
> fsck.ext4 /dev/disk/by-label/hassos-data
> ```

What do I have to stop to unmount /mnt/data? I stopped HA core, but still get 'target is busy' when trying to unmount.

agners, 2021-06-21:
Hm, if you started Home Assistant OS completely, then there are quite some services which use that partition.

I think we can ask systemd to fix the file system. Can you edit `cmdline.txt` on the boot partition (on the target in `/mnt/boot`, or the first partition of the disk) and add `fsck.repair=yes` and then reboot?

MrAlfabet, 2021-06-21:
> Hm, if you started Home Assistant OS completely, then there are quite some services which use that partition.
>
> I think we can ask systemd to fix the file system. Can you edit `cmdline.txt` on the [...]

Check passed, now I just get the growfs FAILED error on boot (like I did in the beginning)

agners, 2021-06-21:
Hm, there are still errors:

```
[    7.335143] EXT4-fs warning (device sda8): ext4_resize_begin:72: There are errors in the filesystem, so online resizing is not allowed
```

When it gets stuck during boot up, and you use "enter" instead of Ctrl+D, you should get a root prompt where a manual fsck with the following commands should be possible:

```
umount -A /dev/disk/by-label/hassos-data
fsck.ext4 /dev/disk/by-label/hassos-data
```

From a running system, you get to a similar state using `systemctl isolate rescue.target` and `systemctl stop systemd-journald`.

MrAlfabet, 2021-06-21:
umount: /var/log/journal: target is busy.

Even at the boot hang....

MrAlfabet, 2021-06-21:
Alright, after stopping systemd-journald I could unmount the data volume. Check started and gave me the following things:

Inodes that were part of a corrupted orphan linked list found. (8 fixed)
Deleted inode XXXXXX has zero dtime.
Block bitmap differences: a lot of numbers here
Free blocks count wrong for group #30 (8458, counted 9002).
Same for group #209
Free block count wrong
Inode bitmap differences: more numbers
more lines with free inodes count wrong

rebooted, and everything works fine now!

So, to recap:

```
Press <enter> when the system stops during boot
systemctl stop systemd-journald
umount -A /dev/disk/by-label/hassos-data
fsck.ext4 /dev/disk/by-label/hassos-data
Keep pressing <enter> through all the prompts
```

closed 2021-06-21 by MrAlfabet

agners, 2021-06-21:
Cool, thanks for the update.

> I did get the 'cant fscheck' error again though, even though the cmdline.txt still contains the fsck=yes line.

Did you had the `fsck.repair=yes` (note the `repair`) option set? From what I understand this should be equivalent to run `ext4.fsck -y`, so I am a bit surprised that this did not resolve the issue on its own.

MrAlfabet, 2021-06-21:
> Did you had the `fsck.repair=yes` (note the `repair`) option set? From what I understand this should be equivalent to run `ext4.fsck -y`, so I am a bit surprised that this did not resolve the issue on its own.

Yea, I did have it set. Removed it again now that everything is working.

bgautrea, 2021-07-08:
> So, to recap:
> 
> ```
> Press <enter> when the system stops during boot
> systemctl stop systemd-journald
> umount -A /dev/disk/by-label/hassos-data
> fsck.ext4 /dev/disk/by-label/hassos-data
> Keep pressing <enter> through all the prompts
> reboot
> ```

Thank you @MrAlfabet for this summary... worked for me too!

tvl9ops, 2021-07-24:
Had the same problem, your solution worked for me, too. Thanks!

Panzer23, 2021-08-11:
Had the same problem, your solution worked for me, too. Thanks!

Silverridge, 2021-09-16:
Thanks @MrAlfabet, this resolved my most annoying problem too!

josedeone, 2022-01-06:
> > So, to recap:
> > ```
> > Press <enter> when the system stops during boot
> > systemctl stop systemd-journald
> > umount -A /dev/disk/by-label/hassos-data
> > fsck.ext4 /dev/disk/by-label/hassos-data
> > Keep pressing <enter> through all the prompts
> > reboot
> > ```
> 
> Thank you @MrAlfabet for this summary... worked for me too!

thank you it wokk here aswell

agners, 2023-01-25:
It seems that often `systemd-journald` is immeaditly started after stopping. To disable systemd-journald entirly to successfully unmount, use this command before stopping `systemd-journald`:

```
systemctl disable --now systemd-journald-audit.socket systemd-journald-dev-log.socket systemd-journald.socket
```

So:

```
Press <enter> when the system stops during boot
systemctl disable --now systemd-journald-audit.socket systemd-journald-dev-log.socket systemd-journald.socket
systemctl stop systemd-journald.service 
umount -A /dev/disk/by-label/hassos-data
fsck.ext4 /dev/disk/by-label/hassos-data
Keep pressing <enter> through all the prompts
reboot
```

MrAlfabet, 2023-01-25:
@agners two remarks: the `disable --now` already performs a `stop` of the service, and if you fully disable these services, you'll have to enable them afterwards (which you're not doing for some reason).

agners, 2023-01-25:
> @agners two remarks: the disable --now already performs a stop of the service

Hm, are you sure this is the case? I only list `.socket` in that command.

> you'll have to enable them afterwards

HAOS does not have a writeable /etc, so disable will only be a runtime disable. But yeah, `--now` just stops them immediately, probably `systemctl stop` is what we want actually. I've updated the comment.

Fonsos, 2023-07-21:
I'm having this issue after update to Home Assistant OS 10.4, but after running 

> umount -A /dev/disk/by-label/hassos-data

I got this:
[image]

So I can't continue.

Is there any way I can get the system running or should I get a new VM created and use yesterday's backup?

Thanks for your help!

agners, 2023-07-21:
It would be interesting what is the culprit that the Grow service fails. Do you have EXT4 warnings as well in dmesg? (you can also check `journalctl -k -b 0`, or `journalctl -k -b -1` for the previous boot).

Fonsos, 2023-07-21:
This are the last lines of the log:
[image]

mykm203, 2023-07-29:
I also have the same problem with rescue mode.
After
Press <enter> when the system stops during boot
systemctl stop systemd-journald-audit.socket systemd-journald-dev-log.socket systemd-journald.socket systemd-journald.service 
umount -A /dev/disk/by-label/hassos-data
Sill have /mnt/data/ - target is busy
Could you help with restoring HA.

 HA Last version

costa-alex, 2023-07-30:
> I also have the same problem with rescue mode. After Press when the system stops during boot systemctl stop systemd-journald-audit.socket systemd-journald-dev-log.socket systemd-journald.socket systemd-journald.service umount -A /dev/disk/by-label/hassos-data Still have /mnt/data/ - target is busy Looks like systemd-journald immediately restarted. Could you help with restoring HA.
> 
> HA Last version

Having the same issue. Even after the systemctl stop command, the umount always returns target is busy.

Any help is appreciated. Thanks

agners, 2023-07-31:
Since HAOS 10.x swap space is active on the same partition. Besides stopping all the service, use `swapoff -a` as well.

mykm203, 2023-07-31:
Thanks a lot. It's unmounted, but stillnot fixed. Given error:
"Couldn't fix parent of inode 137432: Couldn't find parent directory entry"

In previous msg stated NULL inode, should be /lost+found

agners, 2023-07-31:
This sounds like a corrupted file system, unfortunately :cry: I'd recommend to set up a new system, potentially replace hardware if you suspect hardware failure.

mykm203, 2023-07-31:
Yes.it's corrupted file system, it was VirtualBox disk. Really I need obly to extract last full backup. Can you assist how to do it?

agners, 2023-07-31:
This really gets into data recovery. It is also possible that your backups are not recoverable anymore. But this issue tracker is not really the place to discuss these kinds of recovery steps.

I'd suggest to search for VirtualBox data recovery how-to's. Once you have access to the disk/data, the partition labeled `hassos-data` contains the backups. They are stored in the `supervisor` -> `backups` directory. The name is part of the metadata inside the backups so its a bit hard to pick the last. Probably best is to sort by date and pick the last large backup (which presumably is a full backup).

piotrek204, 2024-01-21:
Hi, 
I have a very similar issue, but in my case it's `executable/dev/bin/plymouth is missing, skipping: No such file or directory.`
I can do "unmounted" as described above but after reboot still the same situation.

remarkov, 2024-02-21:
Hi @piotrek204 , not sure if you managed to fix your issue, but I just stumbled across the similar issue with my HA setup after a double power failure and had to spend a couple of hours before I figured out how to fix it.
First of all, I also had this `executable/dev/bin/plymouth is missing, skipping: No such file or directory.` as the last entry in the journal but that's not the reason for the failure to boot.
During many reboots I did I notice that there's another error that quickly shows up during the boot. It was saying 
```
Failed to start Docker Application Container Engine.
See 'systemctl status docker.service' for details.
```
That gave me the clue that something is wrong with the docker service. I tried to start the service manually and got immediately rebooted. After some more searching I found [this](https://techoverflow.net/2022/01/05/how-i-fixed-docker-panic-assertion-failed-write-circular-dependency-occurred/) article and decided to try the trick with `local-kv.db`. Apparently in HA the file location is different so I ended up with just 
```
# mv /mnt/data/docker/network/files/local-kv.db /tmp/old-local-kv.db
# reboot
```
And the next boot HA loaded properly without any errors and all my config was present, no issues, no backup restore needed.
Even if you already managed to fix your issue I am leaving my solution here to save time for someone else in the future.

sylwinc, 2024-07-12:
> Hi @piotrek204 , not sure if you managed to fix your issue, but I just stumbled across the similar issue with my HA setup after a double power failure and had to spend a couple of hours before I [...]

Thank you very much for your post, it worked for me as well!!! I saved few hours to find the solution.

mjaverto, 2024-09-04:
For me, I didn't have to do any of this. I was using proxmox and grub was defaulting to a rescue shell. Maybe a power loss caused this, but just selecting the first shell and I'm running fine even between reboots now.

[image]