home-assistant/operating-system #1427: Add support for non-on-board WiFi/Bluetooth

agners, 2021-06-22:
**Hardware Environment**

- [x] Raspberry Pi [3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Generic x86-64 (like Intel NUC)
- [ ] OVA (Open Virtualization Appliance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [ ] Fresh installation of release x.y
- [ ] Updated from version x.y
- Additional information (if accessible):

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**

The on-board WiFi/Bluetooth has sometimes its downsides like limited range. Support external WiFi/Bluetooth adapters.

kamtschatka, 2021-06-23:
Is there a way to manually add the libraries? Right now when I store it in the correct folder and restart, then the folder gets wiped automatically.
I am using a raspberry pi 3

agners, 2021-06-23:
Currenlty there is no way to manually add the firmware. We use a read-only file system for root on purpose.

Takaszi, 2021-07-20:
My FSC-BP119 CSR8510 doesn't work either, zero readings unlike the sluggish RaspberryPi4 board.

agners, 2021-07-20:
@Takaszi this is a different model and likely needs different firmwares etc. Can you open a new issue along with dmesg (Host logs) from when pluggin the device in?

nepozs, 2021-07-20:
@agners 
Subject here is "Add support for non-on-board WiFi/Bluetooth" so it looks like proper place.

I can provide log (from amd64 installation), it is probably the same chipset as @Takaszi has
```
[678593.000816] usb 1-1.3: new full-speed USB device number 5 using ehci-pci
[678593.093816] usb 1-1.3: New USB device found, idVendor=0a12, idProduct=0001, bcdDevice=88.91
[678593.093830] usb 1-1.3: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[678593.093836] usb 1-1.3: Product: CSR8510 A10
```

agners, 2021-07-27:
@nepozs the device should be supported by `CONFIG_BT_HCIBTUSB` and not require a firmware from what I understand.

Which device are you on? What is `lsusb` showing on Home Assistant OS? Is `hciconfig dev` showing something?

agners, 2021-07-29:
Additional non-on-board devices have been enabled in HAOS 6.2.

@nepozs Please open a new issue with your system details for the CSR8510 if it still doesn't show up. The driver really should be enabled and from what I understand there is no firmware required for this device.

closed 2021-07-29 by agners

nepozs, 2021-07-29:
@agners 
In my installation (amd64) it just works
```
~ $ lsusb 
Bus 001 Device 001: ID 1d6b:0002
Bus 001 Device 004: ID 0a12:0001
Bus 001 Device 002: ID 1a40:0201
Bus 002 Device 001: ID 1d6b:0001
Bus 002 Device 003: ID 0451:3410
```
It is native x64 HAOS installation so form HA console `hciconfig dev` is unavailable.