home-assistant/operating-system #1428: Error: "system is not ready with state: setup"

oneseventhree, 2021-06-22:
**Hardware Environment**

- [ 3] Raspberry Pi [1/2/3/4]

**Home Assistant OS release:**
- [z] Fresh installation of release x.y
- Additional information (if accessible):

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**

Fresh install of HA on Rasberry Pi 3 and I get this error:
[image]

I have installed both 32bit and 64bit versions on this device before and had them working for months so I am not sure why these installs won't work:
1) https://github.com/home-assistant/operating-system/releases/download/6.0/haos_rpi3-64-6.0.img.xz
2) https://github.com/home-assistant/operating-system/releases/download/6.0/haos_rpi3-6.0.img.xz

I can not get into the UI like I used to after a fresh install.

agners, 2021-06-22:
You probably run into timeouts, e.g. the system takes a while to initialize and hence the CLI isn't ready in time. Likely, if you wait longer, you can start the the CLI by running `hassos-cli` in the second case, or show the banner by using the `banner` command in the first case.

If its a timeout, then probably you are using a rather slow SD card. Are you using an A2 (Application Performance Class 2 (A2) card)?

jhchenchong, 2021-06-29:
When I installed the 64-bit system on the Raspberry Pi 4, I had the same error as you, but when I installed the 32-bit system, it worked, did you finally solve the problem?

oneseventhree, 2021-06-29:
> When I installed the 64-bit system on the Raspberry Pi 4, I had the same error as you, but when I installed the 32-bit system, it worked, did you finally solve the problem?

I tried both 64 and 32 multiple times and it didn’t work. Eventually (doing the same steps Ive always done) the 32 version started working. 

It’s been working ever since but I am worried that if I ever have to reformat my HA hard drive I will get this error again.

jhchenchong, 2021-06-29:
> I tried both 64 and 32 multiple times and it didn’t work. Eventually (doing the same steps Ive always done) the 32 version started working.
>
> It’s been working ever since but I am worried that if I [...]

I have installed other versions of the 64-bit system and it works fine, but the latest version has the error, so I don't know why it happened, I will try again at the end of the week.

DivanX10, 2022-11-28:
I have Home Assistant installed on Synology in virtual machine and use an ova image

[image]

Error `Error returned from Supervisor: System is not ready with state: setup`. Periodically appears after the HAOS update, when an update for the supervisor arrives. I demolish the Home Assistant completely, install the new Home Assistant through the ova image and restore it from the backup, and it works fine until the next supervisor update for Home Assistant. It turns out that updating the supervisor somehow affects and causes this error error `Error returned from Supervisor: System is not ready with state: setup`

[image]
[image]

agners, 2022-11-29:
> It turns out that updating the supervisor somehow affects and causes this error error Error returned from Supervisor: System is not ready with state: setup

This typically means that Supervisor takes longer to startup than expected, but it shouldn't be "fatal". What is the problem after that? Is the web interface not working?

You can check the supervisor logs, by running `supervisor logs` in that command prompt.

Since this is Supervisor related, ideally you create a new issue in the Supervisor project https://github.com/home-assistant/supervisor.

CqLaowang, 2023-08-26:
I also followed the official website tutorial to install the Home Assistant OS version on my Raspberry Pi 4B. After the automated installation completed, I connected a monitor and restarted several times, but the same error screen continued to appear. However, what's strange is that I can still access the backend through the web interface and view the dashboard. I've been researching for three days, and I'm still unsure how to resolve this issue.

NullyIsHere, 2024-08-15:
Getting this too, getting a ERR_CONNECTION_REFUSED while trying to access the frontend, tomorrow ill try reinstalling homeassistant as i really need it to remote control things while in vacation.