home-assistant/operating-system #1431: Failed to to call /addons/core_cec_scan/install - 'AddonManager.install' blocked from execution, no host internet connection

ChrisSutton123, 2021-06-23:
**Hardware Environment**

NUC8i3BEH

**Home Assistant OS release:**
- haos_generic-x86-64-6.1.img.xz
- and updated from 5.12
- Additional information (if accessible):

## System Health

version | core-2021.6.6
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
virtualenv | false
python_version | 3.8.9
os_name | Linux
os_version | 5.10.45
arch | x86_64
timezone | Australia/Sydney

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 6.1
-- | --
update_channel | stable
supervisor_version | supervisor-2021.06.3
docker_version | 20.10.6
disk_total | 219.4 GB
disk_used | 5.5 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.3.1), Tor (3.0.5), Node-RED (9.1.4), Frigate NVR (1.13), Home Assistant Google Drive Backup (0.104.3), SSH & Web Terminal (8.2.4), Mosquitto broker (6.0.1), Z-Wave JS to MQTT (0.20.0)

</details>

<details><summary>Lovelace</summary>

dashboards | 1
-- | --
resources | 0
views | 1
mode | storage

</details>
**Supervisor logs:**

21-06-23 23:23:19 INFO (MainThread) [supervisor.snapshots] Found 1 snapshot files
21-06-23 23:23:19 INFO (MainThread) [supervisor.discovery] Loaded 2 messages
21-06-23 23:23:19 INFO (MainThread) [supervisor.ingress] Loaded 2 ingress sessions
21-06-23 23:23:19 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.SETUP
21-06-23 23:23:19 INFO (MainThread) [supervisor.resolution.check] System checks complete
21-06-23 23:23:19 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.SETUP
21-06-23 23:23:19 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
[... 79 lines cut ...]
21-06-23 23:34:28 WARNING (MainThread) [supervisor.host.network] Can't update connectivity information: Error: Timeout was reached
21-06-23 23:35:08 WARNING (MainThread) [supervisor.host.network] Can't update connectivity information: Error: Timeout was reached
21-06-23 23:35:49 WARNING (MainThread) [supervisor.host.network] Can't update connectivity information: Error: Timeout was reached
21-06-23 23:36:29 WARNING (MainThread) [supervisor.host.network] Can't update connectivity information: Error: Timeout was reached
21-06-23 23:37:09 WARNING (MainThread) [supervisor.host.network] Can't update connectivity information: Error: Timeout was reached
21-06-23 23:37:49 WARNING (MainThread) [supervisor.host.network] Can't update connectivity information: Error: Timeout was reached
21-06-23 23:38:29 WARNING (MainThread) [supervisor.host.network] Can't update connectivity information: Error: Timeout was reached
21-06-23 23:39:09 WARNING (MainThread) [supervisor.host.network] Can't update connectivity information: Error: Timeout was reached
21-06-23 23:39:49 WARNING (MainThread) [supervisor.host.network] Can't update connectivity information: Error: Timeout was reached
21-06-23 23:40:29 WARNING (MainThread) [supervisor.host.network] Can't update connectivity information: Error: Timeout was reached
21-06-23 23:41:10 WARNING (MainThread) [supervisor.host.network] Can't update connectivity information: Error: Timeout was reached
**Journal logs:**

**Kernel logs:**

**Description of problem:**

After upgrade to 6.x, I cannot install/upgrade any addons or do any further upgrades to the system.

2021-06-24 09:24:55 ERROR (MainThread) [homeassistant.components.hassio] Failed to to call /addons/core_cec_scan/install - 'AddonManager.install' blocked from execution, no host internet connection

Problem is reproducible. System works fine on OS version 5.12, If I upgrade to 6.x problem occurs
If rebuild system using latest 6.x image I get same problem.

I also have a NUC7JY, with same HA configuration, this system works without problem

agners, 2021-06-24:
It seems that the Supervisor cannot get connectivity information from NetworkManager.

OS 6.0 uses a https less connection to check connectivity (since SSL is problematic in case the time is not correct, which leads to a chicken-egg problem when trying to synchronize time with a NTP server). See: https://github.com/home-assistant/operating-system/pull/1359/files. Do you use maybe a firewall of some sort which prevents this connection?

ChrisSutton123, 2021-06-24:
I also have a NUC7JY, with same HA configuration, this system works without problem

ChrisSutton123, 2021-08-18:
Just tried again to update this NUC with latest releases Home Assistant OS 6.2, and core-2021.8.7.
Problem still occurs. Are there any additional logs I can get to help problem resolution?

agners, 2021-08-18:
> Are there any additional logs I can get to help problem resolution?

You can login on the underlying OS using SSH and try to debug the internet connectivity there (see https://developers.home-assistant.io/docs/operating-system/debugging).

Helpful commands:
```
resolvectl query www.google.com
ping www.google.com
```

Do you use maybe a firewall of some sort which could prevent certain connections?

ChrisSutton123, 2021-08-19:
Results from Helpful commands:-
# resolvectl query www.google.com
www.google.com: 2404:6800:4006:80b::2004       -- link: eno1
                172.217.167.100                -- link: eno1

-- Information acquired via protocol DNS in 17.4ms.
-- Data is authenticated: no

# ping www.google.com
www.google.com is alive!

Some background information:-
I currently have access to two Intel NUC systems
A Celeron based system NUC7JY and a i3 based system NUC8i3BEH
Both systems are connected to the same internet router, both have the same Home Assistant OS 6.2 versions
The NUC7JY works perfectly, the NUC8i3BEH does not.
The only way I can get the NUC8i3BEH to work is to downgrade the Home Assistant OS to version 5.12

nickwild-999, 2021-08-24:
I also have this problem but on a KVM on Ubuntu. Also I cannot restore any backups either

ChrisSutton123, 2021-09-05:
I reinstalled the system using the latest images, problem has "gone away"

closed 2021-09-05 by ChrisSutton123