home-assistant/operating-system #1432: Zwave not updates after update HA OS to 6.1

allmoney, 2021-06-24:
**Hardware Environment**
- [x] Raspberry Pi [1/2/3/4]

**Home Assistant OS release:**
- [x] Updated from version 6.0 to 6.1

## System Health

version | core-2021.6.6
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
virtualenv | false
python_version | 3.8.9
os_name | Linux
os_version | 5.10.17-v7l
arch | armv7l
timezone | Europe/Moscow

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 5000
Installed Version | 1.13.2
Stage | running
Available Repositories | 849
Installed Repositories | 4

</details>

<details><summary>AccuWeather</summary>

can_reach_server | ok
-- | --
remaining_requests | 43

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 6.1
-- | --
update_channel | beta
supervisor_version | supervisor-2021.06.4
docker_version | 20.10.6
disk_total | 57.8 GB
disk_used | 16.4 GB
healthy | true
supported | true
board | rpi4
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.3.1), Duck DNS (1.12.5), Samba share (9.5.1), Backup Hassio to Google Drive (1.7.10), Terminal & SSH (9.1.3), motionEye (0.14.0), Zigbee2mqtt (1.18.1-1), Mosquitto broker (6.0.1)

</details>

<details><summary>Lovelace</summary>
[... 122 lines cut ...]
2021-06-24 11:33:07.621 Info, Node004, Sending (Poll) message (Callback ID=0x1f, Expected Reply=0x04) - SensorMultilevelCmd_Get (Node=4): 0x01, 0x09, 0x00, 0x13, 0x04, 0x02, 0x31, 0x04, 0x25, 0x1f, 0xec
2021-06-24 11:33:08.622 Error, Node004, ERROR: Dropping command, expected response not received after 1 attempt(s)
2021-06-24 11:33:08.623 Detail, Node004, Removing current message
2021-06-24 11:33:08.623 Detail, Node004, Notification: Notification - TimeOut
2021-06-24 11:33:31.150 Info, Node011, Value::Set - COMMAND_CLASS_SWITCH_MULTILEVEL - Dimming Duration - 5 - 1 - 255
2021-06-24 11:33:31.152 Detail, Node011, Refreshed Value: old value=255, new value=255, type=byte
2021-06-24 11:33:31.152 Detail, Node011, Changes to this value are not verified
2021-06-24 11:33:31.153 Detail, Node011, Notification: ValueChanged
2021-06-24 11:33:31.155 Info, Node011, Value::Set - COMMAND_CLASS_SWITCH_MULTILEVEL - Level - 0 - 1 - 255
2021-06-24 11:33:31.155 Info, Node011, SwitchMultilevel::Set - Setting to level 255
2021-06-24 11:33:31.155 Info, Node011,   Duration: Default
2021-06-24 11:33:31.156 Detail, Node011, Queuing (Send) MultiChannel Encapsulated (instance=1): SwitchMultilevelCmd_Set (Node=11): 0x01, 0x0f, 0x00, 0x13, 0x0b, 0x08, 0x60, 0x0d, 0x01, 0x01, 0x26, 0x01, 0xff, 0xff, 0x25, 0x20, 0xaf
2021-06-24 11:33:31.156 Detail, Node011, Queuing (Send) MultiChannel Encapsulated (instance=1): SwitchMultilevelCmd_Get (Node=11): 0x01, 0x0d, 0x00, 0x13, 0x0b, 0x06, 0x60, 0x0d, 0x01, 0x01, 0x26, 0x02, 0x25, 0x21, 0xa1
2021-06-24 11:33:31.170 Detail, 
2021-06-24 11:33:31.170 Info, Node011, Sending (Send) message (Callback ID=0x20, Expected Reply=0x13) - MultiChannel Encapsulated (instance=1): SwitchMultilevelCmd_Set (Node=11): 0x01, 0x0f, 0x00, 0x13, 0x0b, 0x08, 0x60, 0x0d, 0x01, 0x01, 0x26, 0x01, 0xff, 0xff, 0x25, 0x20, 0xaf
2021-06-24 11:33:32.171 Error, Node011, ERROR: Dropping command, expected response not received after 1 attempt(s)
2021-06-24 11:33:32.171 Detail, Node011, Removing current message
2021-06-24 11:33:32.171 Detail, Node011, Notification: Notification - TimeOut
2021-06-24 11:33:32.175 Detail, 
2021-06-24 11:33:32.175 Info, Node011, Sending (Send) message (Callback ID=0x21, Expected Reply=0x04) - MultiChannel Encapsulated (instance=1): SwitchMultilevelCmd_Get (Node=11): 0x01, 0x0d, 0x00, 0x13, 0x0b, 0x06, 0x60, 0x0d, 0x01, 0x01, 0x26, 0x02, 0x25, 0x21, 0xa1
2021-06-24 11:33:33.175 Error, Node011, ERROR: Dropping command, expected response not received after 1 attempt(s)
2021-06-24 11:33:33.176 Detail, Node011, Removing current message
2021-06-24 11:33:33.176 Detail, Node011, Notification: Notification - TimeOut
```
Please, help me fix it!

agners, 2021-06-24:
> It's bug in 6.1 HA OS? How fix?

I don't think this is related to OS 6.1. Can you try to roll back to 6.0 and see if that indeed does not show the problem?

Use the following command in the Web based terminal:

```
ha os update --version 6.0
```

allmoney, 2021-06-24:
[image]

Done, rollbacked to 6.0.
All is OK now with 6.0 - zwave devices ON/OFF from UI now without problems! How can help to find and fix a bug in 6.1?

agners, 2021-07-08:
Can you go back to 6.1 again, and see if it still happens? Also maybe reboot or power cycle the device, in case its some kind of state the Z-Wave is hanging in.

If it is still reproducible, also post the log from the Log Provider "Host" (dmesg).

allmoney, 2021-07-08:
Updated to core-2021.7.0 (with Home Assistant OS 6.1) - all is OK!
Updated to Home Assistant OS 6.1 - now all is OK!
I Will test some days...