home-assistant/operating-system #1433: Can't SSH into the Host System

gmlupatelli, 2021-06-24:
**Hardware Environment**
- Raspberry Pi 4

**Home Assistant OS release:**
- Updated from version 5.13
- Additional information (if accessible):
## System Health

version | core-2021.6.6
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
virtualenv | false
python_version | 3.8.9
os_name | Linux
os_version | 5.10.17-v8
arch | aarch64
timezone | America/Toronto

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 5000
Installed Version | 1.13.2
Stage | running
Available Repositories | 847
Installed Repositories | 7

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | July 9, 2021, 8:00 PM
relayer_connected | true
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | true
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 6.1
-- | --
update_channel | stable
supervisor_version | supervisor-2021.06.3
docker_version | 20.10.6
disk_total | 58.0 GB
disk_used | 39.1 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.3.1), Terminal & SSH (9.1.3), Mosquitto broker (6.0.1), Samba share (9.5.1), Network UPS Tools (0.7.0), Z-Wave JS (0.1.26)

</details>

<details><summary>Lovelace</summary>

dashboards | 1
-- | --
resources | 5
views | 2
mode | storage

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

**Supervisor logs:**
<details><summary>Expand</summary>**

> 21-06-24 12:15:22 INFO (MainThread) [supervisor.host.info] Updating local host information
> 21-06-24 12:15:23 INFO (MainThread) [supervisor.host.services] Updating service information
> 21-06-24 12:15:23 INFO (MainThread) [supervisor.host.network] Updating local network information
> 21-06-24 12:15:25 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
> 21-06-24 12:15:25 INFO (MainThread) [supervisor.host] Host information reload completed
> 21-06-24 12:45:13 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
> 21-06-24 13:13:03 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.RUNNING
> 21-06-24 13:13:03 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.SECURITY/ContextType.CORE
> 21-06-24 13:13:03 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.PWNED/ContextType.ADDON
> 21-06-24 13:13:03 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.FREE_SPACE/ContextType.SYSTEM
> 21-06-24 13:13:03 INFO (MainThread) [supervisor.resolution.check] System checks complete
> 21-06-24 13:13:03 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.RUNNING
> 21-06-24 13:13:03 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
> 21-06-24 13:13:03 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state CoreState.RUNNING
> 21-06-24 13:13:03 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
> 21-06-24 13:14:55 INFO (MainThread) [supervisor.snapshots] Found 1 snapshot files
> 21-06-24 13:14:56 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
> 21-06-24 13:15:13 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
> 21-06-24 13:45:14 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
> 21-06-24 14:13:03 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.RUNNING
> 21-06-24 14:13:03 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.SECURITY/ContextType.CORE
> 21-06-24 14:13:03 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.PWNED/ContextType.ADDON
> 21-06-24 14:13:03 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.FREE_SPACE/ContextType.SYSTEM
> 21-06-24 14:13:03 INFO (MainThread) [supervisor.resolution.check] System checks complete
> 21-06-24 14:13:03 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.RUNNING
> 21-06-24 14:13:04 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
> 21-06-24 14:13:04 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state CoreState.RUNNING
> 21-06-24 14:13:04 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
> 21-06-24 14:15:10 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/home-assistant/addons repository
> 21-06-24 14:15:10 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/hassio-addons/repository repository
> 21-06-24 14:15:13 INFO (MainThread) [supervisor.store] Loading add-ons from store: 63 all - 0 new - 0 remove
> 21-06-24 14:15:13 INFO (MainThread) [supervisor.store] Loading add-ons from store: 63 all - 0 new - 0 remove
> 21-06-24 14:15:14 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
> 21-06-24 14:22:05 INFO (MainThread) [supervisor.host.info] Updating local host information
> 21-06-24 14:22:06 INFO (MainThread) [supervisor.host.services] Updating service information
> 21-06-24 14:22:06 INFO (MainThread) [supervisor.host.network] Updating local network information
> 21-06-24 14:22:09 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
> 21-06-24 14:22:09 INFO (MainThread) [supervisor.host] Host information reload completed
> 21-06-24 14:45:15 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
> 21-06-24 15:13:04 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.RUNNING
> 21-06-24 15:13:04 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.SECURITY/ContextType.CORE
> 21-06-24 15:13:04 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.PWNED/ContextType.ADDON
> 21-06-24 15:13:04 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.FREE_SPACE/ContextType.SYSTEM
> 21-06-24 15:13:04 INFO (MainThread) [supervisor.resolution.check] System checks complete
> 21-06-24 15:13:04 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.RUNNING
> 21-06-24 15:13:04 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
> 21-06-24 15:13:04 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state CoreState.RUNNING
> 21-06-24 15:13:04 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
> 21-06-24 15:14:56 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
> 21-06-24 15:15:15 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
> 21-06-24 15:45:15 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
> 21-06-24 15:49:46 INFO (MainThread) [supervisor.api.middleware.security] /network/info access from core_ssh
> 21-06-24 15:49:46 INFO (MainThread) [supervisor.api.middleware.security] /network/info access from core_ssh
> 21-06-24 15:49:46 INFO (MainThread) [supervisor.api.middleware.security] /host/info access from core_ssh
> 21-06-24 15:49:46 INFO (MainThread) [supervisor.api.middleware.security] /core/info access from core_ssh
> 21-06-24 15:49:55 INFO (MainThread) [supervisor.api.middleware.security] /supervisor/logs access from core_ssh
> 21-06-24 15:50:21 INFO (MainThread) [supervisor.api.middleware.security] /supervisor/logs access from core_ssh

</details>

**Journal logs:**
Cant SSH on OS level

**Kernel logs:**
Cant SSH on OS level

**Description of problem:**

**Current behavior:**
After establishing an SSH connection to port 22222 and having access to the CLI environment I am unable to login into the OS by typing login.
After typing the login command my connection is closed by the host.
[image]

**Expected behavior:**
After typing login I would expect to be logged in the OS level as root.

gmlupatelli, 2021-06-24:
I also have a forum post about that: https://community.home-assistant.io/t/cant-log-in-after-ssh-to-the-the-host/317656

ludeeus, 2021-06-24:
You are already connected to the OS, don't write `login`

gmlupatelli, 2021-06-24:
> You are already connected to the OS, don't write `login`

Does this change on version 6.0? The [Debugging documentation](https://developers.home-assistant.io/docs/operating-system/debugging/) page states that I need to login, and I might be crazy, but I remember not being able to do anything other than CLI commands without logging in.

> You will initially be able to perform normal CLI tasks. If you need access to the host system use the 'login' command. Home Assistant OS is a hypervisor for Docker. See the Supervisor Architecture documentation for information regarding the Supervisor. The Supervisor offers an API to manage the host and running the Docker containers. Home Assistant itself and all installed add-ons run in separate Docker containers.

agners, 2021-10-06:
@gmlupatelli yes, this changed with release 6 with PR #1366. I've seen that you have updated the documentation, thanks!

closed 2021-10-06 by agners