home-assistant/operating-system #1434: CoreDNS and observer constantly crashing breaking all internal DNS resolving

vabene1111, 2021-06-27:
**Hardware Environment**
- [x] Raspberry Pi 4

**Home Assistant OS release:**
- [x] Updated from version (dont know the number but october 2020, updated to any version in between, currently 2021.05.4)
## System Health

version | core-2021.5.0
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
virtualenv | false
python_version | 3.8.9
os_name | Linux
os_version | 5.10.17-v8
arch | aarch64
timezone | Europe/Berlin

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | failed to load: unreachable
can_reach_cloud_auth | failed to load: unreachable
can_reach_cloud | failed to load: unreachable

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 6.0
-- | --
update_channel | stable
supervisor_version | supervisor-2021.05.4
docker_version | 20.10.6
disk_total | 57.8 GB
disk_used | 10.5 GB
healthy | failed to load: Unhealthy
supported | true
board | rpi4-64
supervisor_api | ok
version_api | failed to load: unreachable
installed_addons | Let's Encrypt (4.11.0), AdGuard Home (4.0.0), File editor (5.3.0), AppDaemon 4 (0.6.0), Grafana (6.3.3), InfluxDB (4.0.6), Samba share (9.3.1), SSH & Web Terminal (8.2.1)

</details>

<details><summary>Lovelace</summary>

dashboards | 3
-- | --
resources | 0
views | 4
mode | storage

</details>
**Supervisor logs:**
```
21-06-27 11:10:32 WARNING (MainThread) [supervisor.misc.tasks] Watchdog found a problem with CoreDNS plugin!
21-06-27 11:10:33 INFO (MainThread) [supervisor.plugins.dns] Starting CoreDNS plugin
21-06-27 11:10:33 INFO (SyncWorker_3) [supervisor.docker.interface] Cleaning hassio_dns application
21-06-27 11:10:35 INFO (SyncWorker_3) [supervisor.docker.dns] Starting DNS homeassistant/aarch64-hassio-dns with version 2021.06.0 - 172.30.32.3
21-06-27 11:11:05 WARNING (MainThread) [supervisor.misc.tasks] Watchdog found a problem with CoreDNS plugin!
21-06-27 11:11:05 INFO (MainThread) [supervisor.plugins.dns] Starting CoreDNS plugin
21-06-27 11:11:05 INFO (SyncWorker_5) [supervisor.docker.interface] Cleaning hassio_dns application
21-06-27 11:11:08 INFO (SyncWorker_5) [supervisor.docker.dns] Starting DNS homeassistant/aarch64-hassio-dns with version 2021.06.0 - 172.30.32.3
21-06-27 11:11:38 WARNING (MainThread) [supervisor.misc.tasks] Watchdog found a problem with CoreDNS plugin!
21-06-27 11:11:38 INFO (MainThread) [supervisor.plugins.dns] Starting CoreDNS plugin
21-06-27 11:11:38 INFO (SyncWorker_6) [supervisor.docker.interface] Cleaning hassio_dns application
[... 1415 lines cut ...]
2021-06-27 12:26:15 ERROR (SyncWorker_3) [homeassistant.components.influxdb] Cannot connect to InfluxDB due to 'HTTPConnectionPool(host='a0d7b954-influxdb', port=8086): Max retries exceeded with url: /write?db=homeassistant (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7fa0f782e0>: Failed to establish a new connection: [Errno -3] Try again'))'. Please check that the provided connection details (host, port, etc.) are correct and that your InfluxDB server is running and accessible. Retrying in 60 seconds.
2021-06-27 12:27:30 ERROR (SyncWorker_2) [homeassistant.components.influxdb] Cannot connect to InfluxDB due to 'HTTPConnectionPool(host='a0d7b954-influxdb', port=8086): Max retries exceeded with url: /write?db=homeassistant (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7fa13bc700>: Failed to establish a new connection: [Errno -3] Try again'))'. Please check that the provided connection details (host, port, etc.) are correct and that your InfluxDB server is running and accessible. Retrying in 60 seconds.
2021-06-27 12:28:45 ERROR (SyncWorker_3) [homeassistant.components.influxdb] Cannot connect to InfluxDB due to 'HTTPConnectionPool(host='a0d7b954-influxdb', port=8086): Max retries exceeded with url: /write?db=homeassistant (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7fa12d2610>: Failed to establish a new connection: [Errno -3] Try again'))'. Please check that the provided connection details (host, port, etc.) are correct and that your InfluxDB server is running and accessible. Retrying in 60 seconds.
2021-06-27 12:30:00 ERROR (SyncWorker_1) [homeassistant.components.influxdb] Cannot connect to InfluxDB due to 'HTTPConnectionPool(host='a0d7b954-influxdb', port=8086): Max retries exceeded with url: /write?db=homeassistant (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7fa0f65ee0>: Failed to establish a new connection: [Errno -3] Try again'))'. Please check that the provided connection details (host, port, etc.) are correct and that your InfluxDB server is running and accessible. Retrying in 60 seconds.
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] udev.sh: executing... 
[cont-init.d] udev.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
2021-06-27 12:38:49 WARNING (MainThread) [homeassistant.components.webhook] Received message for unregistered webhook c80bd735e2eeed5c520314350b8ae0eb0ef4d72a7f9da412806116d5dbe1d269 from 192.168.178.23
2021-06-27 12:38:49 WARNING (MainThread) [homeassistant.components.webhook] Received message for unregistered webhook c80bd735e2eeed5c520314350b8ae0eb0ef4d72a7f9da412806116d5dbe1d269 from 192.168.178.23
2021-06-27 12:38:49 WARNING (MainThread) [homeassistant.components.webhook] Received message for unregistered webhook c80bd735e2eeed5c520314350b8ae0eb0ef4d72a7f9da412806116d5dbe1d269 from 192.168.178.23
2021-06-27 12:38:53 WARNING (MainThread) [homeassistant.setup] Setup of zone is taking over 10 seconds.
2021-06-27 12:38:55 ERROR (MainThread) [homeassistant.components.updater] Error requesting Home Assistant update data: Cannot connect to host www.home-assistant.io:443 ssl:default [Try again]
2021-06-27 12:38:55 ERROR (MainThread) [metno] Access to https://aa015h6buqvih86i1.api.met.no/weatherapi/locationforecast/2.0/complete returned error 'ClientConnectorError'
2021-06-27 12:38:58 WARNING (MainThread) [zigpy.zcl] Unknown cluster 61184
2021-06-27 12:38:59 WARNING (MainThread) [homeassistant.setup] Setup of influxdb is taking over 10 seconds.
```

Tanks in advance for any help!

agners, 2021-07-07:
Ugh, that looks like your DNS plug-in add-on is corrupted somehow.

I assume you are running from a SD card?

vabene1111, 2021-07-08:
yes i am, any way to "uncorrupt" the dns plugin ?

vabene1111, 2021-10-06:
I fixed this by reinstalling home assistant