home-assistant/operating-system #1435: On fresh 6.1 installation, if reboot system, it crashes

Rockman18, 2021-06-28:
**Hardware Environment**

- [ ] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [x] Generic x86-64 (like Intel NUC)
- [ ] OVA (Open Virtualization Appliance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [x] Fresh installation of release 6 build 1
- [ ] Updated from version x.y
- Additional information (if accessible):

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**
After a fresh installation of the latest release 6 build 1 on a Intel NUC like system (lenovo M700 tiny, Intel G4400T, 4GB of RAM, 500GB HDD). I use balena etcher to burn haos_generic-x86-64-6.1.img.xz on the HDD, then boot for the first time without issue.
After minimal configuration, I try to restart the system but boot process crash with following issue.

[image]
[image]

I try to reboot several times but nothing works, so i reburn the img OS and retry but it's exactly the same.

agners, 2021-07-07:
Hm, what type of harddrive is this?

It seems the bootloader really has trouble writing or reading from it, as it reports I/O error when trying to save the boot state as well as trying to access the kernel partition.

Rockman18, 2021-07-20:
It was a 500GB Seagate Laptop Thin HDD without any issue by the past. I've replaced it with a PNY SSD and every thing's ok now.

agners, 2021-07-20:
> It was a 500GB Seagate Laptop Thin HDD without any issue by the past. I've replaced it with a PNY SSD and every thing's ok now.

Thanks for the update! I guess an hardware issue then. Just had a HDD fail on my this week as well :( Unrelated to HAOS though.

Closing as I'd consider it as a hardware issue.

closed 2021-07-20 by Rockman18