home-assistant/operating-system #1441: WLAN is broken/unstable in 6.x, works in 5.x

rfvermut, 2021-07-04:
**Hardware Environment**

- [x] Raspberry Pi4 (Raspberry Pi 4 Model B Rev 1.2)

**Home Assistant OS release:**
- [x] Fresh installation of release 6.0/6.1
- [x] Updated from version 5.13
- Additional information (if accessible):
## System Health

version | core-2021.6.6
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
virtualenv | false
python_version | 3.8.9
os_name | Linux
os_version | 5.4.83-v8
arch | aarch64
timezone | Europe/Riga

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 3845
Installed Version | 1.13.2
Stage | running
Available Repositories | 848
Installed Repositories | 9

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 5.13
-- | --
update_channel | stable
supervisor_version | supervisor-2021.06.6
docker_version | 19.03.15
disk_total | 27.8 GB
disk_used | 5.9 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.3.1), Terminal & SSH (9.1.3), Mosquitto broker (6.0.1), Duck DNS (1.12.5), NGINX Home Assistant SSL proxy (3.0.1), Zigbee2mqtt (1.18.1-1), Nextcloud (20.0.3), Nextcloud (0.7), motionEye (0.14.0), Frigate NVR (1.13)

</details>

<details><summary>Lovelace</summary>

dashboards | 2
-- | --
resources | 1
views | 1
mode | storage

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

**Journal logs:**
```
Jul 04 09:39:50 homeassistant NetworkManager[337]: <warn>  [1625391590.6917] device (wlan0): Activation: (wifi) association took too long, failing activation
Jul 04 09:39:50 homeassistant NetworkManager[337]: <info>  [1625391590.6919] device (wlan0): state change: config -> failed (reason 'ssid-not-found', sys-iface-state: 'managed')
Jul 04 09:39:50 homeassistant NetworkManager[337]: <warn>  [1625391590.6973] device (wlan0): Activation: failed for connection 'Supervisor wlan0'
Jul 04 09:39:50 homeassistant NetworkManager[337]: <info>  [1625391590.6986] device (wlan0): state change: failed -> disconnected (reason 'none', sys-iface-state: 'managed')
Jul 04 09:39:50 homeassistant NetworkManager[337]: <warn>  [1625391590.7060] device (wlan0): wifi-scan: active scanning for networks due to profiles with wifi.hidden=yes. This makes you trackable
Jul 04 09:39:50 homeassistant NetworkManager[337]: <info>  [1625391590.7070] device (wlan0): supplicant interface state: scanning -> disconnected
Jul 04 09:40:40 homeassistant NetworkManager[337]: <info>  [1625391640.3376] device (wlan0): Activation: starting connection 'Supervisor wlan0' (b98db188-1d18-4b72-92bd-7cf782913de9)
Jul 04 09:40:40 homeassistant NetworkManager[337]: <info>  [1625391640.3385] device (wlan0): state change: disconnected -> prepare (reason 'none', sys-iface-state: 'managed')
Jul 04 09:40:40 homeassistant NetworkManager[337]: <info>  [1625391640.3425] device (wlan0): state change: prepare -> config (reason 'none', sys-iface-state: 'managed')
Jul 04 09:40:40 homeassistant NetworkManager[337]: <info>  [1625391640.3450] device (wlan0): Activation: (wifi) access point 'Supervisor wlan0' has security, but secrets are required.
Jul 04 09:40:40 homeassistant NetworkManager[337]: <info>  [1625391640.3451] device (wlan0): state change: config -> need-auth (reason 'none', sys-iface-state: 'managed')
Jul 04 09:40:40 homeassistant NetworkManager[337]: <info>  [1625391640.3555] device (wlan0): state change: need-auth -> prepare (reason 'none', sys-iface-state: 'managed')
Jul 04 09:40:40 homeassistant NetworkManager[337]: <info>  [1625391640.3567] device (wlan0): state change: prepare -> config (reason 'none', sys-iface-state: 'managed')
Jul 04 09:40:40 homeassistant NetworkManager[337]: <info>  [1625391640.3579] device (wlan0): Activation: (wifi) connection 'Supervisor wlan0' has security, and secrets exist.  No new secrets needed.
Jul 04 09:40:40 homeassistant NetworkManager[337]: <info>  [1625391640.3876] device (wlan0): supplicant interface state: disconnected -> scanning
Jul 04 09:41:05 homeassistant NetworkManager[337]: <warn>  [1625391665.6922] device (wlan0): Activation: (wifi) association took too long, failing activation
Jul 04 09:41:05 homeassistant NetworkManager[337]: <info>  [1625391665.6924] device (wlan0): state change: config -> failed (reason 'ssid-not-found', sys-iface-state: 'managed')
Jul 04 09:41:05 homeassistant NetworkManager[337]: <warn>  [1625391665.6982] device (wlan0): Activation: failed for connection 'Supervisor wlan0'
Jul 04 09:41:05 homeassistant NetworkManager[337]: <info>  [1625391665.6997] device (wlan0): state change: failed -> disconnected (reason 'none', sys-iface-state: 'managed')
Jul 04 09:41:05 homeassistant NetworkManager[337]: <info>  [1625391665.7085] device (wlan0): supplicant interface state: scanning -> disconnected
Jul 04 09:41:55 homeassistant NetworkManager[337]: <info>  [1625391715.0196] device (wlan0): Activation: starting connection 'Supervisor wlan0' (b98db188-1d18-4b72-92bd-7cf782913de9)
Jul 04 09:41:55 homeassistant NetworkManager[337]: <info>  [1625391715.0211] device (wlan0): state change: disconnected -> prepare (reason 'none', sys-iface-state: 'managed')
Jul 04 09:41:55 homeassistant NetworkManager[337]: <info>  [1625391715.0281] device (wlan0): state change: prepare -> config (reason 'none', sys-iface-state: 'managed')
Jul 04 09:41:55 homeassistant NetworkManager[337]: <info>  [1625391715.0307] device (wlan0): Activation: (wifi) access point 'Supervisor wlan0' has security, but secrets are required.
Jul 04 09:41:55 homeassistant NetworkManager[337]: <info>  [1625391715.0308] device (wlan0): state change: config -> need-auth (reason 'none', sys-iface-state: 'managed')
Jul 04 09:41:55 homeassistant NetworkManager[337]: <info>  [1625391715.0419] device (wlan0): state change: need-auth -> prepare (reason 'none', sys-iface-state: 'managed')
Jul 04 09:41:55 homeassistant NetworkManager[337]: <info>  [1625391715.0429] device (wlan0): state change: prepare -> config (reason 'none', sys-iface-state: 'managed')
Jul 04 09:41:55 homeassistant NetworkManager[337]: <info>  [1625391715.0443] device (wlan0): Activation: (wifi) connection 'Supervisor wlan0' has security, and secrets exist.  No new secrets needed.
Jul 04 09:41:55 homeassistant NetworkManager[337]: <info>  [1625391715.0845] device (wlan0): supplicant interface state: disconnected -> scanning
```

**Description of problem:**

After upgrade to 6.1 system became disconnected from wlan. When rebooting it connects for few minutes being available and them loses connection again.
Downgrading to 6.0 didn't help.

Booted from latest raspbian (swapped SD card) - worked flawlessly.

Tried to debug 6.x to my best ability - situation looks as follows:
- all of `nm device show`, wavemon` and `UI -> Supervisor -> Host -> IP -> Change` they give me a some list of SSIDs around, but not *MY* SSID
- if running wavemon long enough, eventually my SSID appears for short period. All fine and perfect, 100% signal quality as it's supposed to be
- trying to connect to it with NM bring back the error where NM cannot find the SSID (probably it's disappeared from radar again)

Downgrading to 5.13 fixed it immediately. Nothing crazy about AP too - Mikrotik, 2.4Ghz, Ch. 9, been working w/o issues forever.

agners, 2021-07-07:
> After upgrade to 6.1 system became disconnected from wlan. When rebooting it connects for few minutes being available and them loses connection again.

So it works after reboot, but after a while it doesn't work until the next reboot?

Can you post the kernel log taken after the disconnect happens (Log Provider "Host" in the Supervisor section)?

rfvermut, 2021-07-07:
ok I will in a few days. need to find an open slot.

agners, 2021-07-28:
With #1485 HAOS 6.2 will have the latest WiFi firmwares.  There have been some changes to the firmware: https://github.com/LibreELEC/brcmfmac_sdio-firmware-rpi/compare/9c0ffe9a7f0753b36ed88f7981905a989d940ea9...883b72628de1d7efa45b421da0cbf175ac2374f8

This change in particular could explain your issue if you are using a higher 5GHz channel:
> Fix higher 5GHz channel and BLE issues on CYW43455

Can you test again once 6.2 is out?

rfvermut, 2021-07-29:
Absolutely. Just mention that I has issues on 2.4GHz.

rfvermut, 2021-08-01:
@agners I collected both the logs from 5.13 boot (working) and 6.1 boot (not working). There are no messages *AFTER* connection is lost, so I'm attaching everything that I have.

[6.1_bad.txt](https://github.com/home-assistant/operating-system/files/6912718/6.1_bad.txt)
[5.13_good.txt](https://github.com/home-assistant/operating-system/files/6912719/5.13_good.txt)

rfvermut, 2021-08-01:
I joined a beta channel and installed 6.2. It looks stable from first 5 minutes of the boot. Possibly  #1485 resolved the  issue.

Attaching log jic. [6.2_good.txt](https://github.com/home-assistant/operating-system/files/6912728/6.2_good.txt)

deviantintegral, 2021-08-03:
I just upgraded to 6.2 on my Raspberry Pi 4, and it's still not able to see any 5GHz networks. It's right beside an access point so it should be able to pick up at least two SSIDs.

```
# ha network scan wlan0 | grep -i frequency
- frequency: 2437
- frequency: 2437
- frequency: 2412
- frequency: 2412
- frequency: 2462
- frequency: 2412
- frequency: 2462
- frequency: 2462
- frequency: 2437
```

AleZinov, 2021-08-23:
Have the same issue as the topic starter: wifi connection flapping after upgrade from 5.13 to 6.x.
Have no logs ATM (actually, they was same as 
aforementioned), but upgrade to 6.2 didn't resolve the issue. However, it brings more stability: before 6.2. connection was lost after 5-10 minutes, now it works for a day and a half.
While I was playing with 6.[0-1] I managed to stabilize wifi for a half day disabling powersafe mode in profile.

P.S. Sorry for my English.

agners, 2021-09-03:
@deviantintegral cannot reproduce here with 6.2 on a Raspberry Pi 4:

```
# ha network scan wlan0 | grep -i frequency
- frequency: 5220
- frequency: 2462
- frequency: 2437
- frequency: 2412
- frequency: 2462
- frequency: 2412
- frequency: 5520
- frequency: 5520
- frequency: 5520
- frequency: 2462
# ha os info
board: rpi4
boot: B
update_available: false
version: "6.2"
version_latest: "6.2"
```

Can the same Hardware connect to 5GHz networks on Raspberry Pi OS? What exact 5GHz channels is your network using?

@AleZinov could it be that your signal quality is not optimal?

deviantintegral, 2021-09-03:
```
$ ha os info
board: rpi4
boot: A
update_available: false
version: "6.2"
version_latest: "6.2"
```

```
$ ha network scan wlan0 | grep -i frequency
- frequency: 2437
- frequency: 2437
- frequency: 2412
- frequency: 2412
- frequency: 2412
- frequency: 2462
- frequency: 2462
```

I have a Unifi AP right beside it broadcasting on channel 153 / VHT80.

agners, 2021-09-03:
Hm, using `iw list` shows that only some frequencies are enabled:

```
        Band 1:
...
                Frequencies:                                                                                                                                                                                                                                                                                                  
                        * 2412 MHz [1] (20.0 dBm)                                                                                                                                                                                                                                                                             
                        * 2417 MHz [2] (20.0 dBm)                                                                                                                                                                                                                                                                             
                        * 2422 MHz [3] (20.0 dBm)                                                                                                                                                                                                                                                                             
                        * 2427 MHz [4] (20.0 dBm)                                                                                                                                                                                                                                                                             
                        * 2432 MHz [5] (20.0 dBm)                                                                                                                                                                                                                                                                             
                        * 2437 MHz [6] (20.0 dBm)                                                                                                                                                                                                                                                                             
                        * 2442 MHz [7] (20.0 dBm)                                                                                                                                                                                                                                                                             
                        * 2447 MHz [8] (20.0 dBm)                                                                                                                                                                                                                                                                             
                        * 2452 MHz [9] (20.0 dBm)                                                                                                                                                                                                                                                                             
                        * 2457 MHz [10] (20.0 dBm)                                                                                                                                                                                                                                                                            
                        * 2462 MHz [11] (20.0 dBm)                                                                                                                                                                                                                                                                            
                        * 2467 MHz [12] (20.0 dBm)                                                                                                                                                                                                                                                                            
                        * 2472 MHz [13] (20.0 dBm)                                                                                                                                                                                                                                                                            
                        * 2484 MHz [14] (disabled)
        Band 2:
....
                Frequencies:
                        * 5170 MHz [34] (disabled)
                        * 5180 MHz [36] (20.0 dBm)
                        * 5190 MHz [38] (disabled)
                        * 5200 MHz [40] (20.0 dBm)
                        * 5210 MHz [42] (disabled)
                        * 5220 MHz [44] (20.0 dBm)
                        * 5230 MHz [46] (disabled)
                        * 5240 MHz [48] (20.0 dBm)
                        * 5260 MHz [52] (20.0 dBm) (no IR, radar detection)
                        * 5280 MHz [56] (20.0 dBm) (no IR, radar detection)
                        * 5300 MHz [60] (20.0 dBm) (no IR, radar detection)
                        * 5320 MHz [64] (20.0 dBm) (no IR, radar detection)
                        * 5500 MHz [100] (20.0 dBm) (no IR, radar detection)
[... cut ...]
        (5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
        (5490 - 5730 @ 160), (N/A, 20), (0 ms), DFS, PASSIVE-SCAN
        (5735 - 5835 @ 80), (N/A, 20), (N/A), PASSIVE-SCAN
        (57240 - 63720 @ 2160), (N/A, 0), (N/A)
```

You can set the default regulatory domain used by the kernel by adding `cfg80211.ieee80211_regdom=<COUNTRY_CODE>` e.g. `cfg80211.ieee80211_regdom=US` to `cmdline.txt` (present in the first partition of your SD card, or in `/mnt/boot/cmdline.txt` through the OS shell).

agners, 2021-09-03:
Since this is a new/different problem I created a separate issue #1530. I am closing this as the original problem has been resolved.

closed 2021-09-03 by agners