home-assistant/operating-system #1442: haos_rpi3-6.1.img.xz and haos_rpi3-6.1.raucb maybe broken

Megachip, 2021-07-04:
**Hardware Environment**

- [x] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Generic x86-64 (like Intel NUC)
- [ ] OVA (Open Virtualization Appliance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [x] Fresh installation of release 6.1
- [ ] Updated from version x.y

**Description of problem:**

System didn't boot at all. Looks like #1440 is related to that issue. 
64bit image works fine.

agners, 2021-07-28:
Just tried to reproduce this on a Raspberry Pi 3 Model B Rev 1.2: I flashed the image onto the SD card and it seems to boot fine. I also tried upgrading from 6.0, it seems to work fine as well.

Are you sure the SD card is fine?

studio-suman, 2021-08-08:
@agners I did try fresh install of 6.2 rpi3 gets stuck at docker image? Pls help here

icouper, 2021-08-11:
Had a similar issue. I was flashing the 64bit 6.2 image for the RPi 3 and the Pi would not boot at all. Grabbed the 6.0 image instead and flashed it and it booted just fine.

This is a brand new SD card (opened today), which also booted fine when I loaded the RPi OS onto it.

studio-suman, 2021-08-30:
So 6.2 is now stands broken and any solution on that?

agners, 2021-09-03:
@studio-suman just tested 32-bit and 64-bit (since you don't specify) on Raspberry Pi 3 here, both work fine: They boot, and after a while (it seems that it takes up to 5 minutes on Rpi3) I can access them via network. Are you sure you are waiting long enough?

What flashing software are you using? Also make sure to properly eject the SD card from your OS (not just pull it out).

I close this issue for now as I really cannot reproduce the issues.

closed 2021-09-03 by agners