home-assistant/operating-system #1443: nmcli does not recognize Wifi dongle "Ralink RT5572"

sim-san, 2021-07-04:
**Hardware Environment**

- [ ] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Generic x86-64 (like Intel NUC)
- [x] OVA (Open Virtualization Appliance, on Intel NUC or any other hardware, please add the Hypervisor you are using)
=> Proxmox VE 6.4-9

**Home Assistant OS release:**
- [x] Fresh installation of release 6.1
- [ ] Updated from version x.y
- Additional information (if accessible):

```
arch: amd64
channel: stable
docker: 20.10.6
features:
- reboot
- shutdown
- services
- network
- hostname
- timedate
- hassos
hassos: "6.1"
homeassistant: 2021.6.6
hostname: homeassistant
logging: info
machine: qemux86-64
operating_system: Home Assistant OS 6.1
state: running
supervisor: 2021.06.6
supported: true
supported_arch:
- amd64
- i386
timezone: Europe/Berlin
```

**Supervisor logs:**

```
21-07-05 00:43:45 INFO (SyncWorker_0) [supervisor.docker.multicast] Starting Multicast homeassistant/amd64-hassio-multicast with version 2021.04.0 - Host
21-07-05 00:43:45 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
21-07-05 00:43:47 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/qemux86-64-homeassistant with version 2021.6.6
21-07-05 00:43:47 INFO (MainThread) [supervisor.homeassistant.secrets] Loaded 13 Home Assistant secrets
21-07-05 00:43:47 INFO (MainThread) [supervisor.hassos] Detect Home Assistant Operating System 6.1 / BootSlot A
21-07-05 00:43:48 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/a0d7b954 repository
21-07-05 00:43:48 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/core repository
21-07-05 00:43:48 INFO (MainThread) [supervisor.store] Loading add-ons from store: 63 all - 63 new - 0 remove
21-07-05 00:43:48 INFO (MainThread) [supervisor.addons] Found 2 installed add-ons
21-07-05 00:43:48 INFO (SyncWorker_2) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/vscode/amd64 with version 3.5.0
21-07-05 00:43:48 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/ssh/amd64 with version 8.2.4
21-07-05 00:43:49 INFO (MainThread) [supervisor.snapshots] Found 0 snapshot files
21-07-05 00:43:49 INFO (MainThread) [supervisor.discovery] Loaded 0 messages
21-07-05 00:43:49 INFO (MainThread) [supervisor.ingress] Loaded 8 ingress sessions
21-07-05 00:43:49 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.SETUP
21-07-05 00:43:49 INFO (MainThread) [supervisor.resolution.check] System checks complete
21-07-05 00:43:49 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.SETUP
21-07-05 00:43:49 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
21-07-05 00:43:49 INFO (MainThread) [supervisor.jobs] 'ResolutionFixup.run_autofix' blocked from execution, system is not running - CoreState.SETUP
21-07-05 00:43:49 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.SETUP
21-07-05 00:43:49 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
21-07-05 00:43:49 INFO (MainThread) [__main__] Running Supervisor
21-07-05 00:43:49 INFO (MainThread) [supervisor.hassos] Rauc: A - marked slot kernel.0 as good
21-07-05 00:43:49 INFO (MainThread) [supervisor.addons] Phase 'AddonStartup.INITIALIZE' starting 0 add-ons
21-07-05 00:43:49 INFO (MainThread) [supervisor.addons] Phase 'AddonStartup.SYSTEM' starting 0 add-ons
21-07-05 00:43:49 INFO (MainThread) [supervisor.addons] Phase 'AddonStartup.SERVICES' starting 2 add-ons
[... 952 lines cut ...]
[   80.355814] audit: type=1334 audit(1625438657.300:142): prog-id=17 op=LOAD
[   80.356226] audit: type=1334 audit(1625438657.301:143): prog-id=18 op=LOAD
[  110.348119] audit: type=1334 audit(1625438687.297:144): prog-id=16 op=UNLOAD
[  110.348130] audit: type=1334 audit(1625438687.297:145): prog-id=15 op=UNLOAD
[  110.638139] audit: type=1334 audit(1625438687.587:146): prog-id=18 op=UNLOAD
[  110.638148] audit: type=1334 audit(1625438687.587:147): prog-id=17 op=UNLOAD
```

**Description of problem:**

I attached a wifi dongle to my Intel NUC and forward the USB device in Proxmox VE to the HA OS. As I see it we do not load firmware for the device.

Wifi Dongle; Ralink Technology, Corp. RT5572 Wireless Adapter
https://linux-hardware.org/index.php?id=usb:148f-5572

```
[    2.642328] usb 1-1: New USB device found, idVendor=148f, idProduct=5572, bcdDevice= 1.01
[    2.644929] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    2.647511] usb 1-1: Product: 802.11 n WLAN
[    2.649279] usb 1-1: Manufacturer: Ralink
[    2.651598] usb 1-1: SerialNumber: 1.0
```

Are the drivers missing ? 
Am I making another mistake ?

agners, 2021-07-07:
Yes, those drivers and firmware are not included in the image currently.

sim-san, 2021-07-08:
Should such drivers be added ? Is there a schedule for this ?

agners, 2021-07-08:
There are tons of drivers which could be added, I usually add them when I see a good reason or it comes up on this issue tracker like in this case.

I'll add them when I come around. Should be a rather small change.

agners, 2021-07-28:
The firmware will be part of OS 6.2 and later.

closed 2021-07-28 by agners