home-assistant/operating-system #1446: Porter addon unable to access Internet on clean installation

joaomlneto, 2021-07-06:
#### Description of problem:
On a clean installation of Home Assistant the "Porter" addon is not able to access the Internet.

This is a re-report of issue https://github.com/hassio-addons/addon-portainer/issues/115 which has been rudely closed by @frenck without waiting for more feedback nor being more clear on where/how to report it.

1. Is the problem reproducible? Yes
2. Has this been working before? No
3. Has there been an attempt to rule out hardware issues? Yes. Connectivity is fine other than Porter addon.

#### Steps to reproduce:
1. Install Home Assistant OS 6.1
2. Perform initial setup
4. Install and configure Portainer
    1. Disable protection mode
    2. Navigate to Extensions page on the addon interface
    3. Observe the error `Portainer must be connected to the Internet to fetch the list of available extensions.`

#### Hardware Environment
Raspberry Pi 4 (8GB)

#### Home Assistant OS release:
Happens on:
- Fresh installation of 6.1
- Installation of 5.13 and upgrade to 6.1

<details>
  <summary>HA info</summary>

```
arch: aarch64
channel: stable
docker: 20.10.6
features:
- reboot
- shutdown
- services
- network
- hostname
- timedate
- hassos
hassos: "6.1"
homeassistant: 2021.6.6
hostname: homeassistant
logging: info
machine: raspberrypi4-64
operating_system: Home Assistant OS 6.1
state: running
supervisor: 2021.06.6
supported: true
supported_arch:
- aarch64
- armv7
- armhf
timezone: Europe/Lisbon
```
</details>

<details>
  <summary>Supervisor logs</summary>

```
21-07-06 13:40:18 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/aarch64-hassio-cli with version 2021.05.1
21-07-06 13:40:18 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/aarch64-hassio-observer with version 2021.06.0
21-07-06 13:40:18 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/aarch64-hassio-multicast with version 2021.04.0
21-07-06 13:40:18 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
21-07-06 13:40:20 INFO (MainThread) [supervisor.homeassistant.secrets] Loaded 0 Home Assistant secrets
21-07-06 13:40:20 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/raspberrypi4-64-homeassistant with version landingpage
21-07-06 13:40:20 INFO (MainThread) [supervisor.homeassistant.core] Starting HomeAssistant landingpage
21-07-06 13:40:20 INFO (MainThread) [supervisor.hassos] Detect Home Assistant Operating System 6.1 / BootSlot A
21-07-06 13:40:21 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/core repository
[... 687 lines cut ...]
[  110.062282] audit: type=1400 audit(1625575218.093:158): apparmor="STATUS" operation="profile_replace" profile="unconfined" name="hassio-supervisor///usr/bin/gdbus" pid=4605 comm="apparmor_parser"
[  110.062344] audit: type=1400 audit(1625575218.093:158): apparmor="STATUS" operation="profile_replace" profile="unconfined" name="hassio-supervisor///usr/bin/git" pid=4605 comm="apparmor_parser"
[  144.477028] audit: type=1334 audit(1625575252.478:159): prog-id=16 op=UNLOAD
[  144.477060] audit: type=1334 audit(1625575252.478:160): prog-id=15 op=UNLOAD
[  144.529464] audit: type=1334 audit(1625575252.530:161): prog-id=18 op=UNLOAD
[  144.529493] audit: type=1334 audit(1625575252.530:162): prog-id=17 op=UNLOAD
[  821.698471] hassio: port 6(veth2c4e057) entered blocking state
[  821.698492] hassio: port 6(veth2c4e057) entered disabled state
[  821.698816] device veth2c4e057 entered promiscuous mode
[  821.699763] audit: type=1700 audit(1625575929.770:163): dev=veth2c4e057 prom=256 old_prom=0 auid=4294967295 uid=0 gid=0 ses=4294967295
[  822.401577] audit: type=1325 audit(1625575930.454:164): table=nat family=2 entries=0 op=xt_register pid=7098 subj==unconfined comm="iptables"
[  822.425594] audit: type=1325 audit(1625575930.482:165): table=filter family=2 entries=0 op=xt_register pid=7099 subj==unconfined comm="iptables"
[  822.461694] audit: type=1325 audit(1625575930.514:166): table=nat family=2 entries=5 op=xt_replace pid=7102 subj==unconfined comm="iptables"
[  822.486341] audit: type=1325 audit(1625575930.538:167): table=nat family=2 entries=7 op=xt_replace pid=7105 subj==unconfined comm="iptables"
[  822.513605] audit: type=1325 audit(1625575930.566:168): table=nat family=2 entries=8 op=xt_replace pid=7109 subj==unconfined comm="iptables"
[  822.545779] audit: type=1325 audit(1625575930.590:169): table=nat family=2 entries=10 op=xt_replace pid=7110 subj==unconfined comm="iptables"
[  822.577645] audit: type=1325 audit(1625575930.626:170): table=nat family=2 entries=11 op=xt_replace pid=7111 subj==unconfined comm="iptables"
[  822.609700] audit: type=1325 audit(1625575930.658:171): table=nat family=2 entries=12 op=xt_replace pid=7112 subj==unconfined comm="iptables"
[  822.641683] audit: type=1325 audit(1625575930.686:172): table=nat family=2 entries=13 op=xt_replace pid=7113 subj==unconfined comm="iptables"
[  822.712629] eth0: renamed from veth1ae75ec
[  822.730926] IPv6: ADDRCONF(NETDEV_CHANGE): veth2c4e057: link becomes ready
[  822.731071] hassio: port 6(veth2c4e057) entered blocking state
[  822.731081] hassio: port 6(veth2c4e057) entered forwarding state
```
</details>

agners, 2021-07-07:
In https://github.com/hassio-addons/addon-portainer/issues/115 you reported that you are using WiFi. Is that still true?

I don't think that https://github.com/hassio-addons/addon-portainer/issues/33 is the problem here. That was related to the particular Docker version on Debian 10.

The Portainer add-on seems to work fine here on OS 6.1, so I guess it is related to your setup.

joaomlneto, 2021-07-07:
Hi @agners! Thanks for the reply :)

> In [hassio-addons/addon-portainer#115](https://github.com/hassio-addons/addon-portainer/issues/115) you reported that you are using WiFi. Is that still true?

It happens both via WiFi or Ethernet, including without the "configure WiFi" steps stated on that issue.

1. When I opened the [hassio-addons/addon-portainer#115](https://github.com/hassio-addons/addon-portainer/issues/115) issue, I was using WiFi (just for the lack of an extra network cable for the time being)
2. I then decided to disable WiFi and attaching a ethernet cable, which resulted in the same error.
3. I decided to reinstall everything (download 6.1 according to the official docs, and perform no *advanced* steps) as stated in the initial comment on this issue. I had the same issue.

I also SSH into the machine and did a `wget` of something resembling the URL where `extensions.json` should be. I recall testing against https://portainer-io-assets.sfo2.digitaloceanspaces.com/extensions-1.24.1.json and was able to retrieve the file.

I think these steps should rule out networking issues on my specific setup.

> I don't think that [hassio-addons/addon-portainer#33](https://github.com/hassio-addons/addon-portainer/issues/33) is the problem here. That was related to the particular Docker version on Debian 10.

Maybe I'm running into some similar stuff somewhere? I haven't really thought much about it! Would appreciate some pointers if you can think of something. :)

> The Portainer add-on seems to work fine here on OS 6.1, so I guess it is related to your setup.

I was also unable to find anyone with the exact same issue, but I'm completely puzzled.
I like to keep things fairly standard; the more I complicate things, the more annoying it is to maintain them :)
In the last ethernet-only scenario I was connecting the Raspberry directly to my router.
Raspberry gets an IP address via DHCP and uses it easily.
Only change from defaults is that I assign a fixed IP to the Raspberry on the router side.

agners, 2021-07-12:
> I decided to reinstall everything (download 6.1 according to the official docs, and perform no advanced steps) as stated in the initial comment on this issue. I ran into the same error.

Do I understand correctly that this reinstall was using Ethernet from start?

> I also SSH into the machine and

Did you connect to the underlying host (port 22222) or using the SSH & Terminal Add-On?

> Would appreciate some pointers if you can think of something. :)

I don't really have an idea what the problem could be. I'd check the add-on logs, try to `docker exec` into the add-on and try network access from there.

joaomlneto, 2021-07-12:
> Do I understand correctly that this reinstall was using Ethernet from start?

That's correct. 💯

> Did you connect to the underlying host (port 22222) or using the SSH & Terminal Add-On?

If I recall correctly, I tried
1. using the add-on web interface
2. connecting remotely via `ssh user@homeassistant.local` on the default port (where `user` is the one configured in the SSH add-on… so I guess it's the add-on again 🤷🏻‍♂️)

I'll attempt via port 22222 in a few days.

> I don't really have an idea what the problem could be. I'd check the add-on logs, try to `docker exec` into the add-on and try network access from there.

I will try in a few days, and I'll make sure to update this issue with more findings/information! Thanks for the time!

pvizeli, 2021-08-23:
Portainer is not supported by core team

closed 2021-08-23 by pvizeli

joaomlneto, 2021-08-24:
> Portainer is not supported by core team

Do you have any recommendations where to report it then?