home-assistant/operating-system #1450: Current measurement for RPi PoE+ hat not working

georgeboot, 2021-07-09:
**Hardware Environment**

- [x] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Generic x86-64 (like Intel NUC)
- [ ] OVA (Open Virtualization Appliance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [x] Fresh installation of release 6.1
- [ ] Updated from version x.y

**Problem**
I can't measure the power the PoE+ hat provides through the OS.

It's supposed to be measured using:

```
cat /sys/devices/platform/rpi-poe-power-supply@0/power_supply/rpi-poe/current_now
```

Perhaps there are some modules that need to be enabled in the kernel to support the new hat out of the box?

agners, 2021-07-12:
It seems that the PoE+ Hat support has been added with Raspberry Pi Kernel `1.20210527-1`. I'll update with the next release.

agners, 2021-07-29:
Release 6.2 is out, please test and reopen in case this is still a problem.

closed 2021-07-29 by agners

georgeboot, 2022-08-14:
@agners this still doesn't work on the latest version:

```
Logger: homeassistant.components.command_line
Source: components/command_line/__init__.py:48 
Integration: command_line (documentation, issues) 
First occurred: 15:55:35 (331 occurrences) 
Last logged: 21:25:34

Command failed (with return code 1): cat /sys/devices/platform/rpi-poe-power-supply@0/power_supply/rpi-poe/current_now
```

```yml
sensor:
- platform: command_line
  name: RPi CPU Temperature
  command: "cat /sys/class/thermal/thermal_zone0/temp"
  unit_of_measurement: "°C"
  value_template: "{{ value | multiply(0.001) | round(1) }}"
```

agners, 2022-08-18:
Please open a new issue with all the details provided.

Did you load the `rpi-poe-plus` overlay?

georgeboot, 2022-08-18:
A combination of setting the overlay to `rpi-poe-plus` (I was still using the old `rpi-poe` overlay) and reading from `/sys/devices/platform/rpi-poe-power-supply/power_supply/rpi-poe/current_now` (without the `@0`) instead, fixed it for me.