home-assistant/operating-system #1455: Add support for NFS / CIFS to be used by subquential software

alekslyse, 2021-07-10:
I do understand that Home Assistant OS should be quite clean, but I am not sure why mount.nfs / cifs is removed, and when mounting throwing the error "for several filesystems (e.g. nfs, cifs) you might need a /sbin/mount.<type> helper program." - for many would mounting for example the media folder to a NFS, or using a NFS volume in portainer be a very good idea to reduce the load on the local r/w.

I think the missing mount files are also the reason why portainer / docker volume create on NFS / CIFS does not work, or should mounting a NFS on the docker on Home Assistant actually work? (for me it creates, but does not consume from the server, so I assume it fails cause of the missing mounting files).

agners, 2021-07-12:
In general we don't have a good way to mount any storage device persistently. Along with that, mounting network file system is also still "unsolved".

> I think the missing mount files are also the reason why portainer / docker volume create on NFS / CIFS does not work,

Did you test this? Are you sure its not working? At least in a quick test Docker seems to mount cifs.

alekslyse, 2021-07-12:
It works inside the home assistant with ntfs4. And yes i can mount nfs and cifs in portainer as a volume, and the mount file show that it uses remote, but i see no content in the folder even if its files on the remote server. Can you confirm that the mount you made contain remote files? It seems like it say it mounts it, but dont really.

sairon, 2026-09-17:
This is stale and OS isn't involved here - Supervisor already supports mounting NFS or CIFS network filesystems, and local mounts (requested in the last comment) should come soon as well.

closed 2026-09-17 by sairon