home-assistant/operating-system #1458: Cannot SSH on port 22222 (connection closed by remote host)

r-jordan, 2021-07-12:
**Hardware Environment**

-  Raspberry Pi 3B

**Home Assistant OS release:**
-  Fresh installation of release 6.1
- Additional information (if accessible):

```
~ $ ha info
arch: aarch64
channel: stable
docker: 20.10.6
features:
- reboot
- shutdown
- services
- network
- hostname
- timedate
- hassos
hassos: "6.1"
homeassistant: 2021.7.1
hostname: homeassistant
logging: info
machine: raspberrypi3-64
operating_system: Home Assistant OS 6.1
state: running
supervisor: 2021.06.8
supported: true
supported_arch:
- aarch64
- armv7
- armhf
timezone: Europe/Madrid
```

**Supervisor logs:**

```
~ $ ha su logs                                                                                                                                           
[s6-init] ensuring user provided files have correct perms...exited 0.                                                                                    
[fix-attrs.d] applying ownership & permissions fixes...                                                                                                  
[fix-attrs.d] done.                                                                                                                                      
[cont-init.d] executing container initialization scripts...                                                                                              
[cont-init.d] udev.sh: executing...                                                                                                                      
[18:31:28] INFO: Using udev information from host                                                                                                        
[cont-init.d] udev.sh: exited 0.                                                                                                                         
[cont-init.d] done.                                                                                                                                      
[services.d] starting services                                                                                                                           
[services.d] done.                                                                                                                                       
[18:31:28] INFO: Starting local supervisor watchdog...                                                                                                   
21-07-12 18:31:36 INFO (MainThread) [__main__] Initializing Supervisor setup                                                                             
21-07-12 18:31:36 INFO (MainThread) [supervisor.bootstrap] Initializing Supervisor Sentry                                                                
21-07-12 20:31:36 INFO (MainThread) [supervisor.bootstrap] Seting up coresys for machine: raspberrypi3-64                                                
21-07-12 20:31:36 INFO (SyncWorker_0) [supervisor.docker.supervisor] Attaching to Supervisor homeassistant/aarch64-hassio-supervisor with version 2021.06
.8                                                                                                                                                       
21-07-12 20:31:36 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.INITIALIZE                          
21-07-12 20:31:37 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete                                                          
21-07-12 20:31:37 INFO (MainThread) [__main__] Setting up Supervisor                                                                                     
21-07-12 20:31:37 INFO (MainThread) [supervisor.api] Starting API on 172.30.32.2                                                                         
21-07-12 20:31:37 INFO (MainThread) [supervisor.hardware.monitor] Started Supervisor hardware monitor                                                    
21-07-12 20:31:39 INFO (MainThread) [supervisor.host.info] Updating local host information                                                               
21-07-12 20:31:39 INFO (MainThread) [supervisor.host.services] Updating service information                                                              
21-07-12 20:31:40 INFO (MainThread) [supervisor.host.network] Updating local network information                                                         
21-07-12 20:31:42 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information                                                              
21-07-12 20:31:42 INFO (MainThread) [supervisor.host] Host information reload completed                                                                  
21-07-12 20:31:42 INFO (MainThread) [supervisor.host.apparmor] Loading AppArmor Profiles: {'hassio-supervisor'}                                          
21-07-12 20:31:42 INFO (MainThread) [supervisor.host.services] Reloading local service hassos-apparmor.service                                           
21-07-12 20:31:42 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/aarch64-hassio-dns with version 2021.06.0                 
[... 109 lines cut ...]
debug1: SSH2_MSG_NEWKEYS sent
debug1: expecting SSH2_MSG_NEWKEYS
debug1: SSH2_MSG_NEWKEYS received
debug1: rekey after 134217728 blocks
debug1: SSH2_MSG_EXT_INFO received
debug1: kex_input_ext_info: server-sig-algs=<ssh-ed25519,ecdsa-sha2-nistp256,ecdsa-sha2-nistp384,ecdsa-sha2-nistp521,rsa-sha2-256,ssh-rsa>
debug1: SSH2_MSG_SERVICE_ACCEPT received
debug1: Authentications that can continue: publickey
debug1: Next authentication method: publickey
debug1: Offering RSA public key: .ssh/id_rsa2
debug1: Server accepts key: pkalg rsa-sha2-256 blen 279
debug1: Authentication succeeded (publickey).
Authenticated to replaced.duckdns.org ([x.x.x.x]:22222).
debug1: channel 0: new [client-session]
debug1: Entering interactive session.
debug1: pledge: network
debug1: channel 0: free: client-session, nchannels 1
Connection to replaced.duckdns.org closed by remote host.
Connection to replaced.duckdns.org closed.
Transferred: sent 2536, received 1256 bytes, in 0.0 seconds
Bytes per second: sent 10876027.6, received 5386549.9
debug1: Exit status -1
```

I installed hassOS 6.1 from fresh twice with the same result.

agners, 2021-07-13:
Seems to work here:

```
...
debug1: Authentication succeeded (publickey).
Authenticated to hostname ([x.x.x.x]:22222).
debug1: channel 0: new [client-session]
debug1: Entering interactive session.
debug1: pledge: filesystem full
Welcome to Home Assistant OS.

Use `ha` to access the Home Assistant CLI.
# 
```

What distribution/ssh version are you using on client side?

Is the server side log saying something? You can login using a local terminal/console and use:
```
journalctl -u dropbear.service
```

r-jordan, 2021-07-13:
Hi @agners !

The Linux distribution in my laptop is:

```
$ cat /etc/os-release 
NAME="Linux Mint"
VERSION="18 (Sarah)"
ID=linuxmint
ID_LIKE=ubuntu
PRETTY_NAME="Linux Mint 18"
VERSION_ID="18"
HOME_URL="http://www.linuxmint.com/"
SUPPORT_URL="http://forums.linuxmint.com/"
BUG_REPORT_URL="http://bugs.launchpad.net/linuxmint/"
UBUNTU_CODENAME=xenial
```

You're righ, the problem was in the client side. It seems that the ssh-agent that cinnamon starts is malconfigured in my laptop. The same ssh command begins to work when I wrote the following command:

```
unset SSH_AUTH_SOCK
```

Now I can investigate why this is happening but I have a work around and may be I would upgrade my OS in a near future :wink: 

I close the issue because it has nothing to do with hassOS.

Thanks for all anyway.

closed 2021-07-13 by r-jordan