home-assistant/operating-system #1460: HAOS 6.1 triggers timed automations at wrong time on ESXi 7.0.1. HAOS 5.12 did not.

tangix, 2021-07-14:
**Hardware Environment**
- [ ] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Generic x86-64 (like Intel NUC)
- [X] OVA ESXi 7.0.1 and vCenter running on ASUS MB with i5-10600K 4,2GHz and 128GB RAM

**Home Assistant OS release:**
- [X] Updated from version 5.12
- Additional information (if accessible):
```
~ $ ha info
arch: amd64
channel: stable
docker: 20.10.6
features:
- reboot
- shutdown
- services
- network
- hostname
- timedate
- hassos
hassos: "6.1"
homeassistant: 2021.7.2
hostname: home-assistant.***.***.***
logging: info
machine: qemux86-64
operating_system: Home Assistant OS 6.1
state: running
supervisor: 2021.06.8
supported: true
supported_arch:
- amd64
- i386
timezone: Europe/Stockholm
```

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**
Installed HAOS 5.12 on our main automation VM using the latest OVA available in March 2021. We have allowed the system to update when prompted and are now running HAOS 6.1 with HA 2021.7.2. The problem is now that time based automations are not firing. Everything else is working as expected - Zwave USB, hardware triggered automations etc.

As a regression test I created another VM with the HAOS 5.12 OVA in the same ESXi environment, updated to HA 2021.7.2 and time based automations are working in that installation, firing on the second.

See https://community.home-assistant.io/t/haos-6-1-on-esxi-7-0-1-not-triggering-automations-5-12-did/321947 for my additional findings.

agners, 2021-07-14:
There is a lot new in release 6: New build system (which brings new glibc, Docker update to 20.10 and other things) as well as a new Linux kernel version (5.10). But I am not aware that a change should change behavior of timed automations.

In a quick test "Time" and "Time pattern" automations do work on my OS 6.1 OVA installation (using Qemu though) automations seem to work.

Is this also happening when using just "Time" or "Time Pattern" as trigger type? Have you checked the time zone in Configuration -> General -> Time Zone?

tangix, 2021-07-14:
Both Time and Time Pattern. A simple automation like
```
alias: Utebelysning kväll (släck)
description: ''
trigger:
  - platform: time
    at: '23:00:00'
condition: []
action:
  - service: script.sla_av_utebelysningen
mode: restart
```
typically runs 90 to 135 minutes late as shown in the trace:
[image]

Time zone OK and output from the normal troubleshooting looks like:
```
{{ utcnow() }}
{{ now() }}
{{ utcnow().astimezone() }}
{{ now().astimezone() }}
{{ utcnow().tzinfo }}
{{ now().tzinfo }}
{{ now().astimezone().tzinfo }}
```
produces:
```
2021-07-14 13:33:00.000612+00:00
2021-07-14 15:33:00.000649+02:00
2021-07-14 15:33:00.000663+02:00
2021-07-14 15:33:00.000717+02:00
UTC
Europe/Stockholm
CEST
```
I was just prompted by the 5.12 installation to update to 6.1 and I will see if the test automation will start to fail.

tangix, 2021-07-14:
BTW: Everything else from the UI works as expected - hitting a button executes a script directly and no other noticeable slow-down.

agners, 2021-07-14:
That time in the trace is somewhat strange (24:35:29, is that a valid time?)

Just to rule out side effects from other things, can you toggle a simple toggle instead of running a script, e.g.:

```
service: input_boolean.toggle
target:
  entity_id: input_boolean.test
```

tangix, 2021-07-14:
> That time in the trace is somewhat strange (24:35:29, is that a valid time?)

No, it's not - https://github.com/home-assistant/frontend/issues/9379

> Just to rule out side effects from other things, can you toggle a simple toggle instead of running a script, e.g.:

Will try.

pvizeli, 2021-07-14:
can you post `ha os info`?

tangix, 2021-07-15:
> can you post `ha os info`?

```
~ $ ha os info
board: ova
boot: B
update_available: false
version: "6.1"
version_latest: "6.1"
```

> Can you make sure that HA is also set up in your timezone as Stefan asked for: `Have you checked the time zone in Configuration -> General -> Time Zone?`

Time zone is displayed correctly as stated above.

pvizeli, 2021-07-15:
sorry :) `ha host info`

tangix, 2021-07-15:
> sorry :) `ha host info`

```
# ha host info
chassis: vm
cpe: cpe:2.3:o:home-assistant:haos:6.1:*:production:*:*:*:ova:*
deployment: production
disk_free: 21.8
disk_life_time: null
disk_total: 30.8
disk_used: 7.7
dt_synchronized: true
dt_utc: "2021-07-15T09:49:48.926119+00:00"
features:
- reboot
- shutdown
- services
- network
- hostname
- timedate
- hassos
hostname: home-assistant.***.***.***
kernel: 5.10.45
operating_system: Home Assistant OS 6.1
timezone: Etc/UTC
use_ntp: true
use_rtc: false
```

tangix, 2021-07-17:
Researching this further and I think I have solved the issue and timed automations run properly.
I installed the 6.1 OVA and compared the settings with my original installation in vSphere ESXi 7.0.1. 
Two differences were found:
1. VM compatibility set to version 18
2. Setting under VM Option - "Synchronize time periodically" was disabled.

I have updated VM compatibility of my original HAOS installation and also disabled the periodically time synch (HAOS runs its own NTP client so the time seems to be correct) and my timed automations run at the correct time.

closed 2021-07-17 by tangix