home-assistant/operating-system #1466: Host (Raspberry Pi) crashes and goes offline

auanasgheps, 2021-07-16:
**Hardware Environment**

- [x] Raspberry Pi [1/2/3/4] - Pi 4 4GB RAM
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Generic x86-64 (like Intel NUC)
- [ ] OVA (Open Virtualization Appliance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [ ] Fresh installation of release x.y
- [x] Updated from version 2021.6 (installed on 29th June 2021)
- Additional information (if accessible):
## System Health

version | core-2021.7.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
virtualenv | false
python_version | 3.9.5
os_name | Linux
os_version | 5.10.17-v8
arch | aarch64
timezone | Europe/Rome

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 4751
Installed Version | 1.13.2
Stage | running
Available Repositories | 847
Installed Repositories | 13

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 6.1
-- | --
update_channel | stable
supervisor_version | supervisor-2021.06.8
docker_version | 20.10.6
disk_total | 109.3 GB
disk_used | 6.1 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | Home Assistant Google Drive Backup (0.104.3), deCONZ (6.9.0), SSH & Web Terminal (9.0.0), Visual Studio Code (3.6.0), File editor (5.3.2), Check Home Assistant configuration (3.8.0), Let's Encrypt (4.11.0), NGINX Home Assistant SSL proxy (3.0.1)

</details>

<details><summary>Lovelace</summary>

dashboards | 1
[... 79 lines cut ...]
Jul 16 02:40:07 homeassistant kernel: hassio: port 7(vethbee281e) entered blocking state
Jul 16 02:40:07 homeassistant kernel: hassio: port 7(vethbee281e) entered forwarding state
Jul 16 03:09:40 homeassistant kernel: kauditd_printk_skb: 1 callbacks suppressed
Jul 16 03:09:40 homeassistant kernel: audit: type=1334 audit(1626404980.617:831): prog-id=321 op=LOAD
Jul 16 03:09:40 homeassistant kernel: audit: type=1334 audit(1626404980.621:832): prog-id=322 op=LOAD
Jul 16 03:09:40 homeassistant kernel: audit: type=1334 audit(1626404980.965:833): prog-id=323 op=LOAD
Jul 16 03:09:40 homeassistant kernel: audit: type=1334 audit(1626404980.965:834): prog-id=324 op=LOAD
Jul 16 03:10:10 homeassistant kernel: audit: type=1334 audit(1626405010.981:835): prog-id=322 op=UNLOAD
Jul 16 03:10:10 homeassistant kernel: audit: type=1334 audit(1626405010.981:836): prog-id=321 op=UNLOAD
Jul 16 03:10:11 homeassistant kernel: audit: type=1334 audit(1626405011.273:837): prog-id=324 op=UNLOAD
Jul 16 03:10:11 homeassistant kernel: audit: type=1334 audit(1626405011.273:838): prog-id=323 op=UNLOAD
Jul 16 03:36:54 homeassistant kernel: audit: type=1701 audit(1626406516.169:839): auid=4294967295 uid=0 gid=0 ses=4294967295 subj==unconfined pid=373 comm="systemd-logind" exe="/usr/lib/systemd/systemd-logind" sig=6 res=1

```

**Description of problem:**
Every 3 to 5 days, my host crashes.
This time happened during the night, with no load whatsoever or user changes.
HASSIO is installed on an external M2 SSD connected via USB.
When the OS crashes, I'm unable to ping the host, all Pi ligths are on (status, ethernet) but the SSD light goes off.
I use Healthchecks to tell when the system crashed, last ping was 03:35 (logs time)

I've found similar issues but don't know if they are relevant (and/or not resolved)
#1336
#1119

agners, 2021-07-19:
It looks like processes hanging and getting killed by systemd for some reason.

Is it maybe memory related? The system monitor integration might help to track memory usage.

Unfortunately, several USB SSD combinations are known to be problematic with newer Raspberry Pi kernels, but there is no clear understanding yet what the problem is exactly.

Do you have any output on the system console when this happens?

auanasgheps, 2021-07-19:
> Do you have any output on the system console when this happens?

How can I get this? There's no monitor connected to the Pi.

There's no memory issue, RAM is at 25-30% usage.

Tonight happened again. I can post the logs, but this time there's simply no error that can lead to something.

My USB SSD enclosure worked fine with my previous Raspberry Pi 3B. 

Another thing that comes to my mind is that I have restored the Pi 3B backup on the Pi 4B, after the clean install. Is this supported?

flipdream, 2021-07-23:
Same issue here

bschatzow, 2021-07-23:
@auanasgheps 
Have you tried either down grading the HAOS to 5.4 or below? That worked for me. Also debian with ha as supervisor. Neither has crashed in months of use.

auanasgheps, 2021-07-23:
> @auanasgheps
> Have you tried either down grading the HAOS to 5.4 or below? That worked for me. Also debian with ha as supervisor. Neither has crashed in months of use.

I'm using HASSIO.
I tried downgrading to 5.4 but I had the audio container crashing every minute in the logs. I joined the beta, with updated the supervisor again to supervisor-2021.06.8.
I've done it a couple of days ago so I'm expecting a crash soon.

Even if we downgrade to 5.4 or similar, this is not sustainable in the long term and we need to find the problem and a proper fix!

bschatzow, 2021-07-23:
Lots of different post with this issue and no useful logs for the developers to troubleshoot. I moved to Debian and HA supervisor.  I update Debian os manually monthly and so far zero issues.

auanasgheps, 2021-07-27:
I'm starting to lose hope. 
I also quickly booted Raspbian to update the Pi firmware, it was running a build from September 2020 and now from July 2021,  but today, exactly 3 days and 7 hours later the latest reboot, it got stuck again with **zero** relevant information in journalctl logs.

I will soon go on holiday so I won't have time to fiddle with this. For the time being I will just set a nightly reboot of the host.

```
Jul 27 04:45:33 homeassistant 446376b1bc31[471]:
Jul 27 04:45:41 homeassistant 3c7e0aa7ec3b[471]: 06:45:41:870 dev /dev/ttyS0
Jul 27 04:45:42 homeassistant 3c7e0aa7ec3b[471]: 06:45:41:871 GW firmware version: 0x26690700
Jul 27 04:45:51 homeassistant 3c7e0aa7ec3b[471]: 06:45:51:861 dev /dev/ttyS0
Jul 27 04:45:51 homeassistant 3c7e0aa7ec3b[471]: 06:45:51:862 GW firmware version: 0x26690700
Jul 27 04:46:01 homeassistant 3c7e0aa7ec3b[471]: 06:46:01:863 dev /dev/ttyS0
Jul 27 04:46:01 homeassistant 3c7e0aa7ec3b[471]: 06:46:01:864 GW firmware version: 0x26690700
Jul 27 04:46:11 homeassistant 3c7e0aa7ec3b[471]: 06:46:11:871 dev /dev/ttyS0
Jul 27 04:46:12 homeassistant 3c7e0aa7ec3b[471]: 06:46:11:872 GW firmware version: 0x26690700
Jul 27 04:46:14 homeassistant 3c7e0aa7ec3b[471]: 06:46:14:859 Current channel 20
Jul 27 04:46:14 homeassistant 3c7e0aa7ec3b[471]: 06:46:14:872 CTRL got nwk update id 0
Jul 27 04:46:14 homeassistant 3c7e0aa7ec3b[471]: 06:46:14:885 Device TTL 5220 s flags: 0x7
Jul 27 04:46:21 homeassistant 3c7e0aa7ec3b[471]: 06:46:21:862 dev /dev/ttyS0
Jul 27 04:46:21 homeassistant 3c7e0aa7ec3b[471]: 06:46:21:863 GW firmware version: 0x26690700
Jul 27 04:46:22 homeassistant 446376b1bc31[471]: 21-07-27 06:46:22 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.RUNNING
Jul 27 04:46:22 homeassistant hassos-supervisor[979]: 21-07-27 06:46:22 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.RUNNING
Jul 27 04:46:22 homeassistant hassos-supervisor[979]: 21-07-27 06:46:22 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.FREE_SPACE/ContextType.SYSTEM
Jul 27 04:46:22 homeassistant hassos-supervisor[979]: 21-07-27 06:46:22 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.SECURITY/ContextType.CORE
Jul 27 04:46:22 homeassistant 446376b1bc31[471]: 21-07-27 06:46:22 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.FREE_SPACE/ContextType.SYSTEM
Jul 27 04:46:22 homeassistant 446376b1bc31[471]: 21-07-27 06:46:22 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.SECURITY/ContextType.CORE
Jul 27 04:46:22 homeassistant 446376b1bc31[471]: 21-07-27 06:46:22 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.PWNED/ContextType.ADDON
Jul 27 04:46:22 homeassistant hassos-supervisor[979]: 21-07-27 06:46:22 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.PWNED/ContextType.ADDON
Jul 27 04:46:22 homeassistant 446376b1bc31[471]: 21-07-27 06:46:22 INFO (MainThread) [supervisor.resolution.check] System checks complete
Jul 27 04:46:22 homeassistant hassos-supervisor[979]: 21-07-27 06:46:22 INFO (MainThread) [supervisor.resolution.check] System checks complete
Jul 27 04:46:22 homeassistant 446376b1bc31[471]: 21-07-27 06:46:22 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.RUNNING
Jul 27 04:46:22 homeassistant hassos-supervisor[979]: 21-07-27 06:46:22 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.RUNNING
Jul 27 04:46:22 homeassistant 446376b1bc31[471]: 21-07-27 06:46:22 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
Jul 27 04:46:22 homeassistant hassos-supervisor[979]: 21-07-27 06:46:22 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
Jul 27 04:46:22 homeassistant 446376b1bc31[471]: 21-07-27 06:46:22 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state CoreState.RUNNING
Jul 27 04:46:22 homeassistant hassos-supervisor[979]: 21-07-27 06:46:22 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state CoreState.RUNNING
Jul 27 04:46:22 homeassistant 446376b1bc31[471]: 21-07-27 06:46:22 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
Jul 27 04:46:22 homeassistant hassos-supervisor[979]: 21-07-27 06:46:22 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
Jul 27 04:46:31 homeassistant 3c7e0aa7ec3b[471]: 06:46:31:862 dev /dev/ttyS0
Jul 27 04:46:31 homeassistant 3c7e0aa7ec3b[471]: 06:46:31:863 GW firmware version: 0x26690700
Jul 27 04:46:41 homeassistant 3c7e0aa7ec3b[471]: 06:46:41:865 dev /dev/ttyS0
Jul 27 04:46:42 homeassistant 3c7e0aa7ec3b[471]: 06:46:41:866 GW firmware version: 0x26690700
Jul 27 04:46:51 homeassistant 3c7e0aa7ec3b[471]: 06:46:51:862 dev /dev/ttyS0
Jul 27 04:46:51 homeassistant 3c7e0aa7ec3b[471]: 06:46:51:863 GW firmware version: 0x26690700
Jul 27 04:47:01 homeassistant 3c7e0aa7ec3b[471]: 06:47:01:863 dev /dev/ttyS0
Jul 27 04:47:01 homeassistant 3c7e0aa7ec3b[471]: 06:47:01:864 GW firmware version: 0x26690700
Jul 27 04:47:11 homeassistant 3c7e0aa7ec3b[471]: 06:47:11:864 dev /dev/ttyS0
Jul 27 04:47:12 homeassistant 3c7e0aa7ec3b[471]: 06:47:11:865 GW firmware version: 0x26690700
Jul 27 04:47:14 homeassistant 3c7e0aa7ec3b[471]: 06:47:14:857 Current channel 20
[... cut ...]
Jul 27 04:55:21 homeassistant 3c7e0aa7ec3b[471]: 06:55:21:863 dev /dev/ttyS0
Jul 27 04:55:21 homeassistant 3c7e0aa7ec3b[471]: 06:55:21:864 GW firmware version: 0x26690700
Jul 27 04:55:31 homeassistant 3c7e0aa7ec3b[471]: 06:55:31:863 dev /dev/ttyS0
Jul 27 04:55:31 homeassistant 3c7e0aa7ec3b[471]: 06:55:31:864 GW firmware version: 0x26690700
Jul 27 04:55:34 homeassistant 446376b1bc31[471]:
Jul 27 04:55:41 homeassistant 3c7e0aa7ec3b[471]: 06:55:41:861 dev /dev/ttyS0
Jul 27 04:55:41 homeassistant 3c7e0aa7ec3b[471]: 06:55:41:862 GW firmware version: 0x26690700
```

vukisz, 2021-09-29:
@auanasgheps did you resolve this? I am having similar issue

auanasgheps, 2021-09-29:
@vukisz nope, I haven't. I started discussing it in #1119 which is a similar issue which has more traction.

So far looks like there could be several root causes, but HASSOS 5.4 is the latest stable build for everyone that is facing these crashes.

I've offered my help in that discussion and I hope together we'll find the root cause. If your `journalctl `logs show something relevant, please post it there.

I will keep this issue open and update it when possibile, I'm very active on Github and I don't like to leave things behind!

vukisz, 2021-09-30:
@auanasgheps thank you I will continue in there then :-)

bschatzow, 2021-10-20:
@auanasgheps Oliver, Great job!  Hopefully you can confirm it works and @agners can create a custom build for more of us to test. Are you sure it is firmware?  This is the link to the pi firmware: 
``` https://github.com/raspberrypi/rpi-eeprom/tree/master/firmware/stable```
I believe from an ssh 22222 window you can run vcgencmd version to get the version.  On my current setup I ran docker container exec homeassistant /opt/vc/bin/vcgencmd bootloader_version and it shows 3_18_21
If you are really talking firmware, the firmware that was stable when 5.4 was released is a Sept 2020 firmware.  This firmware and anything above 5.4 did not work for me.

rizer1980, 2021-10-31:
## System Health

version | core-2021.10.6
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.7
os_name | Linux
os_version | 5.10.17-v8
arch | aarch64
timezone | Europe/Moscow

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 6.5
-- | --
update_channel | stable
supervisor_version | supervisor-2021.10.6
docker_version | 20.10.7
disk_total | 116.7 GB
disk_used | 6.7 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.3.3), Mosquitto broker (6.0.1), Samba share (9.5.1), Terminal & SSH (9.0.2), Zigbee2mqtt (1.21.2-1)

</details>

<details><summary>Lovelace</summary>

dashboards | 1
-- | --
resources | 0
views | 2
mode | storage

</details>

The same problem as many here, pi 4 hangs every 2-5 days.
I connected a monitor, that's what was in the logs when it hung up, I hope it helps.
ps. pi 4 with ssd only.
[image]
[image]

auanasgheps, 2021-11-01:
@rizer1980 please check out #1119. We've found a (hacky) workaround: use HASSOS 5.4 FW files on HASSOS 6.5 For many of us is stable. We're still investigating, but if you can confirm that is stable for you it would be great.
All details can be found in the referenced issue.

auanasgheps, 2022-01-04:
Following #1119, this issue is fixed in HASSOS 7.1, thanks to an updated RPi firmware.
Hurray!

closed 2022-01-04 by auanasgheps