home-assistant/operating-system #1467: host_internet: false - after OS 6.1 installed

Masterz69, 2021-07-16:
**Hardware Environment**

- [X] Raspberry Pi [4]

**Home Assistant OS release:**
- [X] Updated from version 5.?
- Additional information (if accessible):

```
➜  ~ ha info
arch: aarch64
channel: stable
docker: 20.10.6
features:
- reboot
- shutdown
- services
- network
- hostname
- timedate
- hassos
hassos: "6.1"
homeassistant: 2021.6.6
hostname: ha
logging: info
machine: raspberrypi4-64
operating_system: Home Assistant OS 6.1
state: running
supervisor: 2021.06.8
supported: true
supported_arch:
- aarch64
- armv7
- armhf
timezone: Europe/Riga
```

**Supervisor logs:**
```
21-07-15 17:21:28 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
21-07-15 17:21:28 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
21-07-15 17:21:49 INFO (MainThread) [supervisor.snapshots] Creating new partial-snapshot with slug c63238c1
21-07-15 17:21:49 INFO (MainThread) [supervisor.snapshots] Snapshotting c63238c1 store Add-ons
21-07-15 17:21:49 INFO (MainThread) [supervisor.addons.addon] Building snapshot for add-on a0d7b954_phpmyadmin
21-07-15 17:21:49 INFO (MainThread) [supervisor.addons.addon] Finish snapshot for addon a0d7b954_phpmyadmin
21-07-15 17:21:49 INFO (MainThread) [supervisor.snapshots] Creating partial-snapshot with slug c63238c1 completed
21-07-15 17:21:49 WARNING (MainThread) [supervisor.jobs] 'AddonManager.update' blocked from execution, no host internet connection

21-07-16 17:29:26 INFO (MainThread) [supervisor.misc.scheduler] Shutting down scheduled tasks
21-07-16 17:29:26 INFO (MainThread) [supervisor.api] Stopping API on 172.30.32.2
21-07-16 17:29:26 INFO (MainThread) [supervisor.hardware.monitor] Stopped Supervisor hardware monitor
21-07-16 17:29:26 INFO (MainThread) [supervisor.core] Supervisor is down - 100
21-07-16 17:29:26 INFO (MainThread) [__main__] Closing Supervisor
21-07-16 14:29:30 INFO (MainThread) [__main__] Initializing Supervisor setup
21-07-16 14:29:30 INFO (MainThread) [supervisor.bootstrap] Initializing Supervisor Sentry
21-07-16 17:29:30 INFO (MainThread) [supervisor.bootstrap] Seting up coresys for machine: raspberrypi4-64
21-07-16 17:29:30 INFO (SyncWorker_0) [supervisor.docker.supervisor] Attaching to Supervisor homeassistant/aarch64-hassio-supervisor with version 2021.06.8
21-07-16 17:29:30 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
21-07-16 17:29:30 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.INITIALIZE
21-07-16 17:29:30 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
21-07-16 17:29:30 INFO (MainThread) [__main__] Setting up Supervisor
21-07-16 17:29:31 INFO (MainThread) [supervisor.api] Starting API on 172.30.32.2
21-07-16 17:29:31 INFO (MainThread) [supervisor.hardware.monitor] Started Supervisor hardware monitor
21-07-16 17:29:32 INFO (MainThread) [supervisor.host.info] Updating local host information
21-07-16 17:29:32 INFO (MainThread) [supervisor.host.services] Updating service information
21-07-16 17:29:32 INFO (MainThread) [supervisor.host.network] Updating local network information
21-07-16 17:29:45 WARNING (MainThread) [supervisor.host.network] Can't update connectivity information: Error: Timeout was reached
21-07-16 17:29:45 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
21-07-16 17:29:45 INFO (MainThread) [supervisor.host] Host information reload completed
21-07-16 17:29:45 INFO (MainThread) [supervisor.host.apparmor] Loading AppArmor Profiles: {'hassio-supervisor'}
[... 1093 lines cut ...]
          TX packets:190868 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:472850224 (450.9 MiB)  TX bytes:16102521 (15.3 MiB)

wlan0     Link encap:Ethernet  HWaddr E4:5F:01:05:EE:6F
          inet addr:192.168.1.11  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fe80::4979:b09a:2173:3e6f/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:3219459 errors:0 dropped:0 overruns:0 frame:0
          TX packets:3572923 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:1976310324 (1.8 GiB)  TX bytes:4072284784 (3.7 GiB)

➜  ~ netstat -rn
Kernel IP routing table
Destination     Gateway         Genmask         Flags   MSS Window  irtt Iface
0.0.0.0         192.168.1.1     0.0.0.0         UG        0 0          0 wlan0
0.0.0.0         192.168.1.1     0.0.0.0         UG        0 0          0 eth0
172.17.0.0      0.0.0.0         255.255.0.0     U         0 0          0 docker0
172.30.32.0     0.0.0.0         255.255.254.0   U         0 0          0 hassio
192.168.1.0     0.0.0.0         255.255.255.0   U         0 0          0 eth0
192.168.1.0     0.0.0.0         255.255.255.0   U         0 0          0 wlan0

```
See no 172.30.32.3 in 'ifconfig', but may be it fine.

Masterz69, 2021-07-16:
Forgot to add. Yesterday after host restart - all was fine, Supervisor was able to update add-ons.
But today again it fails. Highly likely, because of host_internet: **false**.

Masterz69, 2021-07-16:
```
➜  ~ arp -n | sort -k 1
172.17.0.2               ether   02:42:ac:11:00:02   C                     docker0
172.30.32.2              ether   02:42:ac:1e:20:02   C                     hassio
172.30.32.3              ether   02:42:ac:1e:20:03   C                     hassio
172.30.32.6              ether   02:42:ac:1e:20:06   C                     hassio
172.30.33.0              ether   02:42:ac:1e:21:00   C                     hassio
172.30.33.1              ether   02:42:ac:1e:21:01   C                     hassio
172.30.33.2              ether   02:42:ac:1e:21:02   C                     hassio
172.30.33.3              ether   02:42:ac:1e:21:03   C                     hassio
192.168.1.1              ether   74:d2:1d:be:66:53   C                     eth0
192.168.1.1              ether   74:d2:1d:be:66:53   C                     wlan0
```

agners, 2021-07-19:
Hm, it seems the Supervisor cannot talk to the Host OS through D-Bus for some reason:

```
21-07-16 17:29:45 WARNING (MainThread) [supervisor.host.network] Can't update connectivity information: Error: Timeout was reached
```

Do you see any negative impact due to this?

agners, 2021-07-28:
This might be related to #1483.

vmartinv, 2021-08-05:
Hello I'm experiencing the same problem after updating OS from 6.1 to 6.2.

supervisor logs
```
21-08-05 11:08:46 INFO (SyncWorker_5) [supervisor.docker.addon] Starting Docker add-on ghcr.io/hassio-addons/ssh/aarch64 with version 9.0.0
21-08-05 11:08:50 ERROR (MainThread) [supervisor.api.ingress] Ingress error: Cannot connect to host 172.30.32.1:64499 ssl:default [Connect call failed ('172.30.32.1', 64499)]
21-08-05 11:09:00 INFO (MainThread) [supervisor.api.middleware.security] /network/info access from a0d7b954_ssh
21-08-05 11:09:00 INFO (MainThread) [supervisor.api.middleware.security] /network/info access from a0d7b954_ssh
21-08-05 11:09:00 INFO (MainThread) [supervisor.api.middleware.security] /host/info access from a0d7b954_ssh
21-08-05 11:09:00 INFO (MainThread) [supervisor.api.middleware.security] /core/info access from a0d7b954_ssh
21-08-05 11:09:06 WARNING (MainThread) [supervisor.host.network] Can't update connectivity information: Error: Timeout was reached
21-08-05 11:09:11 INFO (MainThread) [supervisor.api.middleware.security] /network/info access from a0d7b954_ssh
21-08-05 11:09:46 WARNING (MainThread) [supervisor.host.network] Can't update connectivity information: Error: Timeout was reached
21-08-05 11:11:01 WARNING (MainThread) [supervisor.host.network] Can't update connectivity information: Error: Timeout was reached
21-08-05 11:11:42 WARNING (MainThread) [supervisor.host.network] Can't update connectivity information: Error: Timeout was reached
21-08-05 11:12:04 INFO (MainThread) [supervisor.api.middleware.security] /network/info access from a0d7b954_ssh
21-08-05 11:12:04 INFO (MainThread) [supervisor.api.middleware.security] /network/info access from a0d7b954_ssh
21-08-05 11:12:04 INFO (MainThread) [supervisor.api.middleware.security] /host/info access from a0d7b954_ssh
21-08-05 11:12:04 INFO (MainThread) [supervisor.api.middleware.security] /core/info access from a0d7b954_ssh
21-08-05 11:12:17 INFO (MainThread) [supervisor.api.middleware.security] /network/info access from a0d7b954_ssh
21-08-05 11:12:22 WARNING (MainThread) [supervisor.host.network] Can't update connectivity information: Error: Timeout was reached
21-08-05 11:13:02 WARNING (MainThread) [supervisor.host.network] Can't update connectivity information: Error: Timeout was reached
21-08-05 11:13:04 WARNING (MainThread) [supervisor.jobs] 'HomeAssistantCore.update' blocked from execution, no host internet connection
21-08-05 11:13:57 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
21-08-05 11:14:53 WARNING (MainThread) [supervisor.host.network] Can't update connectivity information: Error: Timeout was reached
21-08-05 11:15:00 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret google_drive_backup_password
21-08-05 11:15:33 WARNING (MainThread) [supervisor.host.network] Can't update connectivity information: Error: Timeout was reached
21-08-05 11:16:48 WARNING (MainThread) [supervisor.host.network] Can't update connectivity information: Error: Timeout was reached
21-08-05 11:17:08 WARNING (MainThread) [supervisor.jobs] 'HomeAssistantCore.update' blocked from execution, no host internet connection
```

core logs
```
2021-08-05 11:13:04 ERROR (MainThread) [homeassistant.components.hassio] Failed to to call /core/update - 'HomeAssistantCore.update' blocked from execution, no host internet connection
```

system info
```
➜  ~ ha network info                                  
docker:
  address: 172.30.32.0/23
  dns: 172.30.32.3
  gateway: 172.30.32.1
  interface: hassio
host_internet: false
interfaces:
- connected: true
  enabled: true
  interface: eth0
  ipv4:
    address:
    - 192.168.0.97/24
    gateway: 192.168.0.1
    method: static
    nameservers:
    - 192.168.0.1
  ipv6:
    address:
    - fe80::87f3:f7f6:b1f1:4caa/64
    gateway: null
    method: auto
    nameservers: []
  primary: true
  type: ethernet
  vlan: null
  wifi: null
- connected: false
  enabled: false
  interface: wlan0
  ipv4:
    address: []
    gateway: null
    method: disabled
    nameservers: []
  ipv6:
    address: []
    gateway: null
    method: disabled
    nameservers: []
  primary: false
  type: wireless
  vlan: null
  wifi: null
supervisor_internet: true
➜  ~ ha info        
arch: aarch64
channel: stable
docker: 20.10.6
features:
- reboot
- shutdown
- services
- network
- hostname
- timedate
- hassos
hassos: "6.2"
homeassistant: 2021.7.3
hostname: homeassistant
logging: info
machine: raspberrypi4-64
operating_system: Home Assistant OS 6.2
state: running
supervisor: 2021.06.8
supported: true
supported_arch:
- aarch64
- armv7
- armhf
timezone: Europe/London
➜  ~ ping www.hp.com
PING www.hp.com (23.55.59.11): 56 data bytes
64 bytes from 23.55.59.11: seq=0 ttl=57 time=20.365 ms
64 bytes from 23.55.59.11: seq=1 ttl=57 time=26.016 ms
64 bytes from 23.55.59.11: seq=2 ttl=57 time=16.085 ms
64 bytes from 23.55.59.11: seq=3 ttl=57 time=16.070 ms
➜  ~ curl https://version.home-assistant.io/online.txt
NetworkManager is online
```

agners, 2021-08-05:
Can you login using SSH to the underlying OS using [this guide](https://developers.home-assistant.io/docs/operating-system/debugging#ssh-access-to-the-host).

From there, check the following things:
```
curl https://version.home-assistant.io/online.txt
resolvectl query version.home-assistant.io
resolvectl status
```

karpaz, 2021-08-05:
> Can you login using SSH to the underlying OS using [this guide](https://developers.home-assistant.io/docs/operating-system/debugging#ssh-access-to-the-host).
>
> From there, check the following [...]

Hmm, It's pretty weird. 
```# journalctl -f -u hassos-supervisor.service
-- Journal begins at Tue 2021-02-02 15:29:48 UTC. --
Aug 05 17:06:43 homeassistant hassos-supervisor[939]: 21-08-05 19:06:43 INFO (MainThread) [supervisor.host.services] Updating service information
Aug 05 17:06:43 homeassistant hassos-supervisor[939]: 21-08-05 19:06:43 INFO (MainThread) [supervisor.host.network] Updating local network information
Aug 05 17:06:48 homeassistant hassos-supervisor[939]: 21-08-05 19:06:48 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
Aug 05 17:06:48 homeassistant hassos-supervisor[939]: 21-08-05 19:06:48 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state CoreState.RUNNING
Aug 05 17:06:48 homeassistant hassos-supervisor[939]: 21-08-05 19:06:48 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
Aug 05 17:06:51 homeassistant hassos-supervisor[939]: 21-08-05 19:06:51 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
Aug 05 17:06:51 homeassistant hassos-supervisor[939]: 21-08-05 19:06:51 INFO (MainThread) [supervisor.host] Host information reload completed
Aug 05 17:10:38 homeassistant hassos-supervisor[939]: 21-08-05 19:10:38 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.ADD usb hardware /dev/bus/usb/001/006
Aug 05 17:11:24 homeassistant hassos-supervisor[939]: 21-08-05 19:11:24 INFO (MainThread) [supervisor.hassos] Synchronizing configuration from USB with Home Assistant Operating System.
Aug 05 17:11:24 homeassistant hassos-supervisor[939]: 21-08-05 19:11:24 INFO (MainThread) [supervisor.host.services] Restarting local service hassos-config.service```

```
# docker logs hassos_supervisor
Error: No such container: hassos_supervisor
```

```
# docker logs homeassistant
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] udev.sh: executing... 
[cont-init.d] udev.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
2021-08-05 18:57:59 ERROR (MainThread) [homeassistant.components.hassio] Failed to to call /addons/core_ssh/stats - Addon is not installed
2021-08-05 18:58:03 ERROR (MainThread) [homeassistant.components.hassio] Failed to to call /addons/core_ssh/stats - Addon is not installed
[cont-finish.d] executing container finish scripts...
[cont-finish.d] done.
[s6-finish] waiting for services.
2021-08-05 19:00:11 ERROR (MainThread) [homeassistant.components.hassio.handler] Client error on /host/shutdown request Server disconnected
2021-08-05 19:00:11 ERROR (MainThread) [homeassistant.components.hassio] Failed to to call /host/shutdown - 
s6-svscanctl: fatal: unable to control /var/run/s6/services: supervisor not listening
[s6-finish] sending all processes the TERM signal.
[s6-finish] sending all processes the KILL signal and exiting.
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] udev.sh: executing... 
[cont-init.d] udev.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
2021-08-05 19:06:15 WARNING (MainThread) [homeassistant.setup] Setup of zone is taking over 10 seconds.
2021-08-05 19:06:15 WARNING (MainThread) [homeassistant.setup] Setup of media_source is taking over 10 seconds.
2021-08-05 19:06:15 WARNING (MainThread) [homeassistant.setup] Setup of system_health is taking over 10 seconds.
2021-08-05 19:06:15 WARNING (MainThread) [homeassistant.setup] Setup of logbook is taking over 10 seconds.
2021-08-05 19:06:15 WARNING (MainThread) [homeassistant.setup] Setup of input_boolean is taking over 10 seconds.
2021-08-05 19:06:15 WARNING (MainThread) [homeassistant.setup] Setup of input_text is taking over 10 seconds.
2021-08-05 19:06:15 WARNING (MainThread) [homeassistant.setup] Setup of timer is taking over 10 seconds.
```

These are my logs now, but I must admit that I tried firstly to install SSH from addon shop, then uninstalled that and acces to HomeAssistant with your tutorial.

It's problem, but I swear that I already saw the message `'HomeAssistantCore.update' blocked from execution, no host internet connection`
It will appear every day. It's start with message in logs, after that I cannot go to Supervisor, and every addon stop responding. When I want to restart HA, than it will show something about shutdown call taking over 10 seconds. 
After that all my sensors say that they are unavaiable
When I manually restart my Rpi 3B+ it will work normally, but next day it happen again. I try it tomorrow and update my logs. This logs doesn't say nothing.

vmartinv, 2021-08-11:
I was able to solve this... I was using manual mode because I have my own DNS servers running but this appear to have messed up with the OS.
I went to Supervisor->System->In the IP address section, pressed the change button->In IPv4 section of my connection I set an external DNS address (Google is 8.8.8.8), the previous one was wrong. It would be good if there were some link to troubleshooting in that error message.

Masterz69, 2021-08-25:
There no such an issue noticed before 2021.6.x versions. Connectivity stays FULL for some time (last time >12 hours), then become LIMITED. IMHO root cause not in external/local DNS servers.

gitarman94, 2021-09-23:
Just reinstalled hassio right off the website using latest of everything and I too am still receiving this error.  Previously I was using 6.4 and updated the OS version first, which then prevented me from updating the core. But ever since that update, I can't even download plugins now.

Masterz69, 2021-11-12:
So far in my case - after setting different IP addresses for wlan0 & eth0 - issue appear no more.

agners, 2022-01-13:
It seems that the issue in general is not reported often anymore. It seems that the update of systemd (which brings a new version of the DNS resolver systemd-resolved used on the host OS) helped. In any case, thanks for your contributions!

closed 2022-01-13 by agners

nagosuka, 2023-03-26:
I am currently having this issue which started sometime the past week...

nagosuka, 2023-03-27:
Nevermind. DNS issue with pi-hole