home-assistant/operating-system #1468: Container in use errors when updating HA core version or add-ons

sparkydave1981, 2021-07-19:
**Hardware Environment**

- [ ] Generic x86-64 (Intel NUC)

**Home Assistant OS release:**

- [ ] 2021.06.8

```
arch: amd64
audio_input: None
audio_output: None
boot: true
image: ghcr.io/home-assistant/generic-x86-64-homeassistant
ip_address: 172.30.32.1
last_version: 2021.7.3
machine: generic-x86-64
port: 8123
ssl: false
update_available: false
version: 2021.7.3
version_latest: 2021.7.3
wait_boot: 600
watchdog: true
```

**Supervisor logs:**

```
21-07-19 11:26:43 INFO (SyncWorker_5) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/unifi/amd64 with version 0.23.2
21-07-19 11:26:43 INFO (SyncWorker_10) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/motioneye/amd64 with version 0.15.0
21-07-19 11:26:43 INFO (SyncWorker_11) [supervisor.docker.interface] Attaching to homeassistant/amd64-addon-zwave_js with version 0.1.29
21-07-19 11:26:43 INFO (SyncWorker_9) [supervisor.docker.interface] Attaching to ghcr.io/einschmidt/caddy-2/amd64 with version 1.0.3
21-07-19 11:26:43 INFO (SyncWorker_8) [supervisor.docker.interface] Attaching to f8f3f8ff/amd64-addon-gmusicproxy with version 0.5.1
21-07-19 11:26:43 INFO (MainThread) [supervisor.snapshots] Found 5 snapshot files
21-07-19 11:26:43 INFO (MainThread) [supervisor.discovery] Loaded 3 messages
21-07-19 11:26:43 INFO (MainThread) [supervisor.ingress] Loaded 3 ingress sessions
21-07-19 11:26:43 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.SETUP
21-07-19 11:26:43 INFO (MainThread) [supervisor.resolution.check] System checks complete
21-07-19 11:26:43 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.SETUP
21-07-19 11:26:43 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
21-07-19 11:26:43 INFO (MainThread) [supervisor.jobs] 'ResolutionFixup.run_autofix' blocked from execution, system is not running - CoreState.SETUP
21-07-19 11:26:43 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.SETUP
21-07-19 11:26:43 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
21-07-19 11:26:43 INFO (MainThread) [__main__] Running Supervisor
21-07-19 11:26:43 INFO (MainThread) [supervisor.hassos] Rauc: B - marked slot kernel.1 as good
21-07-19 11:26:43 INFO (MainThread) [supervisor.addons] Phase 'AddonStartup.INITIALIZE' starting 0 add-ons
21-07-19 11:26:43 INFO (MainThread) [supervisor.addons] Phase 'AddonStartup.SYSTEM' starting 2 add-ons
21-07-19 11:26:43 WARNING (MainThread) [supervisor.addons.options] Option 'anonymous' does not exist in the schema for Mosquitto broker (core_mosquitto)
21-07-19 11:26:43 INFO (SyncWorker_11) [supervisor.docker.interface] Cleaning addon_core_mosquitto application
21-07-19 11:26:43 WARNING (SyncWorker_11) [supervisor.addons.options] Option 'anonymous' does not exist in the schema for Mosquitto broker (core_mosquitto)
21-07-19 11:26:43 INFO (SyncWorker_11) [supervisor.docker.addon] Starting Docker add-on homeassistant/amd64-addon-mosquitto with version 6.0.1
21-07-19 11:26:43 INFO (SyncWorker_4) [supervisor.docker.addon] Starting Docker add-on homeassistant/amd64-addon-mariadb with version 2.4.0
21-07-19 11:26:44 INFO (MainThread) [supervisor.services.modules.mqtt] Set core_mosquitto as service provider for mqtt
21-07-19 11:26:45 INFO (MainThread) [supervisor.services.modules.mysql] Set core_mariadb as service provider for MySQL
21-07-19 11:26:46 INFO (MainThread) [supervisor.auth] Auth request from 'core_mosquitto' for 'hassio_mqtt'
21-07-19 11:26:46 INFO (MainThread) [supervisor.auth] Home Assistant not running, checking cache
21-07-19 11:26:48 INFO (MainThread) [supervisor.addons] Phase 'AddonStartup.SERVICES' starting 6 add-ons
21-07-19 11:26:48 INFO (SyncWorker_11) [supervisor.docker.interface] Cleaning addon_core_ssh application
21-07-19 11:26:48 INFO (SyncWorker_11) [supervisor.docker.addon] Starting Docker add-on homeassistant/amd64-addon-ssh with version 9.1.3
21-07-19 11:26:49 INFO (SyncWorker_4) [supervisor.docker.interface] Cleaning addon_a0d7b954_unifi application
21-07-19 11:26:49 INFO (SyncWorker_4) [supervisor.docker.addon] Starting Docker add-on ghcr.io/hassio-addons/unifi/amd64 with version 0.23.2
21-07-19 11:26:49 INFO (SyncWorker_9) [supervisor.docker.interface] Cleaning addon_a0d7b954_vscode application
21-07-19 11:26:49 INFO (SyncWorker_9) [supervisor.docker.addon] Starting Docker add-on ghcr.io/hassio-addons/vscode/amd64 with version 3.6.0
21-07-19 11:26:49 INFO (SyncWorker_5) [supervisor.docker.interface] Cleaning addon_6a3d001f_assistant_relay application
21-07-19 11:26:49 INFO (SyncWorker_5) [supervisor.docker.addon] Starting Docker add-on apipa169/amd64-hassio-assistant_relay with version None
21-07-19 11:26:49 WARNING (MainThread) [supervisor.addons.options] Option 'interface' does not exist in the schema for Samba share (core_samba)
21-07-19 11:26:49 INFO (SyncWorker_1) [supervisor.docker.interface] Cleaning addon_core_samba application
21-07-19 11:26:49 WARNING (SyncWorker_1) [supervisor.addons.options] Option 'interface' does not exist in the schema for Samba share (core_samba)
21-07-19 11:26:49 INFO (SyncWorker_1) [supervisor.docker.addon] Starting Docker add-on homeassistant/amd64-addon-samba with version 9.5.1
[... 43 lines cut ...]
21-07-19 11:27:08 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
21-07-19 11:29:00 INFO (MainThread) [supervisor.api.middleware.security] /network/info access from core_ssh
21-07-19 11:29:00 INFO (MainThread) [supervisor.api.middleware.security] /network/info access from core_ssh
21-07-19 11:29:00 INFO (MainThread) [supervisor.api.middleware.security] /host/info access from core_ssh
21-07-19 11:29:00 INFO (MainThread) [supervisor.api.middleware.security] /core/info access from core_ssh
21-07-19 11:29:05 INFO (MainThread) [supervisor.api.middleware.security] /core/info access from core_ssh
21-07-19 11:29:23 INFO (MainThread) [supervisor.api.middleware.security] /core/stop access from core_ssh
21-07-19 11:29:32 INFO (MainThread) [supervisor.api.middleware.security] /core/update access from core_ssh
21-07-19 11:29:32 INFO (MainThread) [supervisor.homeassistant.core] Updating Home Assistant to version 2021.7.3
21-07-19 11:29:32 INFO (SyncWorker_7) [supervisor.docker.interface] Updating image ghcr.io/home-assistant/generic-x86-64-homeassistant:2021.7.2 to ghcr.io/home-assistant/generic-x86-64-homeassistant:2021.7.3
21-07-19 11:29:32 INFO (SyncWorker_7) [supervisor.docker.interface] Downloading docker image ghcr.io/home-assistant/generic-x86-64-homeassistant with tag 2021.7.3.
21-07-19 11:29:37 INFO (MainThread) [supervisor.homeassistant.core] Successful started Home Assistant 2021.7.3
21-07-19 11:34:41 INFO (MainThread) [supervisor.api.middleware.security] /supervisor/info access from core_ssh
21-07-19 11:35:16 INFO (MainThread) [supervisor.api.middleware.security] /core/info access from core_ssh
21-07-19 11:36:09 INFO (MainThread) [supervisor.api.middleware.security] /supervisor/logs access from core_ssh
```

**Journal logs:**

**Kernel logs:**

**Description of problem:**

Whenever I try to update HA Core to a new version I have major issues.  I get errors relating to the Container being is use / name already existing.  The same happens if I try to update an Add-on which is running.
The only solution (temporary) has been to stop HA (or the add-on), update, then start again.  In this case it sometimes works however it also requires a Host Reboot.

sparkydave1981, 2021-07-19:
https://community.home-assistant.io/t/multiple-add-ons-fail-to-start-with-same-error/320487

agners, 2021-07-19:
This seems more related to Supervsior, moving.