home-assistant/operating-system #1470: Memory and Swap spike resulting in crash of HASS on VMware

awdark, 2021-07-19:
**Hardware Environment**
- [X] OVA (Open Virtualization Appliance, on Intel NUC or any other hardware, please add the Hypervisor you are using)
- On VMWare workstation allocated 4gb ram 16gb drive space

**Home Assistant OS release:**
- [ ] Fresh installation of release x.y
- [x] Updated from version - core-2021.6.x

- Additional information (if accessible):
## System Health

version | core-2021.7.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
virtualenv | false
python_version | 3.9.5
os_name | Linux
os_version | 5.10.45
arch | x86_64
timezone | America/Los_Angeles

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 5000
Installed Version | 1.13.2
Stage | running
Available Repositories | 848
Installed Repositories | 3

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 6.1
-- | --
update_channel | stable
supervisor_version | supervisor-2021.06.8
docker_version | 20.10.6
disk_total | 14.1 GB
disk_used | 9.3 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | Duck DNS (1.13.0), File editor (5.3.2), Mosquitto broker (6.0.1), TasmoAdmin (0.15.0), Nginx Proxy Manager (0.11.0), InfluxDB (4.1.1), Node-RED (9.2.0), ESPHome (1.19.4), MariaDB (2.4.0), Terminal & SSH (9.1.3), AirCast (3.1.1), Samba Backup (4.5.0)

</details>

<details><summary>Lovelace</summary>

dashboards | 3
-- | --
resources | 2
views | 5
mode | storage

</details>

**Supervisor logs:**
future: <Task finished name='Task-32046' coro=<Repository.update() done, defined at /usr/src/supervisor/supervisor/store/repository.py:106> exception=StoreGitError()>
Traceback (most recent call last):
  File "/usr/src/supervisor/supervisor/store/git.py", line 155, in pull
    await self.sys_run_in_executor(
  File "/usr/local/lib/python3.9/concurrent/futures/thread.py", line 52, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/local/lib/python3.9/site-packages/git/remote.py", line 828, in fetch
    res = self._get_fetch_info_from_stderr(proc, progress)
  File "/usr/local/lib/python3.9/site-packages/git/remote.py", line 702, in _get_fetch_info_from_stderr
    proc.wait(stderr=stderr_text)
  File "/usr/local/lib/python3.9/site-packages/git/cmd.py", line 447, in wait
    raise GitCommandError(remove_password_if_present(self.args), status, errstr)
git.exc.GitCommandError: Cmd('git') failed due to: exit code(128)
  cmdline: git fetch -v --update-shallow --depth=1 origin
  stderr: 'fatal: unable to access 'https://github.com/home-assistant/addons/': SSL connection timeout'
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/usr/src/supervisor/supervisor/store/repository.py", line 110, in update
    await self.git.pull()
  File "/usr/src/supervisor/supervisor/jobs/decorator.py", line 107, in wrapper
    raise err
  File "/usr/src/supervisor/supervisor/jobs/decorator.py", line 105, in wrapper
    return await self._method(*args, **kwargs)
  File "/usr/src/supervisor/supervisor/store/git.py", line 195, in pull
    raise StoreGitError() from err
supervisor.exceptions.StoreGitError
21-07-19 07:25:29 ERROR (MainThread) [asyncio] Task exception was never retrieved
future: <Task finished name='Task-32044' coro=<Repository.update() done, defined at /usr/src/supervisor/supervisor/store/repository.py:106> exception=StoreGitError()>
Traceback (most recent call last):
  File "/usr/src/supervisor/supervisor/store/git.py", line 155, in pull
    await self.sys_run_in_executor(
  File "/usr/local/lib/python3.9/concurrent/futures/thread.py", line 52, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/local/lib/python3.9/site-packages/git/remote.py", line 828, in fetch
    res = self._get_fetch_info_from_stderr(proc, progress)
  File "/usr/local/lib/python3.9/site-packages/git/remote.py", line 702, in _get_fetch_info_from_stderr
    proc.wait(stderr=stderr_text)
  File "/usr/local/lib/python3.9/site-packages/git/cmd.py", line 447, in wait
    raise GitCommandError(remove_password_if_present(self.args), status, errstr)
git.exc.GitCommandError: Cmd('git') failed due to: exit code(128)
  cmdline: git fetch -v --update-shallow --depth=1 origin
  stderr: 'fatal: unable to access 'https://github.com/esphome/hassio/': SSL connection timeout'
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/usr/src/supervisor/supervisor/store/repository.py", line 110, in update
    await self.git.pull()
  File "/usr/src/supervisor/supervisor/jobs/decorator.py", line 107, in wrapper
    raise err
  File "/usr/src/supervisor/supervisor/jobs/decorator.py", line 105, in wrapper
    return await self._method(*args, **kwargs)
  File "/usr/src/supervisor/supervisor/store/git.py", line 195, in pull
    raise StoreGitError() from err
supervisor.exceptions.StoreGitError
21-07-19 07:25:31 INFO (MainThread) [supervisor.jobs] 'StoreManager.update_repositories' blocked from execution, no supervisor internet connection
21-07-19 07:25:31 INFO (MainThread) [supervisor.store] Loading add-ons from store: 66 all - 0 new - 0 remove
21-07-19 07:25:31 INFO (MainThread) [supervisor.plugins.observer] Starting observer plugin
21-07-19 07:25:31 INFO (MainThread) [supervisor.homeassistant.core] Wait until Home Assistant is ready
21-07-19 07:25:33 INFO (SyncWorker_1) [supervisor.docker.observer] Starting Observer ghcr.io/home-assistant/amd64-hassi

**Journal logs:**

**Kernel logs:**

**Description of problem:**

On the Home assistant console my best understanding is my ram and swap is filling up resulting in a crash of the python process  responsible for home assistant.

[24943.464628] Out of memory: Killed process 12573 (python3) total-vm:2041988kB, anon-rss:1524880kB, file-rss:0kB, shmem-rss:0kB, UID:0 pgtables:3996kB oom_score_adj:0

[image]
[image]
[image]

I understand it can obviously be an integration issue, user issue, etc.  This issue became very apparent after the recent updates in June or July but I have screwed myself over with my personal limited backup retention policy.  I know it started with the July update and I thought it was a reverse proxy issue but it was actually the HA application crashing rather than an access issue. 

I have tried to disable many of my sensors and integrations but it feels like I am going into this blindly so I am not sure what to target.  Is there a way to increase the verbosity of the logging?

agners, 2021-07-21:
I'd check if its Home Assistant Core or the Supervisor process which is crashing (both are services run by a `python3` processes). You can check by listing all `python3` processes using `ps xfa | grep python3` and then once the process crashes again check which PID it matches with.

awdark, 2021-07-22:
Thank you for your help, I apologize but I think I am running it incorrectly.

I deployed this a while back using the vmware OVA option so I might be in a shell or wrong shell.  
I tried using the Terminal & SSH add on but the output seems limited
[image]

In what I believe is the console from VMware I see this which does not recognize the PS command but I do see that I am in a ha> prompt (home assistant command line) rather than a proper SSH.  Ctrl+C does not quit it and just takes me back to the top of the command line. 
[image]

agners, 2021-07-22:
On the VMware console, you have to use `login` to access the OS shell.

awdark, 2021-07-23:
Nice, that worked 

[image]

After it ran out of memory and process was killed
[image]
so python3 -m homeassistant --config config

agners, 2021-07-23:
Ok, this seems to be caused by a memory leak in Core then. Moving.