home-assistant/operating-system #1471: Boot Hangs on Odroid N2

marcosgmeli, 2021-07-19:
**Hardware Environment**

- [x] ODROID N2

**Home Assistant OS release:**
- [x] Fresh installation of release 6.1

- Additional information (if accessible):

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**

Hi there, my Odroid N2, cant boot, with fresh 6.1 HassOS(first installation), flashed on a 16GB eMMC.
Sometimes it bootloops and sometimes it just hangs, Its not always the same point where it hangs, there can be seen some errors about memory but not sure if its the cause.
Already tried other OS on the same eMMC and works flawlessly.
I will share some photos  of what I see on the monitor, is there a way to access those logs in text?.
The most recurrent log is the "Mem abort info:" and "Unable to handle kernel ...(different options)"
[image]
[image]
[image]
[image]

agners, 2021-07-23:
You can use a serial console using the [USB-UART Module Kit](https://www.hardkernel.com/shop/usb-uart-module-kit/).

This looks like serious crashes, probably cause by memory issues. I have OS 6.1 running on my ODROID-N2+ instance since it came out, so I don't think that this is a general problem. Is this a ODROID-N2 or ODROID-N2+?

Is that other OS you are using working stable over time? Can you run a memory test tool (e.g. stress-ng or similar)?

marcosgmeli, 2021-07-25:
Its ODROID-N2(not plus), it have been stable, I will try with stress-ng to see if something come up

marcosgmeli, 2021-07-25:
Tested on Ubuntu Mate 20.04 with the next stress-ng options:
```
stress-ng --cpu 6 --io 4 --vm 2 --vm-bytes 75% -t 60s --verify --metrics-brief
stress-ng: info:  [242878] dispatching hogs: 6 cpu, 4 io, 2 vm
stress-ng: info:  [242878] successful run completed in 60.22s (1 min, 0.22 secs)
stress-ng: info:  [242878] stressor       bogo ops real time  usr time  sys time   bogo ops/s   bogo ops/s
stress-ng: info:  [242878]                           (secs)    (secs)    (secs)   (real time) (usr+sys time)
stress-ng: info:  [242878] cpu                7467     60.19    164.77      7.99       124.06        43.22
stress-ng: info:  [242878] io               404161     60.00      1.53     56.19      6735.60      7002.10
stress-ng: info:  [242878] vm               461360     60.10     44.34      6.71      7676.06      9037.41
```
Nothing strange as far as I understand

agners, 2021-07-26:
> Ubuntu Mate 20.04

What kernel is that build using (`uname -a`)?

marcosgmeli, 2021-07-27:
`Linux odroid 4.9.241-109 #1 SMP PREEMPT Mon Jan 25 14:34:28 UTC 2021 aarch64 aarch64 aarch64 GNU/Linux`

marcosgmeli, 2021-07-27:
Maybe is related to this other issue? https://github.com/home-assistant/operating-system/issues/1096

agners, 2021-07-28:
Possibly. 

One thing which would explain this issue is wrong/bad memory timings. Memory is setup by the early boot code. But as far as I am aware we use the same memory timing as Hardkernel is using in their Ubuntu builds.

Is this the 2GB or 4GB variant?

marcosgmeli, 2021-07-28:
Let me know if you need to test something else on the board, to compare with the plus version, I suppose that the problem its a memory difference between those two.

agners, 2021-07-29:
Just checked, the last changes to memory timing have been made early last year, and those are part of HAOS:
https://github.com/hardkernel/u-boot/commit/3b8b5cbd956b22080925dd3beaaac463452c12f1

One thing which would be interesting is how another mainline Linux based distribution behaves. Can you test [Armbian](https://www.armbian.com/odroid-n2/), it uses the 5.10 kernel as well.

marcosgmeli, 2021-07-29:
Tested Armbian buster and Armbian buster with xfce , both are not showing anything over HDMI, cant say if the same problem because just have a black screen, but looks like its not being able to boot.

marcosgmeli, 2021-07-29:
6.2 is crashing at boot 
[image]

marcosgmeli, 2021-08-02:
I was able to connect a uart adapter and get the logs:
[haoslogs.txt](https://github.com/home-assistant/operating-system/files/6913655/haoslogs.txt)
Also reported it at armbian forums: https://forum.armbian.com/topic/18732-odroid-n2not-plus-not-booting/

marcosgmeli, 2021-08-04:
Hi @agners, founded this thread : https://forum.odroid.com/viewtopic.php?f=176&t=40915
looks like the solution to the issue, but not sure...

agners, 2021-08-06:
@marcosgmeli that is unlikely to be related. Our boot script uses the load addresses from upstream U-Boot anyways. If load addresses would be wrong, you would not get that far anyways (those load addresses are set to load the kernel into memory. In your situation the kernel boots fine).

One thing I just noticed: You are using a rather high resolution screen. Have you tried with a lower resolutions screen/no screen?

marcosgmeli, 2021-08-06:
Yes I have tried without screen, the file of logs that i got from tx/rx was without screen

roybosch, 2021-08-08:
I'm having the same problem on my N2+. The 6.1 update didn't succeed, same goes for 6.2. It just never comes back online.

agners, 2021-09-03:
@roybosch you are seeing similar stack traces on the screen?

roybosch, 2021-09-20:
> @roybosch you are seeing similar stack traces on the screen?

Today I again tried an update. This time to 6.4. Again, home assistant did not come back online after. I was afraid I had to reïnstall and restore the backup again.

I took the Odroid board to my computer and connected the HDMI to my monitor. When I did that, I found out that the CLI came online and after a little while it came back online. The update was finished. 

Then I took the board back to the server cabinet, powered it up, and it didn't come back online. Because I used the same power adapter, I first thought it could be some noise on the power line as I'm having more devices powered in that cabinet. I measured the voltage on the adapter, but it was good. Next possibility was the internet cable. A CAT5E cable. I measured the speed and I concluded that it wasn't as high as it could be (94Mbit/s). When trying another CAT5E cable connected to the same switch, the speeds reached up to 300+Mbit/s. 

Leaving the better ethernet cable connected I powered the board up again. This time it came back online in no time. I find it very strange that after OS 5.13, it stopped working with the same cable that worked at 5.13 and older. Is there a simple explanation for this?

TLDR; Everything works now I changed the UTP cable.

liudger, 2021-09-22:
> Then I took the Odroid N2+ board back to the server cabinet, powered it up, and it didn't come back online. Because I used the same power adapter, I first thought it could be some noise on the power line as I'm having more devices powered in that cabinet. I measured the voltage on the adapter, but it was good and stable. Next possibility was the internet cable. A CAT5E cable. I measured the speed and I concluded that it wasn't as high as it could be (94Mbit/s). But high enough, right? When trying another CAT5E cable connected to the same switch, the speeds reached up to 300+Mbit/s.
> 
> Leaving the better ethernet cable connected I powered the board up again. This time it came back online in no time. I find it very strange that after OS 5.13, it stopped working with the same cable that worked at 5.13 and older. Is there a simple explanation for this?

Can it be the UTP cable? I have small patch cable. So this seems very strange that something would be wrong with it. 
Let's see how long it stays up! 🤞

roybosch, 2021-09-22:
Yes it was definitely the UTP cable. But I can't explain why the same cable works on 5.13 but stops working from 6.0 and higher.

I'm using another cable now, which doesn't give me the trouble.

closed 2021-12-28 by github-actions[bot]