home-assistant/operating-system #1472: HA OS seems to use way to much ram for cache and swaps too much -> slower performance

FaserF, 2021-07-21:
**Hardware Environment**

- [x] Raspberry Pi [4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Generic x86-64 (like Intel NUC)
- [ ] OVA (Open Virtualization Appliance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [ ] Fresh installation of release x.y
- [x] Updated from version 6.0 to 6.1 (problem could already be there prior 6.0)
- Additional information (if accessible):

## System Health

version | core-2021.7.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
virtualenv | false
python_version | 3.9.5
os_name | Linux
os_version | 5.10.17-v8
arch | aarch64
timezone | Europe/Berlin

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 4303
Installed Version | 1.13.2
Stage | running
Available Repositories | 915
Installed Repositories | 20

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 6.1
-- | --
update_channel | stable
supervisor_version | supervisor-2021.06.8
docker_version | 20.10.6
disk_total | 234.0 GB
disk_used | 42.1 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | SSH & Web Terminal (9.0.0), ADB - Android Debug Bridge (0.8.0), AdGuard Home (4.1.6), Home Assistant Google Drive Backup (0.104.3), MariaDB (2.4.0), Mosquitto broker (6.0.1), AirCast (3.1.1), Assistant Relay (0.7.4), Visual Studio Code (3.6.0), Bitwarden (Vaultwarden) (0.12.0), Google Assistant SDK (2.5.0), Nextcloud (20.0.3), heimdall (0.1.1), Nginx Proxy Manager (0.11.0), Wiki.JS (1.1.0), Apache2 Minimal with MariaDB Client (2.1.0), Bash Script Executer (1.2.0), xqrepack (1.2.0), Freenom-dns-updater (1.0.0), Switch LAN Play Server (1.2.0), Switch LAN Play Client (1.1.0)

</details>

<details><summary>Lovelace</summary>

dashboards | 2
-- | --
resources | 9
views | 12
mode | storage

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**

[image]

HA OS seems to cache way too much stuff (yellow bar) and then swaps out much stuff. This makes my raspberry pi 4 feel very slow.

A friend of mine (Raspberry Pi 4) has the same issue.

Is there anything to improve this? Is this a bug or a "feature"?

agners, 2021-07-21:
> HA OS seems to cache way too much stuff

Yellow is probably mostly page cache, and is very important to make reads faster. Essentially Linux tries to keep files (code or data) in memory as much as possible to speed up subsequent reads.

Usually things only get swapped out when its really not used for a long time (but needs to be in memory).

Imagine Home Assistant reads data (e.g. from database) every time you open your dashboard. You want your Raspberry Pi to keep that in memory. Whereas some program code from a service started in the background which is used very seldom, should be written out to swap to not take up valuable memory.

HAOS is built on Linux, which in terms has a very sophisticated algorithms to manage memory. We don't tune memory management special, and it has served well so far.

Btw: In HAOS, swap is also in memory, but compressed using zram.  This is done to avoid wearing out SD cards. One might ask why use swap then even? Its a trade off: Compressed swap allows to store more in memory at the cost of CPU time. We basically trade more memory with some CPU cycles.

> Is there anything to improve this? Is this a bug or a "feature"?

There are tunables in Linux, and in theory you can minimize caching. But your system is pretty much guaranteed to feel slower. In the end just "tune" for the sake of moving a bar is not helpful. You need to dig into why your Raspberry Pi 4 feels slow exactly. I don't have a straight forward recipe how to do that. Monitoring CPU/memory load, and trying different configurations might help to find what improves the situation.

I am closing this issue as its not really an issue.

closed 2021-07-21 by agners

FaserF, 2021-07-21:
Wow, thank you so much for such a detailed answer!