home-assistant/operating-system #1478: ESXi 7.0 on ARM issues

cladand, 2021-07-27:
I'm trying to install haos 6.1 on ESXi 7.0 on ARM.

1) I tried to use the vmdk linked to a brand new VM and no matter what I change in the parameters, I receive the attached error
[image]
I also tried to copy the vmdk with vmkfstools -i but same problem.

2) I tried to use the ESXi OVA, but it gives me a VMX compatibility error.
How can I proceed?

closed 2021-07-28 by cladand

agners, 2021-07-28:
@cladand  sooo.. works?

auanasgheps, 2021-07-28:
@agners  nope it doesn't, but by design. I helped @cladand and figured out he was trying to boot [the generic x86/64 image](https://www.home-assistant.io/installation/generic-x86-64) on a Raspberry Pi (ARM) with ESXi. 
The problem is that nobody, at least now, does instruction emulation over virtualization, so will never work unless you'll build an ARM image.

agners, 2021-07-28:
@auanasgheps yeah that make sense.

Note that this also came up a while ago here: #1050. While it should't be too hard to make a vmdk from a Raspberry Pi built image, the problem will be the boot flow: From what I understand ESXi on ARM uses UEFI boot flow. That is not what our RPi OS builds to currently... Not impossible, but involves certainly some work. Currently this is a bit too niche use case to focus on.