home-assistant/operating-system #1483: Host internet considered down because online.txt returns hCAPTCHA

agners, 2021-07-28:
@DexterityGerm notes in https://github.com/home-assistant/supervisor/issues/3011#issuecomment-888013566

> In my case, this message appears due to the fact that when contacting https://version.home-assistant.io/online.txt , it issues hCAPTCHA, so HAOS does not receive the desired response “NetworkManager is online”.
> 
> This is my problem. Having connected via VPN, everything works correctly.
> 
> Question: how to convince hCAPTCHA that I am not a bot and remove my IP from blocking?

_Originally posted by @DexterityGerm in https://github.com/home-assistant/supervisor/issues/3011#issuecomment-888013566_

agners, 2021-07-28:
Interesting find. We updated the configuration for `online.txt` to avoid hCAPTCHA being issued. This should fix the problem in your case.

closed 2021-07-28 by agners

isaeedm, 2021-12-22:
I am behind a proxy that keeps getting blocked by captcha , and i am facing the same error with OS6.6, Core 2021.11, Supervisor 2021.10.

Is there a way for me to override this setting by pointing it somewhere without the captcha?

agners, 2021-12-23:
@isaeedm the hCAPTCHA on the server end is no longer active, so if you get this message its some other reason (e.g. your proxy blocks access to that file/serves a captcha or does something weird).

 Currently there is way to override the URL (it is configured in `/etc/NetworkManager/NetworkManager.conf` which is in the read-only section of the OS).