home-assistant/operating-system #1489: SSL problem with the new HA OS 6.2

victorputroagung, 2021-08-03:
**Hardware Environment**
Raspberry Pi [4]

**Home Assistant OS release:**
Updated from version 6.1 to 6.2

**Description of problem:**
after updating to 6.2 from 6.1
can't access to grafana graph or tasmo admin gui at another port.
i suspect that the SSL is the problem.

nothing changed at my config, nor do my network.
tried to fresh install through balena again but the problem still persist.

[image]
[image]

agners, 2021-08-03:
Hm, those we interface are not provided by the operating system itself, but by the Tasmo/Graphana add-on.

To validate that the OS indeed is not the problem, can you downgrade to 6.1? Using the following command in the terminal should do it:

```
ha os update --version 6.1
```

victorputroagung, 2021-08-04:
> Hm, those we interface are not provided by the operating system itself, but by the Tasmo/Graphana add-on.
> 
> To validate that the OS indeed is not the problem, can you downgrade to 6.1? Using the following command in the terminal should do it:
> 
> ```
> ha os update --version 6.1
> ```

ill try to downgrade it today, will post an update later :)

victorputroagung, 2021-08-04:
> Hm, those we interface are not provided by the operating system itself, but by the Tasmo/Graphana add-on.
> 
> To validate that the OS indeed is not the problem, can you downgrade to 6.1? Using the following command in the terminal should do it:
> 
> ```
> ha os update --version 6.1
> ```

you were right, seems like the problem is at the ISP itself. ill close the issue, thanks for helping! :)

closed 2021-08-04 by victorputroagung