home-assistant/operating-system #1492: Xen Guest tools not being detected

Maelstrom96, 2021-08-04:
**Hardware Environment**

- [ ] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Generic x86-64 (like Intel NUC)
- [x] OVA (Open Virtualization Appliance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [ ] Fresh installation of release x.y
- [x] Updated from version 6.1 to 

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**

See: https://community.home-assistant.io/t/add-xen-guest-tools-package-to-hass-os/219573/7

PR #1459 added support for Xen Tools, but my preliminary testing doesn't show the expected result. I've installed the nightly version 7.0.dev20210719 which does seem to have the PR code in it, but the hypervisor is not seeing the management agent.

Maelstrom96, 2021-08-04:
This issue was a reporting error from XCP-ng console and can be closed, I've updated the community thread with the relevant informations.

closed 2021-08-04 by Maelstrom96

Trufax, 2021-12-29:
@Maelstrom96 hey can you tell me how you fixed the issue ? According to the pull requests guest tools should already be in, however XCP-NG does not recognize it.

Maelstrom96, 2021-12-29:
@Trufax I haven't confirmed if the PR is found in the 7.0 release, but you shouldn't have to do anything in order for the agent to work. I'm still running 7.0.dev20210719 and haven't tested the latest release yet. I'll see if I can validate it soon.

Maelstrom96, 2022-01-24:
@Trufax Updated to 7.1 and the management agent looks to be working correctly. It won't show that the management agent is detected in XCP-ng console, or XenServer console, but Xen-Orchestra displays that the agent is detected. To make sure, I've attempted to hotplug a new network interface and it worked as expected, which indicates that the agent is working.

Trufax, 2022-01-24:
Hi, i dont know when (so with which update) but today i checked again and the Agent is properly detected by Xcp-NG. So all works as expected now. Thanks for the feedback.